jgyates/genmon #67: updating question

ts4s, 2018-01-26:
Hello, It's been a while and i'm looking at update my genmon software (just a few version behind). What the best way to go about doing it?

jgyates, 2018-01-26:
No problem.

If you use Windows, I would install PuTTY as it contains a graphical ssh / scp tool. These tools (ssh and scp) are included in linux and macOS.
http://ged.msu.edu/angus/tutorials/using-putty-on-windows.html

Use a secure shell (ssh) to make a terminal connection on your PC into your pim then refresh your git repository. 

Open a command prompt on a linux or macOS system and type (use PuTTY for Windows):

     ssh pi@IPADDRESS

you will be prompted for the username of the pi user on your pi. Then change directories into your genmon directory and update the git repository by typing:
    
    git pull
    
this will update to the latest, or, if you have modified any files and want to overwrite them then type:

    git reset --hard origin/master

This command will update your entire repository to the latest. It will not overwrite your /etc/genmon.conf file.  You will need to change the If you want to use any new options you will need to update your file manually.  If you do a "git reset"  you will need to type "sudo chmod 755 ./startgenmon.sh" to change the permissions again like you did during install. Once your files are updated you can restart genmon via the ./startgenmon.sh scripts, or just reboot your system (sudo reboot now)

ts4s, 2018-01-26:
OK, i'm struggling here... i type git pull and i get this back: fatal: Not a git repository (or any of the parent directories): .git

any suggestion? i did some googling but i'm not getting anywhere... git noob

jgyates, 2018-01-26:
Using git to update will only work if you installed via this command:

     git clone https://github.com/jgyates/genmon.git

If you did then, verify that you are in the directory that you installed genmon. For example on my system I logged into the system via ssh as the user pi. So when I log in it will be in the /home/pi directory. From there I issue the command "cd genmon" so I am now in the /home/pi/genmon directory. Once in the /home/pi/genmon directory then I issue the command "get pull"

The way git works is that it stores information about the original files in hidden files when you first do the clone. When you update via the "git pull" command it checks what is on your system vs what is in the repository online and updates as needed.

jgyates, 2018-01-26:
If you are still having problems then send the output of this command"

     ls -la

ts4s, 2018-01-26:
well, i just cheated the system... move the genmon to genmon-old and then ran the git clone. test the git pull and it work so i should be good for next time. 

here's the file rights: 
[image]

jgyates, 2018-01-26:
ok. Glad to hear it is working.

closed 2018-01-26 by jgyates

ts4s, 2018-01-27:
[image]
 
sorry, long story short... went back to orginal sd card and go it function like it was before trying to update. Now i ran the update and this screenshot is where i am at. just wondering how i should proceed?

jgyates, 2018-01-27:
It looks like the files listed have been modified or at least the time stamp has changed since the original checkout. You can delete them and to a pull again or you can issue this command:

     git reset --hard origin/master

This will overwrite any changed files with the latest version.

jgyates, 2018-01-27:
FYI, I added a wiki page that includes this info:

https://github.com/jgyates/genmon/wiki/Updating-to-the-Latest-Version

ts4s, 2018-01-27:
ok, well i back at square one trying to figure out what "new change" i need to make to get it up and running.

here where i am at: 

pi@GeneratorPi:~/genmon $ ./startgenmon.sh stop
Stopping genmon python scripts
pi@GeneratorPi:~/genmon $ ./startgenmon.sh start
Starting genmon python scripts
Traceback (most recent call last):
  File "/home/pi/genmon/genmon.py", line 21, in <module>
    import datetime, time, sys, smtplib, signal, os, threading, socket, serial, pytz
ImportError: No module named pytz
pi@GeneratorPi:~/genmon $ Traceback (most recent call last):
  File "/home/pi/genmon/genserv.py", line 13, in <module>
    import sys, signal, os, socket, atexit, configparser
ImportError: No module named configparser

webpage will not open, i've only edit and copied the genmon.conf too /etc. i'm not seeing an log files as of yet. guessing i need to check on the pytz?

jgyates, 2018-01-27:
If it has been a while since your install, you will need to install a time zone module with this command:

    sudo pip install pytz

And possibly

    sudo pip install configparser

These modules were added with new features.