jgyates/genmon #68: Status screen reported frequency

jeepjungle, 2018-01-31:
Really enjoying using this genmon!  I have a question about the status screen I get when my 36KW Liquid Cooled Nexus controlled Generac is running.

The RPM (1800) and Voltage (240) readings look right but the frequency displayed is 30HZ.  I guess I would expect this to read 60Hz.  Do I have something set wrong in my config file?

Thanks,
Rob

jgyates, 2018-02-01:
Hi Rob,

Hmm, that is a bit odd. It could be that for your controller the software needs to double this value. If you could validate one thing first before I make a change: Validate that your generator is not utilizing a quiet mode for the test cycle. Quiet mode will run the engine at a lower speed when running the weekly exercise. I am almost sure this is not the case since your RPMs are 1800 (this is the non quiet mode RPM level for liquid cooled gensets). My gen is a liquid cooled to but it is Evolution, not Nexus.

Once you validate this let me know and we can go from there.

Thanks
Jason

jeepjungle, 2018-02-01:
Remote Started it Just now:

[image]

proppilot, 2018-02-01:
I suspect it is in quiet mode.  At half the RPM you would see half the frequency.

jeepjungle, 2018-02-01:
Quiet mode is 1500 RPM.  Standard run Mode is 1800 RPM.  Some generators are made to run at 3600 RPM but not ones like this that have larger engines (2.4L turbo).  I was just wondering if other folks who have water cooled Nexus controlled Generacs that run at 1800 RPM are also seeing 30Hz in the status display.  Not a big deal, just has me scratching my head.

jgyates, 2018-02-01:
@jeepjungle , agreed. Since you are a diesel, 1800 RPM should be full speed (3600 full speed for NG/LP), so 30Hz is wrong. I just uploaded a version of genmon that should multiply the number I read from the frequency register by 2. Test this and see if it works. One other thing you can do to validate is check the Frequency on the display (From the main menu, select "Status") and validate you are truly running at 60Hz. If the numbers on the display match what the new version of genmon is reporting we are golden. Let me know how it goes.

jeepjungle, 2018-02-01:
Did git pull and reboot.  It now display 60Hz and 1800RPM while in normal run mode.  I double checked the Nexus status display and it also shows 1800RPM and 60Hz.  So the code change fixed my genmon display.

But to be clear, this has a 2.4L turbo Mitsubishi engine running from NG supply and is designed to run at 1800 RPM.  Hopefully the code change won't throw off the folks with 3600 RPM NG generators.  I appreciate your help greatly.  Let me know if you want me to perform any other tests or get any register/mem values for you.

Thanks,
Rob

jgyates, 2018-02-01:
ok. my mistake on the diesel. If the generator is air cooled it will not effect them. This change only affects Nexus models that are liquid cooled. If it does affects someone else with a Nexus liquid cooled I will have to either do some more testing (with some volunteers sending me register dumps) or add a option in the config file to let the use pick how they want the RPMs displayed. For now we will see if this causes any other problems. 

Also, I am modifying the title of this thread to be Nexus and liquid cooled specific to draw some attention to others who may have a Nexus Liquid Cooled unit.

jgyates, 2018-02-01:
If anyone else has a Nexus Liquid Cooled generator, please test the latest and let me know if the frequency is reporting accurately when the generator is running.

Thanks
Jason

griffths, 2018-02-01:
Just a hazy datapoint on this topic but if I remember correctly, the alternators on some of the larger liquid cooled units are not direct drive so depending upon where the frequency value is calculated, it is possible that it might not correspond with expected results as they are driven through a gear reduction unit and the engine doesn't turn at the same RPM as the alternator.   

Jay

jgyates, 2018-02-16:
Hi @jeepjungle 

I think the fix I posted for this is correct, but to validate can you give me two data points?

1) Is your generator a Generac QuiteSource QT036
2) Can you send me the output of the "generator registers" command?

Thanks
Jason

jeepjungle, 2018-02-16:
Yes, QT036 Generac QuietSource is correct.   Side note, I just had the Nexus controller replaced a few weeks ago and now the serial number in the controller is all f's. 

OK : Auto, Off - Ready
>generator: registers
Registers:
Num Regs: 159 
Not Changed: 35278755 Changed 8626110 Total Changed 0.24
0038:01f4			0034:01f4
0037:01f4			0032:01f4
05f5:001e			05f4:0032
05f1:01f4			01f1:0000
05ed:0018			023b:0000
023e:0000			003b:0000
05fa:01fe			0059:01f4
0058:01f4			0053:01f4
0052:01f4			0057:01f4
0056:01f4			0055:01f4
0054:01f4			0208:03f4
005c:01f4			005a:01f4
005f:01f4			005d:01f4
003c:001e			000d:0000
000e:1108			000f:0210
000a:008c			000c:0001
0005:0c00			0006:0301
0000:0006			0001:00000000
0008:0000			0009:00fa
01f2:0000			0012:0000
0011:0090			0010:0512
0007:0000			0019:0000
002b:0000			002c:0000
002a:65d7			002f:01f4
002d:01f4			002e:01f4
0020:0000			0021:0000
001f:4774			001e:00f8
001d:4550			001c:0030
001b:4607			001a:0062
01f4:ffffffffffffffffffff

jgyates, 2018-02-16:
Thanks! 

Yes, the generator serial number reading all "F" is normal for a controller that is changed out. That happened to me also. the controller will require an activation code if it is not all "F" so all controllers that are replaced are sold with "F"s for the serial number. 

I made the mistake of updating my serial number on my new controller (since I can write to that register and it made me enter the activation code again (I was glad I had kept it). The generator will not transfer power unless it is activated or the serial number is all "F"

jeepjungle, 2018-02-16:
Seems like you can go to the Generac website, punch in the serial number from stamped plate and it will still give you an activation code.  I suspect it must write to some register once the correct code is entered.

JeanB51, 2018-02-19:
I ran the updated version today,
I have a Generac 36KW, Nexus controller, liquid-cooled, running on Propane.

Running the test (without load transfer), I have:

- 1800 RPM
- 60HZ
- 240V

I see this both in the responses from Genmon, and on the Nexus head itself

So evrything look to be in top shape.

Only one regret, I had to stop Genmon after it had run for 185 consecutive days.. this is one steady animal. I'll see how long I go this time.

Thanks again Jason.
enjoy some beer from my contribution on the way.

jgyates, 2018-02-19:
Thanks Jean! It is funny, I also look how long it has been running and feel sad if I have to restart, haha.

Everyone on this thread, thanks for helping test. I will close this issue now but feel free to comment if you have any other questions or issues.

closed 2018-02-19 by jgyates

moab11, 2020-04-29:
Hello, 
It looks like this change has affected my QTO3015ANSX, it is a 3600 RPM unit(the software picks this up properly) but the Frequency is shown as 120Hz. Is there somewhere to adjust this in the software?
[image]

jgyates, 2020-04-29:
Hi @moab11 

I would guess that his is something specific to your installation. The last updates were only to add alarm values (this is not related to the frequency reporting code). The frequency reported is a value read from the controller. On some controllers it is multiplied by 2 on others it is not.  It is possible  that you have a serial communication issue that is causing your controller to be mis-identified. You can check this on the monitor page under "Communication Stats". Is your controller being identified correctly on the Maintenance page? It is probably more likely that your controller is mis-identified due to a serial comm issue. 

Also, I you want to troubleshoot this please create a new thread. This allows me to better track the issues and allows others to search on thread more efficiently.

jgyates, 2020-11-18:
This issue is likely related to a difference in the firmware. If your frequency is still reported incorrectly see the resolution on this thread: https://github.com/jgyates/genmon/issues/473