jgyates/genmon #70: Configuration GUI

buschauer, 2018-02-11:
Did a small enhancement to allow updating email configuration via the web gui. Would be happy to share. Let me know if anybody is interested:

[image]

jgyates, 2018-02-11:
Looks wonderful. I can give you access and you can add it or you can post it in this thread as a zip attachment and I will check it in. Let me know what you prefer.

Thanks
Jason

buschauer, 2018-02-11:
Jason,

First and foremost, thanks for sharing the great Raspberry Pi App for 
Generac monitoring. I absolutely love it and it's super well documented.

To elaborate:

  * I enhanced the application a bit to allow me to update the email
    settings via the web app (see screenshot below). I've attached the
    source code. Feel free to use it and distribute it without credits
    or the like. It's all yours - you can do with it what you like
    (inclusive deleting it).
    The change also includes a small enhancement to allow me to use SSL
    for SMTP access and to use a different sender address than my email
    address to simplify my email filtering.

  * To avoid having to buy an expensive tool, I used the below cable and
    simply cut one side. Then I used the adapter listed below. So it
    took me all of 3 minutes to make my own cable and all I needed was a
    small screwdriver:
    https://www.amazon.com/gp/product/B000M802RG
    https://www.amazon.com/gp/product/B01C2V2E8U

Again - thanks for putting all of this together.

Michael Buschauer

jgyates, 2018-02-11:
Great info on the cable. I will add this to the docs.

I did not see the source code attachment. Once I get that I will merge test and repost.

jgyates, 2018-02-11:
One note, if you are replying to this thread by email, attachments are stripped off when using email. To attached a file you will need to post it to the thread at this link:

https://github.com/jgyates/genmon/issues/70

buschauer, 2018-02-11:
Here the attachment:

[Email Settings Enhancements.zip](https://github.com/jgyates/genmon/files/1713985/Email.Settings.Enhancements.zip)

jgyates, 2018-02-11:
Hi Michael,

Thanks for the work on the web interface. I think it is a great addition. I have merged the code, tested and uploaded it along with updates to the docs. I kept almost everything as it was presented. The only minor tweaks I made were:

1) Made the HTTP port default to 8000 instead of 80. Some folks use other apps on their pi so I do not want this to conflict with a full blown web server and this may cause problems with some folks who are upgrading a previous installation.
2) I made sender_address in mymail.conf an optional parameter and if sender_address is not present then use email_account as this value. this will probably work better to help folks upgrading from previous version.
3) I added a confirmation dialog when hitting save on the settings page, mostly to let people know that it is working and to remind them that genmon must be restarted for the changer to take effect.

Other than that I took the rest of your code as is. Great work and thanks again.

closed 2018-02-11 by jgyates

buschauer, 2018-02-11:
Jason,

  * All fine with your changes, looking good
  * Would be good to make the HTTP port a config parameter too. That way
    I don't have to edit the code after doing a "git pull"
  * Good point on the configuration dialog. I couldn't find a
    non-intrusive way to get the mail config reloaded (let me know if
    you have a simple idea). That may be a next item to fix if I have
    time. I also wanted to introduce an ability to configure which email
    address gets what type of notification (ie each email address can
    subscribe either to INFO, WARN and/or ERROR type of notifications
    separately. so basically a matrix) Then I was thinking of a monthly
    digest email that would be useful (notification type DIGEST).

Michael

jgyates, 2018-02-11:
Hi Michael,

Good point on the HTTP port. I just checked in a version that has a config option 'http_port'. Now genserv will read genmon.conf and look for an entry named 'http_port' for the default port. If this parameter is not there it still defaults to 8000. Now you can override it without modifying the code.

Jason

buschauer, 2018-02-11:
Hi Jason,

worked, thanks for the quick turnaround.

Michael