jgyates/genmon #71: Question. Upgrade

usmc124, 2018-02-16:
How do I upgrade from the current version to V1.4.1.  I have been replacing the updated files individually up till this point but assuming their is a way to upgrade the whole package.  Barely know enough about python to be dangerous.

jgyates, 2018-02-16:
I have received this question a few times so it prompted me to write a wiki page about this:

https://github.com/jgyates/genmon/wiki/Updating-to-the-Latest-Version

Let me know if you have any questions.

usmc124, 2018-02-16:
Perfect- exactly what I was looking for.  I found numerous methods to update the package but each followed with 25 entries of people disagreeing or offering alternative methods.  I figured I should go to the experts.  Thank you!

closed 2018-02-18 by jgyates