jgyates/genmon #73: MyEmail problem

UPS99, 2018-02-20:
I recently set up genmon and it's working very well through the web interface.   This is a great project. However, I have been unable to get email working.  I only need to have alerts sent, and did not try setting up IMAP.  I have tried configuring mymail.conf both by editing directly, and also by the web gui, however, starting mymail.py logs just these 2 lines in mymail.log and aborts:

"Loggin in 0 <mailservername fqdn>:587"
"Error in SendMailThread, retrying: character mapping must return integer, None or unicode"

The first part of the second line is in mymail.py, and it seems to be adding the Python error "character mapping.... " as a variable.
 
I have the outgoing email account and password entered, recipient account, the email server, and port 587.  The IMAP server is blank, and I have tried with SSL enabled and disabled.  The web gui includes a box "Sender Account", that doesn't seem to match anything in mymail.conf - it contains the word "undefined".  

This is running on a Raspberry Pie 3, with Python version 2.7.13,  I haven't tried updating Python yet. Genmon code 1.4.0 and later 1.4.1.

Has anyone seen this?

jgyates, 2018-02-20:
Based on this message in your log: ""Loggin in 0 :587" it looks like your SMTP server is not set in /etc/mymail.conf. The statement should print something like this: "Loggin in 0 smtp.gmail.com:587". This indicates that your SMTP server in /etc/mymail.conf is not set. You may want to validate that you are updating the mymail.conf file in the /etc directory, not the one in the directory that genmon.py is located.

If you are still having problems then can you post your mymail.conf file (removing / replacing your password and email name with something basic like "password". Also, does your web interface settings match your mymail.conf information?

jgyates, 2018-02-20:
One other question, how are you editing your mymail.conf file. What is odd is that you are printing out nothing for your SMTP server, yet if this was blank in the mymail.conf then the thread that sends mail (and prints out the message we are seeing) would not be launched. This makes me thing that you may have some odd non-ASCII characters in your mymail.conf file. If you are editing your mymail.conf file on a Windows system and copying it over this could introduce problems with CR/LF, depending on the editor your are using.

UPS99, 2018-02-20:
Hi,

 

I’ll update GitHub also – here’s the contents of /etc/mymail.conf  and below that the settings from the web gui – password lines skipped.  I saw your other email, and I had edited it in Windows using UltraEdit, a programmers file editor.  However just now I took a fresh copy of mymail.conf from the zip file, and edited it in nano – but the problem is still there.

 

I have the same user account in both places at the moment, but have tried using different accounts also.  I also tried a very plain password in case any characters were causing an issue.

jgyates, 2018-02-20:
If you want to include and attachment you must post it in the github forum. If you just reply to the email the attachment is stripped of by github. I did not get your mymail.conf file.

UPS99, 2018-02-20:
I only sent screen shots in the email reply.  Maybe the email also has html disabled?  I will update github in more detail.

 

 

Ed

jgyates, 2018-02-21:
I did some testing and found one problem in mymail.py that I just uploaded a fix for. Based on your output it looks like your SMTP server is not set ("Loggin in 0 :587" vs "Loggin in 0 SERVERNAME:587"). This will caused a crash in the previous version of mymail.py. The fix for this is now checked in. 

The problem you describe is that you are not using IMAP, but you do want to use SMTP, but your output in the above thread suggest that the program was reading an invalid or blank SMTP server. I suggest trying the new mymail.py and if you still have a problem then send the contents of your mymail.conf file (with your passwords and username redacted).

Let me know how it goes.

Thanks
Jason

UPS99, 2018-02-21:
Thanks, Jason.

 

 

I’ll try this in a few minutes.  I never did get time earlier today to get back to GitHub.  Too many work issues came up…

 

 

Ed

UPS99, 2018-02-21:
I just tried it – and received this error:

 

File “/home/pi/genmon/mymail.py, line 270

If  self.SMTPServer !=”” # if only sending is disabled, do not queue^

 

 

I will upload mymail.conf to GitHub.

 

 

Ed

UPS99, 2018-02-21:
[mymail.zip](https://github.com/jgyates/genmon/files/1742880/mymail.zip)

Here is the mymail.conf file,

jgyates, 2018-02-21:
my bad, I left out a colon in the new line I added. I just uploaded a new mymail.conf

jgyates, 2018-02-21:
I looked at your mymail.conf file. Nothing I jumping out at me as incorrect. If you still get an error with the new file then send your /var/log/mymail.conf file.

UPS99, 2018-02-21:
It is back to the original error -  character mapping...
[mymail.log.zip](https://github.com/jgyates/genmon/files/1742986/mymail.log.zip)

jgyates, 2018-02-21:
ok. I was able to duplicate by using the SMTP server in your conf file. After some searching I think I have a solution checked in. When I did my testing, the change I made now returns the error (535, 'Authentication failed') which is expected. Try this again with the latest version of mymail.py.

The issue is related to unicode handling in a python library and how some mail servers handle authenticating sending mail. Let me know how it goes in your testing.

UPS99, 2018-02-21:
Very good.  It works now, authenticates, and sends the email "Generator Monitor Starting ".

jgyates, 2018-02-21:
Great!  This was an issue I had never seen before so thanks for helping resolve this.

UPS99, 2018-02-21:
Unfortunately, there is still a small problem.  It is sending the "Generator Monitor Starting" email every 2 minutes.  In the mymail.log file - logged every 2 minutes is "Logging in: SMTP Server <servername>:Port <587>".

jgyates, 2018-02-21:
Are there any other messages in the log? If there is an error it retires every 2 min, but that should only happen if an error occurs. I will do some more testing, it is possible I have a logic error, but it is a bit difficult to find it since I don't have your exact situation here. Don't worry, we will find the issue.

UPS99, 2018-02-21:
That is all there is in the mymail.log file (above) - it just repeats exactly 2 minutes apart.  There is nothing repeating being logged in the genmon.log, genserv.log, messages, etc.  No errors logged at all, either.  The PIDs for genmon.py and genserv.py have not changed since the last reboot - so I don't think the message is coming from genmon actually restarting.

jgyates, 2018-02-21:
are you restarting genmon after you update mymail.py?

jgyates, 2018-02-21:
to restart you can call the startgenmon.sh file from the command line:

cd genmon
./startgenmon.sh stop
./startgenmon.sh start

UPS99, 2018-02-21:
Sometimes I have done that, sometimes I reboot to be sure nothing is left running or hung.  After the last update I rebooted.

 

 

Ed

jgyates, 2018-02-21:
How are you updating the repository?

Try updating this way:

git reset --hard origin/master

If you updated any files "git pull" will not overwrite them.

I just checked in one change to mymail.py, but I doubt it will change things, but it is worth a test.

This is odd. I am not seeing this behavior.

jgyates, 2018-02-21:
If you stop and restart from the command line using the ./startgenmon.sh, do you see any error messages? genserv.py will outut someting like this for any traffic on the web page: 

192.168.1.100- - [21/Feb/2018 13:02:14] "GET /cmd/getbase HTTP/1.1" 200 -

jgyates, 2018-02-21:
Just a FYI, under normal operation you should see two emails from genmon when it starts. One that states genmon is starting, and the 2nd one with the initial status of the generator.

UPS99, 2018-02-21:
This is interesting - when the script is run manually to stop and then start genmon, I get 2 emails – first Generator Monitor Starting at.., followed by Generator Notice: ...  and  the emails don’t repeat.

 

When the script is launched at boot via crontab – I get only the Generator Monitor Starting at.. email and it repeats every 2 minutes.  

 

I don’t think it is critical that it start at boot – since a reboot would normally be a rare occurrence.  

 

 

Ed

jgyates, 2018-02-21:
That is strange. Question: if you type "sudo crontab -e" do you see the commands to start startgenmon.sh? If you leave off the sudo, and just type "crontab -e" this would be the crontab for a the currently logged in user (I assume the pi user) user and should not have the startgenmon.sh command (e.g. each user has the ability to have a separate crontab running with the permissions of that user). If you setup the crontab as a non root user then this could potentially cause the problem.

jgyates, 2018-02-21:
I just tried loading the script via the pi user's crontab and it still works for me (since startgenmon.sh script contains sudo before each command). I guess it is something else. 

I know there are a few things that will not work in a crontab (relative paths and some environment variables). Have you modified the script?

This pages has some info and links to other threads that have problems with scripts behaving differently between crontab and command line execution.

https://www.unix.com/answers-to-frequently-asked-questions/13527-cron-crontab.html

jgyates, 2018-02-21:
Since crontab jobs are run in a different environment, depending on how you installed the python libraries, it could make a difference, but it does not make sense as to why no errors are logged. This page has a description on how to track down differences in the crontab environment vs the normal environment:

https://superuser.com/questions/519585/script-acts-differently-whe-run-by-cron

UPS99, 2018-02-21:
I didn’t modify the startgenmon.sh script.  However, I have had a problem more than once in the past on other server where something launched in crontab, or a drive mount launched in fstab, fails because something else is not ready yet, such as network access.

 

I can try adding a delay – I think this is right:

 

@reboot /bin/sleep 60 ; /home/pi/genmon/startgenmon.sh start

 

https://www.unix.com/answers-to-frequently-asked-questions/13527-cron-crontab.html

jgyates, 2018-02-21:
I have a delay in mine:

@reboot sleep 30 && /home/pi/genmon/startgenmon.sh start

I found this sleep was needed as it caused some with the serial subsystem not be ready on the pi.

UPS99, 2018-02-21:
I tried it with 60 seconds – probably doesn’t need that long – and it works fine – rebooted - both emails sent, no repeating messages.

 

 

Ed

closed 2018-02-22 by jgyates