jgyates/genmon #74: Power PI from Generator Controller

jonathanpisarczyk, 2018-02-21:
Hello Everyone,

I was curious if anyone has had success powering the PI using the generator controller connector as stated in the instructions? 

"alternate way to power your Pi is to take 5V and ground from the generator controller connector and power your Pi via the power and ground connections on the GPIO connector. The wiki page on making a cable has additional detials on the controller pinout."

If so can you please provide a wiring diagram? I'm not 100% sure how to connect molex power and ground to which GPIO pins on PI?  I'm using the RS232 9pin female with GPIO pins 04, 06, 08, 10. 

I'm guessing I could connect molex PIN1 (5V) to GPIO 02 (5V) 
and
molex PIN2 (ground) to GPIO 39 (Ground) 

If not I plan on using the usb style connector. 

Any comments apperciated.  What a great project and idea.  I love it.  Makes so much more sense than the 3g option.  I've got no cell signal where our generator is located.

krisasmith, 2018-02-21:
I am powering my Pi though the GPIO. I am using PIN 2 & 14.

[image]

Black wire is ground and green wire is 5v both coming from the genset controller.
[image]

jgyates, 2018-02-21:
Hi @jonathanpisarczyk 

This thread:

https://github.com/jgyates/genmon/issues/3

has some discussion on using 5V and GND from the controller connector. I personally have not done this but others who have used this forum have.

jonathanpisarczyk, 2018-02-21:
Thanks @jgyates I missed thread #3... 
Also thanks @krisasmith 
I suspect you meant to say **Black wire is 5v and green wire is ground** both coming from the genset controller.  

I will try it out!

krisasmith, 2018-02-21:
Yes - I had them backwards. I have updated the post.

jonathanpisarczyk, 2018-02-21:
Picture of my new setup for anyone else starting out. 
[image]
[image]

krisasmith, 2018-02-21:
That should do it, looks just like mine.

closed 2018-02-22 by jgyates

UPS99, 2018-02-28:
I had tried this with an Evolution controller and found that everything worked, however the load of the Pi pulled the 5 volt line down to about 4.7 volts, which may put the Pi close to the edge of its operating range, and also could overheat a regulator or other component in the controller.

jgyates, 2018-02-28:
That is a good data point. Thanks!