jgyates/genmon #75: Incorrect Engine State: Cooling Down

ColtB45, 2018-02-26:
I've got a 7030 with a Evolution controller that incorrectly reports a state of Cooling Down while running in an outage. It's the only one I have seen do that.

It does briefly go to Running, but then (during a 3 hour outage) stays on cooling down for the duration. Once utility power is restored it goes back to running, and then Stopped.

```
Registers:
Num Regs: 209
Not Changed: 158309 Changed 5085 Total Changed 0.03
0038:0000                       0034:0000
0037:10a5                       0032:4000
05f5:0000                       05f4:0000
05f1:0000                       01f1:0000
05ed:0064                       023b:0000
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:0000
0052:0000                       0057:0000
0056:0000                       0055:0000
0054:0000                       0208:03f2
005c:0000                       005a:0000
005f:0000                       005d:0000
003c:0000                       000d:0000
000e:0a15                       000f:021a
000a:0085                       000c:0005
0005:0a0a                       0006:0000
0000:0009                       0001:00000000
0008:0000                       0009:00f5
01f2:0000                       0012:0000
0011:009c                       0010:0112
0007:0000                       0019:0001
002b:0005                       002c:0a0a
002a:6a77                       002f:0000
002d:0002                       002e:0001
0020:0000                       0021:0000
001f:4a49                       001e:018a
001d:0000                       001c:0000
001b:476f                       001a:00c2
01f4:33303032323537353634
```

jgyates, 2018-02-26:
Thanks for the emails, it looks like you are getting duplicate emails for "Generator Notice" does this always occur? You should only getting "Generator Notice" emails if register 0001 changes state. It looks like you are not getting duplicate emails of Power Outage Notification. Is this generator using 2 Wire start (i.e. some other device is signaling the generator to start)?

ColtB45, 2018-02-26:
Hey @jgyates ,

Thanks for the quick reply.

**Comm Errors:** I'm not noticing any. This is from the monitor tab:
```
Monitor:

Genmon Stats: 
    Monitor Health: OK
    Generator Type Selected : Evolution Controller, Air Cooled
    Generator Monitor running for 2:32:30.

Serial Stats: 
    PacketCount M: 279386, S: 279385, Buffer Count: 0
    CRC Errors : 0  Percent : 0.00
    Discarded  : 0  Restarts: 0  TimeOut: 0
    Packets per second: 61.07
    Average Transaction Time: 0.0316 sec
```

**Validate the outage:** I can confirm the entire duration, except for a brief period in the beginning and again at the end, the generator was running and genmon was reporting cooling down.

**Transfer Switch:** It has a dumb transfer switch.

**Emails:** You got 'em. :)

**Outage Log:** The Outage log is destroyed. :( I run on a non-persistant install that is wiped at reboot (overlayroot). I had rebooted when updating genmon. I can definitely save it in the future.

**Testing:** I can probably do that later this week. This is a fairly remote EMS repeater and it takes a fair bit of effort to get there.

ColtB45, 2018-02-26:
@jgyates I always get duplicate Generator Notice & Outage Notice emails but not duplicated restored emails. It's worth noting this doesn't happen on the other installs and this is a brand new (< 2 month old) genset. Possibly newer firmware causing issues?

jgyates, 2018-02-26:
Sounds good. One other thing we can check is the logs in /var/log. If you have them. I know you said that your system is overlayed, but just thought I would ask. If you have them we would want to see genmon.log, myserial.log (probably empty since you report no comm errors) and mymail.log (odd that you are getting duplicate emails).

It sounds like you are using this on multiple generators, which is great if you are since we can narrow down the differences easily. If you are are all of the generators the same model or are they different models?

Another thing we can do to test is have the software monitor register 0001. genmon.py has the ability to monitor retisters and send an email when they change:

On line 294:

        self.PrimeRegisters = {
                    "0001" : [4, 0],     # Alarm and status register

Change the last zero to a one like this:

        self.PrimeRegisters = {
                    "0001" : [4, 1],     # Alarm and status register

Once you make this change and restart it will send an email every time register 0001 changes.

ColtB45, 2018-02-26:
I will grab those logs next time. I suspect I can do that by initiating a transfer in-place of a real outage? I wont be able to confirm on the controller but I would have the logs.

We have genmon at 3 tower sites (about to be 4). Anywhere the generator is compatible. All are evolution controllers but different models and ages.

I'll try monitoring register 0001 and let you know what happens.

jgyates, 2018-02-26:
Yes, transfer in place should do everything a power outage does except send power outage notification emails.

thanks for the feedback and testing. This really helps compatibility. No rush, just let me know when you have some more data points.

-Jason

ColtB45, 2018-02-26:
First, Let me share some good news/egg on my face. I use a mail forwarder and it seems to be responsible for the duplicate emails. Not sure why it duplicates some and not others, but I removed the other 5 addressees and started testing with a different address and the duplicates stopped.

**genmon.log** is empty less some entries from before I activated the overlay.
**myserial.log** is also empty.
**mymail.log** only has entries about failing to get the incoming folder.

Here are the emails from a manually initiated transfer.

Another fun aside, you're not able to use the web interface to stop the generator during a transfer when the running status is cooling down.

I got it shutdown and transferred back but don't know if that's worth reconsideration if someone needs to stop it _right now_.

```
-----Original Message-----
From: russellgen 
Sent: Monday, February 26, 2018 12:09 PM
To: kg5chm
Subject: Generator Monitor Starting at Russell_Mtn

Time: 12:09:10 PM
Date: Mon 26-Feb-2018
Generator Monitor Starting at Russell_Mtn
```
```
-----Original Message-----
From: russellgen 
Sent: Monday, February 26, 2018 12:09 PM
To: kg5chm
Subject: Generator Notice: Russell_Mtn

Time: 12:09:18 PM
Date: Mon 26-Feb-2018

Notice from Generator: 

Switch State: Auto
Engine State: Off - Ready
Active Relays: 

No Alarms: 0001:00000000

Last Log Entries:
    Start Stop Log: 02/26/18 09:08:23 Stopped - Auto 
    Service Log:    
    Alarm Log:      01/17/13 10:57:01 Low Battery : Alarm Code: 0000

Last Alarm Code: Error Code 0000: No alarms occured since controller has been power cycled.
```
```
-----Original Message-----
From: russellgen 
Sent: Monday, February 26, 2018 12:10 PM
To: kg5chm
Subject: Monitor Register Alert: 0001

Time: 12:09:53 PM
Date: Mon 26-Feb-2018
0001 changed from 00000000 to 00020000
Num Regs: 209 
Not Changed: 1071 Changed 17 Total Changed 0.02
0038:0000			0034:0000
0037:10a5			0032:4000
05f5:0000			05f4:0000
05f1:0000			01f1:0000
05ed:0064			023b:0000
023e:0000			003b:0000
05fa:0000			0059:0000
0058:0000			0053:0000
0052:0000			0057:0000
0056:0000			0055:0000
[... 502 lines cut ...]
```
-----Original Message-----
From: russellgen 
Sent: Monday, February 26, 2018 12:17 PM
To: kg5chm
Subject: Generator Notice: Russell_Mtn

Time: 12:17:18 PM
Date: Mon 26-Feb-2018

Notice from Generator: 

Switch State: Auto
Engine State: Off - Ready
Active Relays: 

No Alarms: 0001:00000000

Last Log Entries:
    Start Stop Log: 02/26/18 12:17:47 Stopped - Auto 
    Service Log:    
    Alarm Log:      01/17/13 10:57:01 Low Battery : Alarm Code: 0000

Last Alarm Code: Error Code 0000: No alarms occured since controller has been power cycled.
```

jgyates, 2018-02-26:
Glad to hear the email issue is resolved. 

It looks like the engine status has the following sequence:

00000000 - Off Ready
00020000 - Cranking - 12:09:53 PM
00030000 - Running - 12:10:01 PM
00050000 - Cooling Down - 12:10:08 PM
00030000 - Running - 12:15:49 PM
00090000 - Stopping - 12:17:05 PM
00000000 - Off Read - 12:17:12 PM

Firmware:  V1.18
Hardware : V1.06

The next steps as I see it are this:

- Check the display on the controller and see if it matches the above sequence. It is possible that generac has changed the display / status bits with new firmware revisions.
- Compare the 7030 Firmware and Hardware revisions and see if there are differences with the other system installed (register 002a).
- Check the 7030 to see if there is additional hardware connected to the controller (two wire, etc)
- If these do ideas do not yield anything useful we can start comparing register dumps between the generators (i.e. there may be another register that will uniquely identify the differences between them)

Presently, the code that interprets the engine status is the same for all models. The code also has exceptions for four types of controllers (Evolution Air Cooled, Evolution Liquid Cooled, Nexus Air Cooled, Nexus Liquid Cooled). If there are more variants or the code should take into account specific firmware versions we can do that.

ColtB45, 2018-02-26:
Is there a way to retrieve the common model number from the registers? Example: G00**7030**0? I see the serial number with `generator: getregvalue=01f4` but that's less helpful.

jgyates, 2018-02-26:
Not that I have found. As far as identity, reg "0000" will tell if it is Nexus or Evolution and Air Cooled vs Liquid Cooled. I have tried looking at the serial number format, but I have not been able to see any patterns. According to the formus here: http://zillerstore.com/forums/ the serial number is not formatted in a way that will reveal the model number. Generac has a page that will look up the model from the serial number but I have not been able to figure that out from registers.

ColtB45, 2018-02-27:
I have other ones with HW v1.06 working. I have FW up to v1.16 working as well. I suspect a change at this point in FW v1.17, 1.18, or 1.19.

G00**6551**0:
HW v1.05
FW v1.10

Works as expected.
```
Registers:
Num Regs: 209
Not Changed: 62111456 Changed 1048490 Total Changed 0.02
0038:0018                       0034:0009
0037:2a94                       0032:4000
05f5:0000                       05f4:0000
05f1:0000                       01f1:1a10
05ed:0064                       023b:0000
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:0000
0052:0000                       0057:0000
0056:0000                       0055:0000
0054:0000                       0208:03f6
005c:0000                       005a:0000
005f:0000                       005d:0000
003c:0000                       000d:0000
000e:0e2c                       000f:021a
000a:0086                       000c:0001
0005:0a00                       0006:0001
0000:0009                       0001:00000000
0008:0000                       0009:00e7
01f2:030b                       0012:0000
0011:009c                       0010:0112
0007:0000                       0019:0011
002b:0005                       002c:0a00
002a:696e                       002f:0000
002d:0002                       002e:0001
0020:0000                       0021:0000
001f:48ce                       001e:018e
001d:0000                       001c:0000
001b:45f4                       001a:00c6
01f4:xxxxxxxxxxxxxxxxxxxx
```

G00**7042**0:
HW v1.06
FW v1.16

Works as expected.
```
Registers:
Num Regs: 209
Not Changed: 27764 Changed 704 Total Changed 0.03
0038:001b                       0034:000a
0037:2db4                       0032:4000
05f5:0000                       05f4:0000
05f1:0000                       01f1:050a
05ed:0064                       023b:0000
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:0000
0052:0000                       0057:0000
0056:0000                       0055:0000
0054:0000                       0208:03f1
005c:0000                       005a:0000
005f:0000                       005d:0000
003c:0000                       000d:0000
000e:0021                       000f:0102
000a:008b                       000c:0004
0005:0a00                       0006:0301
0000:0009                       0001:00000000
0008:0000                       0009:00f0
01f2:0831                       0012:0000
0011:009c                       0010:0212
0007:0000                       0019:0011
002b:0005                       002c:0a00
002a:6a74                       002f:0000
002d:0001                       002e:0003
0020:0000                       0021:0000
001f:4867                       001e:018b
001d:0000                       001c:0000
001b:458d                       001a:00c3
01f4:xxxxxxxxxxxxxxxxxxxx
```

jgyates, 2018-02-27:
I wrote a simple python program to parse some data from register dumps people have submitted. I reached out to one user who has FW V1.18 to see if he can weigh in. From the small data set I have (about 18 register dumps) you are the only person how has a Evo Air Cooled with V1.19

jgyates, 2018-03-23:
@ColtB45 , Just checking in, any additional data on this issue.