jgyates/genmon #76: added a few enhacements

buschauer, 2018-03-04:
Hi Jason

Did a few enhancements again. Feel free to include some or all of them (as you like). 

The changes are:
  - Included all settings (both from mymail.conf & genmon.conf) to the Settings page
  - Added a Notifications page that allows for different email addresses  to receive different
    types of notification (eg. I can send a text message via 123@tmomail.net in case of 
    an outage, but I send all other messages to my normal email account)
  - "cleaned up" the header and footer, to make sure they don't scroll off-the screen.
  - improved startgenmon.sh, to allow for parallel installations and easy restart

For details refer to README.Michael in the attached zip-file.

Michael

[image]
[image]
[image]
[genmon.zip](https://github.com/jgyates/genmon/files/1778243/genmon.zip)

jgyates, 2018-03-04:
You sir, rock! I will merge these tomorrow. If you have any other ideas for enhancements let me know. I can add you as a contributor to the repository if you like.

proppilot, 2018-03-04:
Wow this is becoming a commercial quality product !  With the shutdown of 2G in Canada rendering the Generac solution useless, this has replaced it beautifully and I will never go back.

jgyates, 2018-03-04:
OK. I merged the changes, with a few minor tweaks:

* Made msgtype in sendemail() an optional parameter (I use this module for another project so I don't have to changes the other stuff just yet)
* changed the description of useunknown genmon.conf option to be more descriptive in the UI and moved it to the general section. This option enabled the display of unknown sensors in the output. It does not effect the console output
* changed the spelling of "consol" to "console"
* removed the following settings from the UI 
     - modbus slave address (this will break comms, only for development and debugging)
     - uselegacysetexercise, liquidcooled, evolutioncontroller - If these are enabled they will override the auto-detect settings, since the UI does not have an option to disable this would lead to someone disabling auto-detect if they enabled any of these so it is probably best to remove the the UI and leave them in the documentation for debugging purposes.

In summary, I kept all functionality, just tweaked a few things. 

@buschauer, great additions. Many thanks!

jgyates, 2018-03-04:
also, if you have any ideas for how to make the registers page look better I am all for it. As you can probably tell html/javascrip/css is not my strong suit. Presently the register page (internal.html) shows the base registers, when the registers change the register color changes to red for 5 seconds, then to yellow for 10 seconds, then to brown. This allows easier debugging of what registers change, which allows figuring out which registers change and for what reason. Presently there are a few options that are supported on liquid cooled evolution that are not supported on other platforms (battery charging status for example).

buschauer, 2018-03-04:
Let me think about how to show the registers. Maybe add another menu to the left and put it into a table where the background color changes?

Also there is  single pixel above the navMenu and another single pixel below the right hand side that is off. But couldn't figure out yesterday why :) Will fix it when I have time.

2 other idea: 
- how do you feel about adding an "upgrade" section to the menu, so that you can upgrade to the latest version from the web gui?
- Do you have any good idea on how to trigger the refresh of all the config values. Would love to avoid the manual restart, as it's not "commercial quality" (as @proppilot put it...)

buschauer, 2018-03-04:
Jason, Forgot to say "Thank you" for integrating and merging it. I checked it out and it looks good! Thanks for all your work!

jgyates, 2018-03-04:
Regarding the single pixel, you got me. I have no clue.

I like the upgrade option. This would obviously require a restart.

As far trigging the refresh, off the top of my head, I could make genmon.py restart genserv.py and genserv.py restart genmon.py. This would effectively reload everything. We sort of have a chicken and egg situation as some settings are core to the web interface functionality. 

A potential solution would be to have a third python app that would restart both genmon and genserv.

Let me think on the restart / reload problem....I will get back to you on this.

jgyates, 2018-03-04:
on possibly solution would be to add a command to genserv.py that genmon.js could call. This command would spawn a thread (so we can return to javascript execution),the thread will wait x number of seconds and genserv.py will call startgenmon.sh to restart both processes. The javascript will display to the user that it should wait x seconds and then force a reload on the web page.

Also, I started putting a version string in genmon.py and returning that with the "status" command. If we do add an option to upgrade then we will need to decide if we upgrade from a tagged release or just refresh the repository (or give an option for both). I think the place for this would be in the "monitor" page as that deals with genmon.py

buschauer, 2018-03-04:
that would work. But for the config parameters, I was more thinking on whether it's possible to refresh the cashed variables by re-reading the conf files without restarting. Would be a bit cleaner and less risky (in case the config files are messed up, the application could fail to start. However at the next re-start, the problem would still occur).

jgyates, 2018-03-04:
have just a config reload is doable, but it depends on the variable we are refreshing. Some things would require stopping and starting some threads (which is doable, but other threads may start acting up if that happens) , others would require only updating a variable. Let me try to enumerate what would be needed for each parameter and what the thread dependencies are...

jgyates, 2018-03-04:
I am going to do an experiment to see how many genmon config params I can get to reload. I will look at genserv later as this one is a bit tricky with Flask on one end and the the javascrip on the other.

I should be able to get several easily by updating some variables. Others will require some threads to be restarted.

jgyates, 2018-03-05:
Design trade off time: I have modified genmon.py to add a "reload" option. Currently everything is relaoded except the Server Port parameter. I am leaning toward taking that out of the web UI as it will cut off communications with the web server if it is changed so this parameter does not makes sense to me to change in the UI.

The issue I have is that when I made the changes to restart the serial port, I had to change from a blocking serial read to a non blocking so the thread could exit when signaled. I added in the smallest sleep I could, but the end result is that I can restart the serial port thread, but the change reduces throughput from around 64 packets per second to around 48 packets per second.

We have 2 options:

1) Make changing the serial port an option only changeable in gemon.conf
or
2) All the serial port to be changed in the Web UI and take the performance hit.

 I am leaning toward option 1, but I am open to other suggestions.

thoughts?

jgyates, 2018-03-05:
I guess a third option would be to leave the option to change the serial port in the Web UI and if the user changes it let them know they must restart genmon. Also, if we are going to implement the update software option we will have to restart everything anyway.

So now I am leaning toward have this option in the UI, but have the UI know which options require a complete restart vs a reload of settings.

jgyates, 2018-03-05:
OK. I found a workaround for this to keep the throughput up and still restart the serial. I still have concerns about changing the serial port. I need to test to determine what happens if the port is invalid. Ideally with a GUI you don't want to give the user the ability to break things.

buschauer, 2018-03-06:
Thanks. I like option1. I don't think the ports need to be dynamically configurable and removing them from the settings screen seems fine.
Logically, the generator type, the cooling type, the full type and the ports would belong into a "Setup Wizard" section anyhow. The don't seem to be logically at home on the normal settings page.

jgyates, 2018-03-06:
Ok. I check in some updates to genmon.py, genserv.py, mymail.py, mylog.py and added the file mythread.py.

These updates add the following:

a "reload" command that will reload the settings for genmon.py, if called via the web app it will also reload genserv.py settings.
The rest of the updates are in support of reloading after the "reload" command.
I tweaked a few of the Web UI descriptions to make the more accurate for their function (some of the descriptions on the conf file were not exactly clear (fixed that also)

The new "reload" command is there, but genmon.js does not call it yet.
For now I took out the Server Port (the socket thread is the only thread that does not get restarted on reload), but left in the serial port option. I think I got that fixed were it will not cause problems. We can take it out later if we need to and after more testing.

jgyates, 2018-03-06:
Also, genmon.py reloads the config in the python script and stops / restarts the needed threads to reload. genserv.py reloads automatically via a option built into Flask. If the file is updated, Flask will reload it. In genserv.py I added a function named Reload() that makes a shell call to "touch" the file genserv.py, which reloads genserv.py. Not sure how this will play with the javascript yet, it does not appear to harm it just watching it run. We may need to issue a page refresh for some things to fully take effect (like HTTPS changes).

jgyates, 2018-03-06:
FYI, I added a script "updategenmon.sh" to the repo. It does a hard reset to the master so it will overwrite and changes in your genmon directory if you decide to use it. I thought it might come in hand if we do the update from the web UI path.

jgyates, 2018-03-06:
ok, one more comment, :) . 
I added a "is file present check" for the cert and key files. I don't have a cert or key file handy so I did not test that. If you do, let me know if there are any issues with this code, it was pretty simple so I don't think there are any issues.

buschauer, 2018-03-07:
did a few small changes again:
 - cleaned up my missing pixels 
 - made sure header and footer are properly center & middle aligned
 - added an icon on the top-right to point to the register page.
 - integrated the registers page into the main page and did some 
   formatting. The color is now gradually fading away over 60 secs (configurable)
 - added some icons to the main menu and stream-lined the code behind the scene
   that executes the actions

Let me know what you think.

BTW I didn't adjust the "Maintenance" page yet, which is not consistently left aligned (the bottom part has less margin. Also the hidden section is causing other pages to scroll while hidden). I took a quick look, and the change looks quite involved.  Do you want me to do it? 

[genmon.zip](https://github.com/jgyates/genmon/files/1787558/genmon.zip)

buschauer, 2018-03-07:
As for Reload(), are you sure you checked in the changes, as I can't seem to find them.

jgyates, 2018-03-07:
Reload is line 385 in genserv.py, the code to call the "reload" command in genmon.py is located in the genserv.py ProcessCommand() function on line 83. Line 105 in genserv.py is where it actually happens. The java script should just issue the getJSON command with "reload" as the command.

I will look over these changes and merge....Thanks for the update!

buschauer, 2018-03-07:
hmm.. you're right. I see them on the website. there is something off with my local git... let me check.

jgyates, 2018-03-07:
Wow, this looks outstanding. I really like the new look. 

Regarding the maintenance page, that is your call, if you want to go for it, I am fine with the way it looks so dont feel you have to...

I also really like the new register page. I assume if I wanted to change the descriptions of the register page I just modify the BaseRegistersDescription table? I will probably tweak the descriptions a bit to make them more understandable. One minor request, if possible could you make to so it fades to a color other than the default color. One of the things I am trying to accomplish with this page is to be able to leave the page open for a long period of time and be able to quickly see what changes have occured just by looking at the color. This helps me find registers that are meaningful. I have done this in the past my instrumenting genmon.py to have it send emails when specific registers changed but that is a bit messy as you have to deal with may emails or long emails to get the data you need.

Again, this is awesome stuff. I will be checking this stuff in soon...

buschauer, 2018-03-07:
sure - no problem.  Find attached the updated genmon.js. This will still fade as before, but leave the font in RED for perpetuity once something changed. And yes, you are right, you can change BaseRegistersDescription as required.

[genmon.zip](https://github.com/jgyates/genmon/files/1787760/genmon.zip)

jgyates, 2018-03-07:
I updated some of the labels for the registers and checked everything in.

jgyates, 2018-03-08:
I am guessing one of two things may fix this:

1) we need to have the javascript delay for 5 seconds (this is about how long a reload takes) after the reload, then issue a window.location.reload() to reestablish the javascript. I would imagine that the registers would go stale or at least get partial set of registers back on a reload, until the serial system was up and running. Hopefully the addition of window.location.reload()  will fix that.

or 
2) Have the javascript reset the list of registers it is tracking.

I tried this out and it is working for me. Once I hit OK to save settings it looks like everything restarts and the web interface response after about a seconds or 2. My register page looks like it keeps going. We should do more testing to see if the issue you are seeing is a result of the socket in ClientInterface.py getting dropped (it should not since I leave that thread in place on reload) or an issue with the javascript needing to be reloaded, or just the register page gets confused because it is receiving a partial list of registers. I am guessing the later (register list).

I know that if you get the registers too soon after starting or reloading you could get a partial list of registers because genmon.py has not finished a complete cycle thru the register list. If it turns out a reload only affect the register page we can just have genmon.js reset the list of registers on a reload.

I took your changes and made one minor mod in genmon.js. This makes the "Display Unknown Sensors" setting not dependant on "Output to Console". "Display Unknown Sensors" shows a few more sensors on the Status page.

[genmon.js.zip](https://github.com/jgyates/genmon/files/1791742/genmon.js.zip)

It looks like we implementing the same goal with two approaches. You are calling Reload() and the "reset" command after SaveSettings(). I was thinking it would just be another JSON call after the save settings JSON call. Don't know if it will make a difference in the end result (probably not). If we do it your way we can remove the "reload" at line 85 in genserv.py and the 

     if command == "reload":
          Reload()                # reload Flask App

at line 105 in genserv.py.

Probably does not hurt anything to have both of these approaches in there.

FYI, I am not checking anything in just yet until we nail this down a little better.

jgyates, 2018-03-08:
I timed how long it takes from clicking "Are you sure" to "Monitor Health: OK", total time ~ 13.5 seconds

buschauer, 2018-03-08:
that seems to be the problem. mine is not recovering. Even after 2 mins, I still see "Monitor Health :  Threads are dead. Not receiving data. " I checked the javascript side and the ajax calls recover correctly from the restart. How can I figure out why the threats don't restart?

jgyates, 2018-03-08:
OK. look at two things:

1) If the status is "Threads are dead", from ClientInterface.py you can issue the "generator: getdebug" command and it will tell you what thread is dead.
2) look at /var/log/genmon.log, /var/log/mymail.log, and /var/log/myserial.log to see if there are any error messages.

Also, what change did you make in the settings page to get it to crash a thread?

jgyates, 2018-03-08:
One other thing, for now, I put an email notification ("info") when genmon mod reloads successfully. I may remove this later but for now it is good indicator if reload is working or not

buschauer, 2018-03-08:
I get the following output:

ClientInterface.py
WARNING:  Threads are dead. Not receiving data. : Auto, Off - Ready
>generator: getdebug
SendMailThread

ohh... could this be related to the fact that I have IMAP disabled (not set). I see an error message genmon.log: 

Error in Reload: MyMail instance has no attribute 'threadReceiveEmail'

as for the change I did in Settings. I didn't change anything. I just press save.

And one more things. I realize that once I enable "Display Unknown Sensors", I get on the status page the error message about unknown values. Seems genmon.py line 1834 is too generic.

jgyates, 2018-03-08:
I think I can duplicate this....will fix shortly...

jgyates, 2018-03-08:
Try this..
[mymail.py.zip](https://github.com/jgyates/genmon/files/1791823/mymail.py.zip)

buschauer, 2018-03-08:
works beautifully! thanks!

jgyates, 2018-03-08:
I checked in the version of mymail.py I sent you. Also, I checked in a new version of genmon.py that has one additional check that may cause a crash on reload if you don't have it.

I have not checked in the version of genserv.py and genmon.js that we are working with now. 

Do you think we need a delay or message to the user before we go live with the reload on settings change?

jgyates, 2018-03-08:
I added this message:

"Save settings? Generator Monitor will reload your settings. This will take a few seconds. Are you sure?"

Apparently the built in javascript alert type dialogs will not count down or allow them to be removed. There is likely a better way to do this, but I am not aware of anything just yet. I am going to check this in and continue to test. I will not tag a release until I have more testing in...

Thanks for all of your help. Let me know if you want to tackle something else.

buschauer, 2018-03-09:
see attached. fixed the message (I had a better idea as you suggested). See attached. We can use the same when we do the "Upgrade" from the web-site.

PS: the rest seems to work fine now

Michael

[genmon.zip](https://github.com/jgyates/genmon/files/1795350/genmon.zip)

buschauer, 2018-03-09:
And one more things. I realize that once I enable "Display Unknown Sensors", I get on the status page the error message about unknown values. Seems genmon.py line 1834 is too generic.

jgyates, 2018-03-09:
I just got in, I will look at the update in a few min.

I also saw the Unknown Sensor issue last night. I checked in an update (just changed the label to "Unsupported Registers:" last night for this issue.

jgyates, 2018-03-09:
Wow, very slick. I will check this in now.

jgyates, 2018-03-09:
One very minor issue: if you change the "Site Name" in settings you have to hit refresh for it to show up at the top of the web page. This is due to the fact that genmon.js only calls GetHeaderValues() on $(document).ready(), it should probably be called again after or in saveSettings..

jgyates, 2018-03-09:
I updated the main page READ.ME and Wiki to include a picture of the new web interface near the top.

jgyates, 2018-03-09:
FYI, I found a couple of more issues with the genserv.py:

If the config parameter is passed into the Flask library via app.run() (line 499, HTTPPort and SSLContext) then the settings are not updated on reload. This has something to do with the way Flask is implementing the auto-reload function. This affects most of the secure web settings (except username and password) and also affects Web Server Port.

I am updating the name of this thread to make it specific to the web UI so others who may be using this know this is a work in progress.

jgyates, 2018-03-09:
I am disabling the genserv.py specific settings until I can figure this one out....

buschauer, 2018-03-10:
Fixed the  "Site Name" issue and also fixed the display for register 01f4 (serial number) which was very wide and rendered badly on a small screen

[genmon.zip](https://github.com/jgyates/genmon/files/1798996/genmon.zip)

jgyates, 2018-03-10:
Thanks! I am trying to determine the issue with reload. In doing this I have started looking into the restart option, which also has a problem. I sort of have it sort of working, but it is ugly. It restarts both genmon and genserv, but genserv fails on reload due to Flask not completely terminating, so it retries and succeeds the second time.  As a result you get two emails notifying that genmon is starting. Flask is an odd app in the way that it handles it's own process.....

jgyates, 2018-03-10:
Tony, I am going to remove the "sh" from the instructions on this page regarding setting up crontab.

https://github.com/jgyates/genmon/wiki/Setting-up-your-Pi-and-genmon

I can't remember how the "sh" got the, but it was never needed if you used the files in the repository.

jgyates, 2018-03-10:
Michael,
I checked in a version of genserv.py that restarts both genserv.py and genmon.py on a settings change. This appears to work, however I was seeing a situation where the first restart would fail, and the software would detect that and it would restart it again and it would work on the seconds try (resulting in two emails from genmon that it started). For some reason this behavior stopped and it is working as expected. Not really sure why. 

The short story is that it is working, however we "may" see to emails stating that genmon has started.

I am still looking into this but I thought I would get some additional data points (i.e. see if anyone else is having an issue. It may be a timing issue having to do with how long it takes genserv.py to be killed.

proppilot, 2018-03-10:
now that I did the upgrade and see the fantastic functionality that has been added, I have a question about web security.  How do they work ?  If I click on 'use secure web settings' do I just need to add a username and password and I will get a login prompt on the home page or is a certificate needed ? 

I did just check that one box, added a un and pw and nothing happened.  When I looked back at the settings (after saving successfully) the fields were empty again.

jgyates, 2018-03-10:
Hi Tony,

After you make changes to the settings page you have to click the "Save" button at the bottom.

Regarding the web interface settings: if you make any changes to the "Webserver Security Settings:" section, you will need to refresh your browser page. 

If you make changes to "Port of WebServer" your address of the web server will change. For example if you access the web site from this address: http://192.168.1.15:8000/, changing the Port of Web Server will change the 8000 to the value you specify in the settings.

Let me know if you have any other questions.

proppilot, 2018-03-10:
[image]

These are the settings I am changing (username is a real one and so is password)

I get the email that the utility restarted.  I clear the cache in Safari and when I open the web page there is no login and when I look in the settings, they are unchecked and the u and p fields are empty

jgyates, 2018-03-10:
You will need to have "Use self signed certificate" checked unless you provide a key file or a cert file. What is happening is that you are using settings that do not go together so genserv is reverting to the default. If you select "use secure web settings", "Use self signed cert", "web username", and "web password", then you should be able to access "https://youipaddress" .  Note the https, not http. Also, if you use "User Sedure Web Settings" the software will ignore your "Port of Web Server" since secure web (HTTPS) always uses port 443.

The web interface does not currently check the validity of your web secure settings (or any setting for that matter).

proppilot, 2018-03-11:
hmm... I click on the self signed and use secure web and then create a username and password and save at the bottom.  The restart happens and when I can refresh the page and look at the settings, they are defaulted again.

jgyates, 2018-03-11:
are you changing your URL in your address bar after you make the settings. You need to have "https://ipaddress" with no port at the end. What is the exact address you are using to access the web server. I have tried this and I am unable to duplicate what you describe so something else is different. 

/var/log/genserv.log will also show restart and port number info. I just tested this and set it back to the default and these are my entries:

2018-03-10 20:18:44,223 : Restarting: /home/pi/genmon/startgenmon.sh restart  <- save settings
2018-03-10 20:18:47,917 : Starting /home/pi/genmon/genserv.py Port:443      <- secure web enabled
2018-03-10 20:19:29,673 : Restarting: /home/pi/genmon/startgenmon.sh restart  <- save settings
2018-03-10 20:19:33,385 : Starting /home/pi/genmon/genserv.py Port:8000   <- default settings

proppilot, 2018-03-11:
2018-03-10 22:18:53,651 : Error in app.run:[Errno 98] Address already in use
2018-03-10 22:18:55,654 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 22:18:59,483 : Starting /home/pi/genmon/genserv.py Port:8000

After I make the settings I clear the cache in Safari and then enter https://192.168.0.203 which is the IP address of the PI and get this

[image]

If I just do 192.168.0.3 I get the same thing.  If I amend :8000 I get the web page

Could this be any way related to port forwarding I have set up in my router ?  I have port 1963 forwarded to 8000 for my external DDNS connection

Regardless of how I call up the web page, the settings are reset.  Now just to confirm, I am not entering anything into the two lines for 'key file' or 'certificate file' correct ?

PS : if there was a wiki on setting this up please point me as I really apologize for spending this time going back and forth on something I am just not getting !

jgyates, 2018-03-11:
No worries, this actually helps me determine what should improve the software.

Yes, you should leave key file and cert file empty unless you have your own signed key certificate. 

Help me understand this a little better...are your trying to access the genmon server outside of your local network? is this the reason for port forwarding?

The message: Starting /home/pi/genmon/genserv.py Port:8000 tells me that the port is not being set in your /etc/genmon.conf file. If secure settings was working then this would be 443, not 8000.

For our testing, let's first take the port forwarding out of the picture and validate that this is working without the router doing anything. 

I noticed that in the above picture you sent you are setting your port number to 1963 is this related to your port forwarding?

If you can send the following I can likely nail this down:

*Last several entries in genserv.log
*The settings of that you are trying to make on the settings page
* How are these settings currently set in your genmon.conf file: usehttps ,useselfsignedcert, http_port
* A screen cap of your settings page (just the web config stuff) for now just use dummy un and pw

proppilot, 2018-03-11:
Yes, I am using xxxxxx.ddns.net:1963 to access the pi from outside.  In my router I have port 1963 mapped to 8000.

last several entries from genserv.log

2018-03-10 19:49:42,537 : Starting /home/pi/genmon/genserv.py Port:8000
2018-03-10 19:49:42,545 : Error in app.run:[Errno 98] Address already in use
2018-03-10 19:49:44,548 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 19:49:48,357 : Starting /home/pi/genmon/genserv.py Port:8000
2018-03-10 19:49:48,868 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 19:49:52,656 : Starting /home/pi/genmon/genserv.py Port:8000
2018-03-10 19:49:52,664 : Error in app.run:[Errno 98] Address already in use
2018-03-10 19:49:54,667 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 19:49:58,505 : Starting /home/pi/genmon/genserv.py Port:8000
2018-03-10 19:52:54,043 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 19:52:57,899 : Starting /home/pi/genmon/genserv.py Port:8000
2018-03-10 19:52:57,907 : Error in app.run:[Errno 98] Address already in use
2018-03-10 19:52:59,913 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 19:53:03,753 : Starting /home/pi/genmon/genserv.py Port:8000
2018-03-10 19:53:04,405 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 19:53:08,175 : Starting /home/pi/genmon/genserv.py Port:8000
2018-03-10 19:53:08,183 : Error in app.run:[Errno 98] Address already in use
2018-03-10 19:53:10,186 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 19:53:14,055 : Starting /home/pi/genmon/genserv.py Port:8000
2018-03-10 19:53:16,544 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 19:53:20,413 : Starting /home/pi/genmon/genserv.py Port:8000
2018-03-10 19:53:20,421 : Error in app.run:[Errno 98] Address already in use
2018-03-10 19:53:22,424 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 19:53:26,252 : Starting /home/pi/genmon/genserv.py Port:8000
2018-03-10 19:59:58,961 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 20:00:03,073 : Starting /home/pi/genmon/genserv.py Port:8000
2018-03-10 20:00:03,081 : Error in app.run:[Errno 98] Address already in use
2018-03-10 20:00:05,084 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 20:00:08,894 : Starting /home/pi/genmon/genserv.py Port:8000
2018-03-10 20:00:11,213 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 20:00:14,998 : Starting /home/pi/genmon/genserv.py Port:8000
2018-03-10 20:00:15,006 : Error in app.run:[Errno 98] Address already in use
2018-03-10 20:00:17,009 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 20:00:20,872 : Starting /home/pi/genmon/genserv.py Port:8000
2018-03-10 22:18:49,771 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 22:18:53,643 : Starting /home/pi/genmon/genserv.py Port:8000
2018-03-10 22:18:53,651 : Error in app.run:[Errno 98] Address already in use
2018-03-10 22:18:55,654 : Restarting: /home/pi/genmon/startgenmon.sh restart
2018-03-10 22:18:59,483 : Starting /home/pi/genmon/genserv.py Port:8000

here are the settings I am making

[image]

In genmon.conf I have changed nothing from the default file downloaded today with the update except the site name at the top.  

usehttps, useselfsignedcert and http_port are all commented out still with #.  The port is set to 80 but is #http_port = 80

I think I see where this is going that I have missed reading something on configuring genmon.conf for https use ? !! 

What I cannot figure out is why when I set the parameters I show above and go back into the page, they are empty again.

jgyates, 2018-03-11:
When you are testing on your local network, are you using the xxxxxx.ddns.net:1963 address in your browser or 192.168.0.203? Try only using the 192.168.0.203 address on your local network and disable port forwarding and retest. This will let us know if the router is affecting things. the javascript communicates with the Flask server using the port defined in the settings page (Web Server Port) . Depending on how your network is setup this could be causing a problem. Is your local network one subnet (i.e. everything is 192.168.0.x)?

I think you are looking at your genmon.conf in your /home/pi directory and not your /etc/genmon.conf file. If you had #http_port commented out (# denotes a comment) then it would show "(disabled)" underneath the settings in the web page.

For access outside your house, that is very much dependent on your router, but I would recommend a VPN instead of port forwarding. For example I have a VPN configured in my router and to access the site from my mobile phone I enable the VPN then just pull up the page as if I were on the network locally. It is much more secure that way. The folks who developed Flask do not recommend exposing it to the outside world unless you setup a full blow web server and setup Flask to route the pages through the full server. Having said that, let's focus on getting your system working as expected on the local network and then you can decide if you want to access it from the outside.

jgyates, 2018-03-11:
I checked in a new version of genserv.py that prints a little more info to the log file for debugging purposes.

[23 more replies]

closed 2018-04-04 by jgyates