jgyates/genmon #78: Issue upgrading software

proppilot, 2018-03-10:
Jason,  I finally took the plunge to upgrade to the latest checked in software.  I have a couple of issues.

1)	The sh file startgenmon.sh is failing in the crontab @reboot with an error (I see it on the command line as well )

Startgenmon.sh 3: startgenmon.sh: Bad substitution

2)	The web interface still looks like the original one.  None of the new 'stuff' that has been added.

Did something change in the nomenclature for the startgenmon.sh script file ?

GitHubUserToo, 2018-03-10:
Hi Proppilot:

I've been behind a while and upgrade a week ago to get some of the new awesome adds.  Had both exact same problems.  Don't recall why, but these fixed it for me:
1)  Had to use my pi user (not root) and set the start in 'crontab -e', instead of /etc/crontab (maybe run as that user).  I'm using:
@reboot sleep 15 && /home/pi/genmon/startgenmon.sh start

2)  Had to clear my browser cache to see the new UI.  Refresh didn't work. :)

proppilot, 2018-03-10:
Hi GitHubUserToo:

The cache clearing worked perfectly on my mac.  modifying the crontab for the pi user seems to be another issue.  When i 'crontab -e' instead of 'sudo crontab -e' it opens up the crontab for pi user as a temp file.  I am assuming it is because I am logged in as user pi ?  The 'sudo crontab -e' opens the file in an editor yet without the sudo it is only a temp file.  I can add text but not save it.  So, how is the crontab for pi user modified as you suggest ?

jgyates, 2018-03-10:
Can you run startgenmon.sh as a normal user from the command line.

to run crontab for the pi user, log in as the user, then run "crontab -e". To run as root use "sudo crontab -e". Just leave off the "sudo" to run it as the pi user.

GitHubUserToo, 2018-03-10:
Proppilot:

Hmmm...... just use 'crontab -e' - it shouldn't need elevated perms.  That work?

You could try in /etc/crontab and where 'root', change to 'pi' (the user it runs as).

(pilot also :) )

proppilot, 2018-03-10:
when I run the old version of startgenmon.sh with the start or stop command all runs well either from the command line or crontab.

when I run the latest version of startgenmon.sh I get the error

Startgenmon.sh 3: startgenmon.sh: Bad substitution

I assume this has something to do to the directory pointing that is now in the script (sorry as that is probably the wrong name for it)

When I have the new startgenmon.sh in the crontab file it does not start genmon at reboot as I suspect it is erring out on the same error I see at the command line.

I am trying to figure out how to put the @reboot command into the crontab for user pi but running into the issue I mention above.

proppilot, 2018-03-10:
[image]

when I open crontab with sudo crontab -e there are edit commands at the bottom however without the sudo you see what I posted ( opened in vim if that makes sense ?)

proppilot, 2018-03-10:
when sudo crontab -e looks like this

[image]

buschauer, 2018-03-10:
The problem is that the command reads 
    @reboot sh /home/pi/genmon/startgenmon.sh start
if you change it to 
   @reboot /home/pi/genmon/startgenmon.sh start

it works fine.  

Jason - do you want to convert the script back to sh instead of bash? I missed the fact that the instructions had "sh" in the crontab  as a default

proppilot, 2018-03-10:
Not to question those way smarter than me on this ! however the command line with the sh is commented out.  The issue I am having is not with crontab, it is with the new sh file for startgenmon generating an error.  Putting what you suggest in crontab (root where it was until the upgrade) does not start genmon

jgyates, 2018-03-10:
How are you running the version on the command line. Type exactly what you are typing on the console.

What distribution of linux are you running?

proppilot, 2018-03-10:
> <reply+02039b7af19e9fec546813884776868375b708323753f05192cf0000000116bbddff9
> 2a169ce121ff9df@reply.github.com>

Logged in as pi and in the genmon directory

Sh startgenmon.sh start

Linux GenMon 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l

<https://github.com/jgyates/genmon/issues/78#issuecomment-372049663> , or
mute the thread 
8xt623-ks5tdBN_gaJpZM4SlWRS> .

  
b","name":"GitHub"},"entity":{"external_key":"github/jgyates/genmon","title"
:"jgyates/genmon","subtitle":"GitHub
repository","main_image_url":"https://cloud.githubusercontent.com/assets/143
418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"h
ttps://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6
-9aed-b52498112777.png","action":{"name":"Open in
GitHub","url":"https://github.com/jgyates/genmon"}},"updates":{"snippets":[{
"icon":"PERSON","message":"@jgyates in #78: \r\nHow are you running the
console.\r\n\r\nWhat distribution of linux are you
running?\r\n"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/78#issuecomment-37204
9663"}}}

jgyates, 2018-03-10:
bingo. Don't type "sh"

Just type "./startgenmon.sh start"

typing "sh" will forct to use the Bourne Shell, "bash" will force the use of "Bash" shell, if you don't specify the script will specify by the first line. The script is Bash, but you are telling it to force Bourne.

proppilot, 2018-03-10:
pi@GenMon:~/genmon $ ./startgenmon.sh start
-bash: ./startgenmon.sh: Permission denied
pi@GenMon:~/genmon $

jgyates, 2018-03-10:
have you changed the permissions of the file? 

     sudo chmod 775 startgenmon.sh

This is required to make it executable. Prepending the "sh" likely got around that.

jgyates, 2018-03-10:
Tony,

Are you still having a problem.

Also, I just updated genserv.py. This should not affect your issue, but it will allow you to change all of the settings via the web interface. Before you could not change the ones related to the web server due to the fact that the code to restart the web server was not written yet.

proppilot, 2018-03-10:
Done and done.  The issue in the end was the permissions on startgenmon.sh and using sh instead of ./

Thanks Jason and others for helping out.  I was leery to do the upgrade as my original version has been running so well.  I cloned the SD to have a sandbox version and the upgrade was easy.

Professional grade !

proppilot, 2018-03-10:
okay just did another pull and update with everything working perfectly.

closed 2018-03-11 by jgyates

proppilot, 2018-06-03:
Jason,  I have one ongoing issue.  When I make any changes on the SETTINGS page, it appears to save them but the changes ( IE enabling metric for the temp) do not take.  I have had this issue since this page was introduced.  Could it be a permissions issue ?  Otherwise, where is the file that these configurations are stored as I can manually edit it.

jgyates, 2018-06-03:
maybe, let's take a look:

What is the output of this command if you are in the genmon directory:

     ls -la genmon.py

It should look someting like this:

pi@genmon:~/genmon $ ls -la genmon.py
    -rw-r--r-- 1 pi pi 39726 Jun  2 19:34 genmon.py

specifically the "pi pi" part

While here let's also check this:

     ls -la 

This will show all files an directories. Do you have a direcotry named ".git"?

Finally let's check the permission and size of the /etc/genmon.conf file?

    ls -la /etc/genmon.conf 

-rw-r--r-- 1 root root 8132 Jun  2 22:47 /etc/genmon.conf

This file should be owned by root

Also, look at /var/log/genserv.py to see if you have any errors.

jgyates, 2018-06-03:
Do you have any errors in genserv.log?

You can enter this URL and it will email me your log files:

http://yourIPaddress:8000/cmd/sendlogfiles

Replace yourIPaddress with the address of your pi and replace 8000 if you have changed the default port.

proppilot, 2018-06-03:
> <reply+02039b7a55867f582bfaba29c27c5464e1ad1b74d7f1825492cf00000001172c117a9

Log files sent

Date:  Sunday, June 3, 2018 at 4:28 PM

http://yourIPaddress:8000/cmd/sendlogfiles

V8SNxTuks5t5Eb6gaJpZM4SlWRS> .
type":"ViewAction","target":"https://github.com/jgyates/genmon/issues/78#iss
uecomment-394188974","url":"https://github.com/jgyates/genmon/issues/78#issu
ecomment-394188974","name":"View Issue"},"description":"View this Issue on
GitHub","publisher":{"@type":"Organization","name":"GitHub","url":"https://g
sage_cards/header.png","avatar_image_url":"https://assets-cdn.github.com/ima
ges/email/message_cards/avatar.png","action":{"name":"Open in
GitHub","url":"https://github.com/jgyates/genmon"}},"updates":{"snippets":[{
"icon":"PERSON","message":"@jgyates in #78: Do you have any errors in
genserv.log?\r\n\r\nYou can enter this URL and it will email me your log
files:\r\n\r\nhttp://yourIPaddress:8000/cmd/sendlogfiles\r\n\r\nReplace
changed the default port.\r\n"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/78#issuecomment-39418
8974"}}}{ "@type": "MessageCard", "@context":
"http://schema.org/extensions", "hideOriginalBody": "false", "originator":
"37567f93-e2a7-4e2a-ad37-a9160fc62647", "title": "Re: [jgyates/genmon] Issue
upgrading software (#78)", "sections": [ { "text": "", "activityTitle":
"**jgyates**", "activityImage":
"https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"name": "Add a comment", "@type": "ActionCard", "inputs": [ { "isMultiLine":
true, "@type": "TextInput", "id": "IssueComment", "isRequired": false } ],
"actions": [ { "name": "Comment", "@type": "HttpPOST", "target":
"https://api.github.com", "body": "{\n\"commandName\":
\"IssueComment\",\n\"repositoryFullName\": \"jgyates/genmon\",\n\"issueId\":
78,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}" } ] }, { "targets": [
{ "os": "default", "uri":
"https://github.com/jgyates/genmon/issues/78#issuecomment-394188974" } ],
"@type": "OpenUri", "name": "View on GitHub" }, { "name": "Unsubscribe",
"@type": "HttpPOST", "target": "https://api.github.com", "body":
"{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 311780434\n}" } ],
"themeColor": "26292E" }

proppilot, 2018-06-03:
When the changes are made on the web page, where are they saved ?

http://yourIPaddress:8000/cmd/sendlogfiles

"https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"https://github.com/jgyates/genmon/issues/78#issuecomment-394188974" } ],

jgyates, 2018-06-03:
Changes are save to /etc/genmon.conf or for mail settings they are saved to /etc/mymail.com

I am not seeing an error writing the file. Let's do this and see if this corrects the problem:

make a copy of your config file:

    sudo cp /etc/genmon.conf.bak

Copy the last genmon.conf

     sudo cp /home/pi/genmon.conf /etc/genmon.conf

Restart genmon
 
     bash ./startgenmon.sh restart

Now if you go to the settings page you will need to re-enter you settings. you can copy and past them from /etc/genmon.conf.bak if you dont know them off the top of you head.

Also, based on your mymail.log file it looks like you have incoming mail enabled but your Incoming Mail Folder setting is not a valid IMAP folder. You have a lot of these errors in you mail log file:

    Error selecting mail folder! (select)

If you don't want use incoming email for genmon commands remove the IMAP server. If you do want to use it you must have a valid IMAP folder for the Incoming Mail Folder setting.

proppilot, 2018-06-03:
> <reply+02039b7ad7ef8dcc14573a06a4e67e5b193247920399929c92cf00000001172c14e99

Seems to have fixed the problem.  There was quite a few differences between
the file I was using the the latest one.

I know there have been multiple program upgrades that have likely driven the
format of the genmon.conf file

When I upgrade the software, should I be replacing the genmon.conf in the
/etc directory to ensure it is the latest ?

Date:  Sunday, June 3, 2018 at 4:43 PM

4YJ3uRAks5t5EppgaJpZM4SlWRS> .
uecomment-394189895","url":"https://github.com/jgyates/genmon/issues/78#issu
ecomment-394189895","name":"View Issue"},"description":"View this Issue on
"icon":"PERSON","message":"@jgyates in #78: Changes are save to
/etc/mymail.com\r\n\r\nI am not seeing an error writing the file. Let's do
this and see if this corrects the problem:\r\n\r\nmake a copy of your config
file:\r\n\r\n    sudo cp /etc/genmon.conf.bak\r\n\r\nCopy the last
genmon.conf\r\n\r\n     sudo cp /home/pi/genmon.conf
/etc/genmon.conf\r\n\r\nRestart genmon\r\n \r\n     bash ./startgenmon.sh
restart\r\n\r\nNow if you go to the settings page you will need to re-enter
dont know them off the top of you head.\r\n\r\nAlso, based on your
these errors in you mail log file:\r\n\r\n    Error selecting mail folder!
(select)\r\n\r\nIf you don't want use incoming email for genmon commands
folder for the Incoming Mail Folder setting.\r\n"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/78#issuecomment-39418
9895"}}}{ "@type": "MessageCard", "@context":
"https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"https://github.com/jgyates/genmon/issues/78#issuecomment-394189895" } ],

jgyates, 2018-06-03:
You are not suppose to have to re-copy the file after every upgrade, but there may be a bug in the code or this could be a one off problem due to something odd in your conf file. I will do some testing to see if I can duplicate this. 

The conf file on linux is sort of odd, if you use the standard libraries for reading everything is pretty straight forward. If you use the standard libraries for writing, you can't have comments as they get stripped out on the write. Even when reading, comments and entries can not co-exist on the same line. You can comment out a line and it is 100% a comment, but if you have an entry with a comment at the end things get confusing for the libraries. As a result we ended up writing our own write functions. Long story short, if you ever added comments to the a line after the entry it could cause a problem like this.

Let me know if you have any other issues.