jgyates/genmon #80: SQL for Log

mattm55, 2018-03-16:
First off, nice work on this project. Documentation is A+.

I looked through and didn't see a mention of saving logs to xSQL tables. That might be a nice addition.

I'm in the process of getting a Generace 9k or 11k gen and will definitely go this route. I do run Homeseer and will look into writing a "pi" if that would be OK?

Thanks, Matt

madisdev, 2018-03-16:
It would be a nice addition, feel free to create a pull request with those features implemented :)

I was wondering, what do you mean you will look into writing a "pi"?

@jgyates I was also thinking about possibly adding SQLLite to store data. Thoughts?

mattm55, 2018-03-16:
Thanks, will do.

Ah, those acronyms... bad habit from work I guess. Yes, "pi" would refer to a Homeseer3 (HS3) **plugin**. That way we can integrate the data from the gen output into Homeseer. SQLLite >> Yes, the HS3 Concord plugin (GE/ITI alarm system) I took over ownership uses that db already as do many other plugins for HS as a standard. That would be great.

Thanks, Matt

madisdev, 2018-03-16:
Never heard of Homeseer before, I dont really delve into that home automation commercial products that much. Seems a bit pricey for what is probably a raspberry PI with custom software :)

jgyates, 2018-03-16:
SQLLite sounds like a good piece of functionality. I checked in a java version of ClientInterface to the repository under OtherApps. 

@mattm55 - This page may be useful with integrating with Homeseer. 

https://github.com/jgyates/genmon/wiki/Interfacing-GenMon-to-External-Applications

I have looked at homeseer but have not implemented it. I use Nagios to monitor everything. Once you get going let me know if you have any questions. 

FYI, the generator only keeps 50 logs for each log (maintenance, on/off, and alarms for Evolution and alarm, on/off for Nexus). The outage log is maintained by genmon in a text file. Error logs for the software are stored in /var/logs.

jgyates, 2018-03-16:
one more piece of data. For a quick status of the generator the "getbase" command will return a one line status (RUNNING, MANUAL, EXERCISING, SERVICEDUE, RUNNING-MANUAL, or ALARM). This command is used by the web interface (to set the color of the site based on status) and the Nagios plugin to determine if an alert needs to be triggered by Nagios. It may be useful as a first step in integrating with a smart home hub.

mattm55, 2018-03-16:
Thanks for the info and link to Interface page. I'll take another loo around for sure.

Yes, Homeseer is pricey (50% off most times) and is quite capable. I'm running alarm, temp, thermostat, etc off. Not for everyone for sure. Maybe just the anal people that need to know whats going on at home  ;o)  Al of my development is always free. I'm a Java developer but can work my way thru most code.

Thanks, Matt

dennis48755, 2018-03-17:
I also use HomeSeer (purchased during a 50% off sale) to integrate camera motion detection via Blue Iris, security system, X10 modules, Z-Wave modules, Control By Web devices and Amazon Alexa.

Integration with GenMon would be the icing on the cake. As would SQL logging.

Dennis

closed 2018-03-26 by jgyates