jgyates/genmon #81: 3.3 or 5 volt to power the RS232 driver?

hokie21, 2018-03-26:
Looking at pictures of the builds people have done as well as the schematic, it appears that everyone is using the 5V pin 2 or 4 from the rpi to power the TTL to RS232 converter. Shouldn't it be powered by the 3V3 pin 1 since the GPIOs on the rpi are 3.3 volt logic?

jgyates, 2018-03-26:
When you purchase a TTL to RS-232 converter it should be one that supports 3.3V. This page as a link for two different vendors that are known to work.

https://github.com/jgyates/genmon/wiki/Hardware

The converter is powered by the 5V pin but the the inputs TTL inputs support 3.3V. If you purchase a converter that only supports 5V it will not work.

hokie21, 2018-03-26:
I was concerned about the output from the converter driving the RPI. Won't this drive the 3.3V RPI GPIO input with a 5 volt signal? I assume this is true, but that it doesn't cause a problem due to the design of the RPI inputs?

jgyates, 2018-03-26:
Here is the data sheet for the chip on the converter:

http://www.ti.com/lit/ds/symlink/max3232.pdf

I have been running mine for about 2 years with no issues. It has been a while since I reviewed the data sheet, but I do remember that a 5V only converter would not work.

closed 2018-03-26 by jgyates

hokie21, 2018-03-26:
Here is the rpi GPIO equivalent circuit. http://www.mosaic-industries.com/embedded-systems/microcontroller-projects/raspberry-pi/gpio-pin-electrical-specifications

I think I'm going to use 3.3V from rpi pin 1 to power the RS-232 converter for my build.

jgyates, 2018-03-26:
OK, sounds good Let me know if you find that it works or not. I would love to add that data point to the wiki.