jgyates/genmon #82: RPM Gauge Reading Off Dial

alphaa211, 2018-03-28:
### Expected Behavior

"New" Genmon RPM Gauge Reading in Green between 3550-3650.   

### Actual Behavior

RPM Gauge Reads Past Dial Completely

### Steps to Reproduce (including precondition)

Weekly Exercise tonight or automatic remote start through Genmon; RPM gauge does the same thing when the unit is running

### Screenshot on This Problem (if possible)

[image]

### Your Environment

- Generator Model: 7033 Evolution Air Cooled, Hardware V1.06, Firmware V1.16
- Generator Registers: {Please write here}
- Genmon Version: V1.5.11 – Latest git pull 8pm tonight CST 3/27/18

jgyates, 2018-03-28:
@alphaa211 

Thanks for the feedback. I know what is causing this issue and we are actually working on a fix as we speak. The web UI uses the nominal RPM to set the colors and max on the gauges. The problem is that the nominal RPM value is incorrect. We have been trying to find a way to read the nominal RPM and kw rating of the generator from the controller registers, but as your bug report proves this is not entirely consistent. One that that would help is if you could send me a copy of your registers (use ClientInterface.py and enter "generator: registers" copy and paste the output to a file and post it here). We should have an updated in a day or two that will fix this issue.

Thanks
Jason

alphaa211, 2018-03-28:
Had to read a bit to get that done ;).  Great project and the hard work is much appreciated.

Registers :
    Num Regs : 212
    Not Changed : 23736
    Changed : 424
    Total Changed : 0.02

    Base Registers :
        0038 : 0001
        0034 : 0000
        0037 : 1156
        0033 : 0050
        0032 : 4000
        05f5 : 0000
        05f4 : 0000
        05f1 : 0000
        01f3 : 1211
        01f1 : 110f
        05ed : 0064
        023b : 0000
        023e : 0000
        003b : 0000
        05fa : 0000
        0059 : 0000
        0058 : 0000
        0053 : 0000
        0052 : 0000
        0057 : 0000
        0056 : 0000
        0055 : 0000
        0054 : 0000
        0208 : 03f8
        005c : 0000
        005a : 0000
        005f : 0000
        005d : 0000
        003c : 0000
        000d : 0000
        000e : 1424
        000f : 031b
        000a : 0080
        000c : 0033
        0005 : 111e
        0006 : 0200
        0000 : 0009
        0001 : 00000000
        0008 : 0000
        0009 : 00f6
        01f2 : 0221
        05ee : 0013
        0012 : 0000
        0011 : 00ab
        0010 : 0212
        0007 : 0000
        0019 : 000f
        002b : 0005
        002c : 111e
        002a : 6a74
        002f : 0000
        002d : 0000
        002e : 0002
        0020 : 0000
        0021 : 0000
        001f : 4915
        001e : 015c
        001d : 0000
        001c : 0000
        001b : 463b
        001a : 0094
        01f4 : 33303030373839363736

jgyates, 2018-03-28:
@alphaa211 ,

I just checked in a version of genmon.py that should fix this for you. There will be an update in a few days that will have a more permanent solution. The update I just checked in makes the assumption that if you have an air cooled model your nominal RPM is 3600 (liquid cooled is now 1800). This is true unless you are running at 50Hz, which I am unaware of anyone doing currently. The future release will have similar behavior but give you the option to override the defaults in the settings page.

Let me know if you still have an issue.

alphaa211, 2018-03-29:
Seems to be working fine now.

[image]

closed 2018-04-04 by jgyates