jgyates/genmon #83: Ambient Temperature

alphaa211, 2018-04-02:
Feature Request:

Generator Model: 7033 Evolution Air Cooled, Hardware V1.06, Firmware V1.16

Any plans or ability to show the Generac Ambient Temp on the Status screen?

It looks like all the other data that Generac displays has already been included.  It would be nice for those of us that use block heaters in the winter and for outages in the high heat of summer.  I'm  not sure exactly where that sensor reads from, but I assume it is used to determine if the Cold Smart Start warm-up is active.             

[image]

[image]

liltux, 2018-04-02:
That thermistor is located in harness beneath controller.  Measures temp of battery compartment for charging characteristics as well as other features such as Cold Smart Start.

jgyates, 2018-04-03:
If the Evolution Air Cooled controller exports this to the modbus registers it should be relatively easy to support this feature. I have a Liquid Cooled model so I will need a volunteer to run a few tests. If you are interested in testing and have an Evolution Air Cooled model you can message me at https://hackaday.io. My user name is jgyates. The test involves:

1) I will need a dump of your registers (using the "registers" command  via ClientInterface.py
2) I will send you a python program that will dump all of the registers in your controller to a file, including the ones that genmon does not currently read. When your run this make a note of the ambient temp from your controller display
3) Wait for the temp to drop or rise, run test two again. Make a note of the temp again 

Then send me all three files (register output from genmon, 2 full register dumps) along with the two temp readings.

This will allow me to see if the temp sensor is exported to the modbus registers. If it looks like it is then, depending on what the register values looks like we may need to run another test. The possible second test will log the candidate register(s) to a file along with the current temp (The same program will do this). I have found what I think is the ambient temp sensor on the Liquid Cooled Evolution. I still have not figured out the formula for converting to an actual temp as appears to use a NTC (Negative Temp Coefficient), the value goes down if the temp goes up.

jgyates, 2018-04-03:
I got a volunteer. Thanks. I will update this thread with what I find.

jgyates, 2018-04-03:
Sorry folks. Ambient Temp is not exported via modbus. Thanks @krisasmith for helping test. 

Thanks for the suggestion @alphaa211 . Let me know if you have any other ideas.

closed 2018-04-03 by jgyates