jgyates/genmon #84: Help with wiring

kalex1114, 2018-04-05:
Hello
I finally got all parts and starting to assemble the cable. 
I need a little help with wiring :)
I bought female rs232 to ttl and male break out connector.
I also bought ATX cable with connector on it and will be using Raspberry PI3

The way I understand it is that I need to cut the ATX cable and leave the molex connector with cables attached.  
Is this correct?
Pin 1 of the molex calbe will plug into pin 2 on the PI3
Pin 2 of the molex cable will plug into pin 9 on the PI3
Pin 7 of the molex cable will plug into pin 2 RXD on the breakout connector
Pin 8 of the molex will plug in into pin 3 TXD on the breakout connector
VCC dupont connector on rs232/ttl adapter will go into pin 4 on PI3
RDX dupont connector on rs232/ttl adapter will go into pin 10 on PI3
TXD dupont connector on rs232/ttl adapter will go into Pin 8 on the PI3
GND dupont connector on rs232/ttl adapter will go into pin 6 of the PI3

Did i get these correctly?

Here are pictures just to make sure i got right parts.
Molex from ATX cable:
[image]

Male breakout cable:
[image]

female rs232/ttl adapter:
[image]

[image]

[image]

JustApu, 2018-04-05:
`` Pin 7 of the molex cable will plug into pin 2 RXD on the breakout connector``
``Pin 8 of the molex will plug in into pin 3 TXD on the breakout connector``

I believe you need to swap these two but everything else looks correct to me. Caveat: My parts are still en route so I haven't built mine yet.

jgyates, 2018-04-05:
Not sure if you have seen this, but there is a diagram on this page that should help:

https://github.com/jgyates/genmon/wiki/Making-a-Cable

kalex1114, 2018-04-05:
Yep I saw the making a cable diagram.  I followed it just wanted to double check i got it right.

kalex1114, 2018-04-06:
JustApu - are you sure? I followed the cabling diagram posted here:

https://raw.githubusercontent.com/jgyates/genmon/master/Diagrams/Cable_Connections.jpg

I was following DCE setup female rs232/ttl adapter and make 9 pin breakout connector
Pin 8 molex into pin 3 on the PI
Pin 7 molex into pin 2 on the PI

Is that not correct?

JustApu, 2018-04-06:
Not sure.

In looking at your link, I think you are correct. I was going with the TX/RX on the MAX3232 board and didn't notice the connectors on your parts for DCE vs. DTE.

jgyates, 2018-04-10:
@kalex1114, I will close this thread unless you still have questions. Feel free to comment in this thread if you do.

closed 2018-04-10 by jgyates

jgyates, 2018-04-10:
Nothing stands out to me as incorrect.