jgyates/genmon #85: Help - Cable Connection Inside Generator

jonathanpisarczyk, 2018-04-07:
Hey Guys,

I went to connect my Pi and cable to Generac.  However I can't find the appropriate place to connect my cable.  I've got a Evolution 20KW model.  See pictures below.  Any help would be appreciated. 

[image]
[image]
[image]
[image]

proppilot, 2018-04-07:
Peel off the Accessory Port 1 sticker and plug is there

jonathanpisarczyk, 2018-04-07:
Thank you !!! Never would have guessed that.

closed 2018-04-08 by jgyates

jonathanpisarczyk, 2018-04-09:
I know the issue has been closed.  However the picture may help someone in the future.  Here's my genmon cable going into accessory port 1.  Thanks again...
[image]