jgyates/genmon #86: Clearing Alarms

sgtmaj27, 2018-04-08:
I apologize in advance if this subject has been discussed, I did not find it in my search.

Is it possible to program a way to clear an alarm via the web interface?  

I realize that there are some alarms that you would not want to clear that way, but my model generac, 5870 (8kw Nexus) is famous for the overspeed error.  Then as soon as the alarm is cleared, the generator starts and runs fine.  Many have tried to isolate the problem and correct it, I have learned to live with this “procedure”.  

The way to clear the a latching alarm , at least on my model, is to turn the controller switch from auto to off and hit enter.  This clears the alam and invariably returning to auto restarts the generator with no errors.  A way to clear this alarm via your software would add, in my estimation, an invaluable feature to the package.

Thanks

jgyates, 2018-04-08:
Unfortunately, I know of no way to do this via software.

sgtmaj27, 2018-04-08:
That’s what I thought.  I figured if it could be done you would have
incorporated it....but I had to ask.

Thanks

closed 2018-04-08 by jgyates