jgyates/genmon #87: genmon stop working after serving

ts4s, 2018-04-08:
Last month I had my generate serviced. During the course of service, the battery was replaced. At the time i forgot about the pi til it was to late. Service Tech hard is all disconnected and new battery installed. since then i have been trying to troubleshoot (when its warm enough to be outside). I can pull a webpage but its not talking to the generate. So I'm wondering if the max3232 board or the pi serial port when bad on me.  I'm waiting on some new max3232 to arrive but that another story. 

snippet genmon.log from today:
`2018-04-08 16:29:13,562 : Error: timeout receiving slave packet for register 5f1 Buffer:1
2018-04-08 16:29:16,567 : Error: timeout receiving slave packet for register 1f1 Buffer:1
2018-04-08 16:29:19,570 : Error: timeout receiving slave packet for register 5ed Buffer:1
2018-04-08 16:29:22,573 : Error: timeout receiving slave packet for register 05 Buffer:1
2018-04-08 16:29:25,584 : Error: timeout receiving slave packet for register 06 Buffer:1
2018-04-08 16:29:28,585 : Error: timeout receiving slave packet for register 07 Buffer:1
2018-04-08 16:29:31,588 : Error: timeout receiving slave packet for register 00 Buffer:1
2018-04-08 16:29:34,590 : Error: timeout receiving slave packet for register 01 Buffer:1
2018-04-08 16:29:37,593 : Error: timeout receiving slave packet for register 053 Buffer:1
2018-04-08 16:29:40,596 : Error: timeout receiving slave packet for register 052 Buffer:1
2018-04-08 16:29:43,599 : Error: timeout receiving slave packet for register 5f4 Buffer:1`

snippet of myserial from the day of service: 
`2018-03-16 13:59:17,253 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 13:59:21,281 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 13:59:21,282 : Error in SerialDevice:Flush : /dev/serial0:Attempting to use a port that is not open
2018-03-16 14:00:00,317 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:00:12,408 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:00:12,527 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:00:12,914 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:00:19,303 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:01:12,066 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:01:13,358 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:01:17,648 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:01:18,514 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:01:18,700 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:02:05,794 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:02:08,780 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:02:54,987 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:03:45,271 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:03:52,096 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:04:01,098 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:04:01,366 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:04:07,976 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:04:52,868 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:05:24,230 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:08:10,156 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but$
2018-03-16 14:08:10,157 : Error in SerialDevice:Flush : /dev/serial0:(9, 'Bad file descriptor')
`

proppilot, 2018-04-08:
> <reply+02039b7a2ac37de25d690f5236e16f61ecbeb4f39a07718792cf0000000116e2513f9
> 2a169ce129e032f@reply.github.com>

Are you sure the serial cable is plugged back into PORT 1 on the controller
board underneath ?

O7ffUD0ks5tmoa_gaJpZM4TLueW> .

  
b","name":"GitHub"},"entity":{"external_key":"github/jgyates/genmon","title"
:"jgyates/genmon","subtitle":"GitHub
repository","main_image_url":"https://cloud.githubusercontent.com/assets/143
418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"h
ttps://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6
-9aed-b52498112777.png","action":{"name":"Open in
GitHub","url":"https://github.com/jgyates/genmon"}},"updates":{"snippets":[{
"icon":"DESCRIPTION","message":"genmon stop working after serving
(#87)"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/87"}}}

jgyates, 2018-04-08:
Hi @ts4s 

Based on the output you are not receiving any data, which could mean you are not transmitting to the controller, not receiving what comes back, or both. I would check the cable connections first and if they look good and you still get an error you can use serialtest.py to test the cable and MAX3232. If that fails then it may be a good guess that your have something not working in the hardware.

ts4s, 2018-04-09:
@proppilot - its a nexus unit, only has one port

@jgyates  - well this is interesting:

`pi@GeneratorPi:~/genmon/OtherApps $ python serialtest.py

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful
pi@GeneratorPi:~/genmon/OtherApps $ `

jgyates, 2018-04-09:
@ts4s 

When you tested with serialtest.py, just so we are communicating clearly, you did disconnect the cable from the Nexus controller and loop back TX to RX?

If you did then this tells us the cable is good. You can also do the test without the loop back to validate that it is fails serialtest.py. This will test to make sure that you do not have any shorts between tx and rx. 

If the first test passes (with loopback) and the second test fails (without the loopback), then we have a higher degree of confidence in your cable. 

You may be doing this already, but make sure you have a good contact with your molex connector and the controller. 

It is possible to have a valid loopback through the cable and still have bad RS-232 transceivers in the MAX3232. The likely hood of this would depend on how / if the transceivers were damaged.

Was your firmware upgraded during your service. I don't expect this to be a problem, but it never hurts to ask.

ts4s, 2018-04-09:
I unplugged the molex connector and jumpered the 2 wires with a paperclip. I did run the test while plugged in to the unit and it fail (probably not a wises test to do). I'm pretty sure I'm still running same old firmware that came with the unit, never seen the tech break out any electronics other than a meter. My service call consist of oil change, check valve lashings, and battery replacement every 3 years.

jgyates, 2018-04-09:
The test while connected, while not recommended, it likely fine. The controller will not respond unless it gets the exact sequence of bytes it expects. 

Are you powering your pi from the battery or from the connector on the back of the controller. If from the battery, you may want to check your power connections, if your ground is not connected properly / securely you could see the results you are seeing now.

jgyates, 2018-04-10:
@ts4s , were you able to get this working again?

ts4s, 2018-04-11:
nope, not yet... 

here's the things I've tried:
- switch power source, using my spare battery for cell phone charging (output 5v & 2.1 amps), no luck (some reason my gfci outlet is not working)
- built a new sd card, no luck
- tried a different PI 3, no Luck
- working on building a new cable, still collecting parts

going to look for logs since it's a new sd card now...

ts4s, 2018-04-11:
result of python serialtest.py

no pins shorted:

- Loopback testing for serial port /dev/serial0...
- Serial port opened
- write data: sent test string
- waiting to received data....
- FAILED: Sent data does not match receive. Received 0 bytes"
 
with pins shorted:
- Loopback testing for serial port /dev/serial0...
- Serial port opened
- write data: sent test string
- waiting to received data....
- PASSED! Loopback successful

ts4s, 2018-04-11:
one last piece, maybe this is the issue:

'pi@GenMonZero:/var/log $ /home/pi/genmon/startgenmon.sh restart
Stopping genmon python scripts
Starting genmon python scripts
pi@GenMonZero:/var/log $ WARNING: System Initializing. Not receiving data. : , 
 * Running on http://0.0.0.0:8000/ (Press CTRL+C to quit)
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET / HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET /jquery-3.3.1.min.js HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET /tooltipster.bundle.min.js HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET /vex.combined.min.js HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET /gauge.min.js HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET /jquery.jqplot.min.js HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET /printThis.js HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET /moment.min.js HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET /jquery.CalendarHeatmapGenmon.js HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET /genmon.css HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET /tooltipster.bundle.min.css HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET /vex.css HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET /vex-theme-os.css HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET /jquery.jqplot.min.css HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET /jquery.CalendarHeatmap.css HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:16] "GET /genmon.js HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:17] "GET /images/status.png HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:17] "GET /images/maintenance.png HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:17] "GET /images/log.png HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:17] "GET /images/outage.png HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:17] "GET /images/monitor.png HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:17] "GET /images/notifications.png HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:17] "GET /cmd/monitor_json HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:17] "GET /cmd/getfavicon HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:17] "GET /cmd/getexercise HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:17] "GET /cmd/getsitename HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:17] "GET /cmd/status_json HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:17] "GET /images/settings.png HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:17] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:17] "GET /images/registers.png HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:17] "GET /favicon.ico HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:20] "GET /cmd/getbase HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:22] "GET /cmd/status_json HTTP/1.1" 200 -
192.168.1.40 - - [10/Apr/2018 21:10:22] "GET /cmd/registers_json HTTP/1.1" 200 -'

shouldn't this line ' * Running on http://0.0.0.0:8000/ (Press CTRL+C to quit)' be "'* Running on http://192.168.1.40:8000/ (Press CTRL+C to quit)'

also the IP address(above) is to the on board wifi and I'm accessing the pi through a usb wifi with a different address, so I should probably disable the onboard correct?

jgyates, 2018-04-11:
If you are powering from a source other than the battery, you must make sure you have a ground connection coming from the controller through the cable if you are not powering from the battery, otherwise the controller and the pi will not have a common ground and your RS-232 will not work.

jgyates, 2018-04-11:
The "Running on http://0.0.0.0:8000/" is the normal display. this is the web server listening on all IP addresses of the PI.

I would say, either you have a bad ground connection or bad hardware. I am betting the former since you battery was replaced in your service. There would be no reason that I am aware of for the tech to do anything to the cable connecting to the PI or the controller.

ts4s, 2018-04-11:
lmao... so on a wild Wim, I took the pi and touched the metal case of the generator to the usb shielding and its start talking...so does that mean the ground from the controller need to be carried from the molex to the db9 connection now?

jgyates, 2018-04-11:
You either need to have the ground on the pi connected to either the generator battery or to the ground on the controller via the molex. If you try to power off a wall wart or a separate battery you will not have a common ground between the pi to the generator controller, which makes the RS-232 transmitter and receiver on the controller and pi not read different voltage levels. So, you can fix this one of 2 ways: 1) make sure you are powering off the battery and have a good ground connection, or 2) connect the ground on the pi to the ground on the controller via the cable.

ts4s, 2018-04-11:
sound like one should be carrying the molex ground back to the db9 connection to ensure good grounding. Guess tomorrow I'll work on cleaning the ground cable for the "battery to usb cable" to see if that improves things first. Then look at adding a wire to the molex/db9 cable.

ts4s, 2018-04-12:
Ok, I have my parts for a more simipler version. However I think the max3232 board is wired backward, need to confirm the orientation? See photos below

[image]
[image]

madisdev, 2018-04-12:
Would need to see backside of MAX3232 board to determine if you wired correctly.

madisdev, 2018-04-12:
I had a question. Did you change any wiring after the generator was serviced? Are the pictures we are looking at the hardware that previously worked?

ts4s, 2018-04-12:
@madisdev - these new picture that I've posted today is all new stuff... 

my original setup can be found in another [thread](https://github.com/jgyates/genmon/issues/14)... 
the new setup just eliminates the DB9 DTE & DCE connection thing

only thing that changed prior to this issue was updating the genmon software. Then here last month, the service tech swapped out the battery.

madisdev, 2018-04-12:
I'm not 100% sure, but I had some similar issues. I think you need to solder onto the back of the board and not the front. I got the feeling that the connections don't go through the board.

ts4s, 2018-04-12:
info on max3232: 

- The power and ground go through the board(pads with hole in them). 

- The other pads are for the pairs of tx & rx, chip has 4 pairs to use.  [page12](http://www.ti.com/lit/ds/symlink/max3232.pdf) shows the pin out of the chip. Pin 7 & 8 and 9 & 10 are the top side of the board. the bottom side is the other pins. 

serialtest.py works fine here on the desk with paperclip: 

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful

without paperclip:

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 1 bytes

I have verified that all wires have continuity from point to point for this new harness (picture posted today).

ts4s, 2018-04-15:
switch back to the generac monitoring unit til I can get time to figure out the gremlins in the unit. closing thread

ts4s, 2018-07-18:
Just an update on this since I found some time to work on this again. 

over the weekend, I was able to resolve my controller issue. I want to say thank you to the guys in this thread: 

https://github.com/jgyates/genmon/issues/117 

disconnected the connections as mention fixed my issue (yahoo! wasn't looking forward to having the controller replace out of my pocket)

Today, I built a new sd card with the lastest software and retest existing pi setup, No luck communicating when testing with the serial test script. Sat down and built new cable, DTE version since replacement board was that version, and grab another pi off the shelf(thankfully I have spares). Fired it up tonight and it's talking. 

  So when I have time again I will order another DCE board and see if thats the cause of the issue and retest the original pi.

  In the future I will have to remember to shutdown the pi and disconnect it when service time comes around again.

@jgyates Thanks for the software, it keeps getting better and better with the updates. So much easier to install now 👍

closed 2018-07-18 by ts4s

jgyates, 2018-07-18:
@ts4s , thanks for the feedback. Glad to hear it is working.