jgyates/genmon #89: Stopped working after software upgrade

lqallison, 2018-04-12:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

### Expected Behavior

One module update

### Actual Behavior

After update graphics missing in web monitor and no menu operation
### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot on This Problem (if possible)

{Please write here}

### Your Environment

- Genereator Model:5875
- Generator Registers: {Please write here}
- Genmon Version: file date 3/23/18

jgyates, 2018-04-12:
@lqallison 

Can you describe what you mean by "one module update"? If you only updated one file then that is likely your problem. The modules are designed to work together so if you only update one file you will probably see missing pieces in the user interface.  This wiki entry describes the update process:

https://github.com/jgyates/genmon/wiki/Updating-to-the-Latest-Version

rbrown1883, 2018-04-12:
My Apologies! I clicked on "Fork" on this project hoping to see a list of the 13 forks in the box.
Instead, now it says "14" and I must have unintentionally created another fork.
I do not intend to add anything to this code, just assist my friend lqallison.
If Mr. Yates can delete my unintended fork, it will help not build clutter into this great project.
Thanks go to Mr Yates for a great project, and I apologize for my blunder.

lqallison, 2018-04-12:
Downloaded new master file and changed static/genmon.js after which genmon stopped serving
graphics. Still querying generator but no readouts on web browser. Should I have used all files?
Not knowledgeable about Github and how to update without losing config settings.

jgyates, 2018-04-12:
@rbrown1883 ,When you create a fork it just creates a forked copy in your github account. The forked copy will not be updated or affect the main repository so you are good.

@lqallison , you you need to update all the files in your genmon directory and subdirectories. If you initially downloaded the repository using the "git clone https://github.com/jgyates/genmon.git" command as described in the setup instructions, you   can just issue the following commands in your genmon directory:

     git fetch origin
     git reset --hard origin/master

This will update all the needed files. You specific settings are in the /etc/genmon.conf directory so updating the software will not change your settings.

lqallison, 2018-04-12:
Updated software ran ok until I remote started generator then tried to remote stop it. Had to go to ClientInterface to issue stop command which worked. Did not try changeover.
Password still visible on settings page.

jgyates, 2018-04-12:
@lqallison , I need more info in order to answer any questions.

1) How did you update the software? once you updated, did you refresh your browser. You can have updated software on the pi, but if your browser does not know you updated it the browser will still be using the cached version.
2) What specifically is not working? Your previous message does not describe what is not working.

lqallison, 2018-04-12:
Remote commands on Maintenance Page of Generator Monitor. Stop has green color to button at all times. Start has gray color at all times. Start button worked with no change of color with generator running but does have "Are You Sure " pop up to start. Stop button had green color with generator running and clicking on it produced no visible change to indicate command sent and did not stop the running generator.
Now it will stop the generator  but  no 'Are You Sure" pop up dialog box and no change in color. Soft ware was updated and reboot. Email Password visible in settings page.
 git fetch origin
git reset --hard origin/master
Also clicked upgrade to latest monitor software

jgyates, 2018-04-12:
If the email pw is visible then you are not running the latest software. It is possible that something is wrong with your file permissions, but you did not include the output of the git commands. Please re-run these commands and include the full output. Also:

What directory is your software located and what directory were you in when you ran the git commands? 
What was the output of the git commands, were there any errors. 
It would be useful if you were to include pictures of the terminal commands or at least a copy and paste of the screen output.

lqallison, 2018-04-12:
[image]
Everything is in home/pi/genmon dir. Git commands run from that dir. rebooted after git. chmod +x startgenmon.sh  ran startgenmon.sh start. earlier in the day I also had a KW graph on screen but do not now. password not visible now. tried to add password security to genmon via settings page changed port to 8137 added user name and password 8 char. could not open a browser connection or get in to change back without editing etc/genmon.conf. back in after commenting out all having to do with password/user. 
1) what happen to KW version?
2) why do we have to chmod startgenmon.sh each time?
3) almost ready to look at L1, L2 current monitoring using Pimoroni ExplorerHat board. here is code used in an example to read board written in python. when run lists port and values down left side with
(port,value) . would not let me put python version here but can be found at Pimoroni.com Pimoroni/explorerhat/examples
[analog-event4.py.txt](https://github.com/jgyates/genmon/files/1905068/analog-event4.py.txt)

jgyates, 2018-04-12:
OK, your problem appears to be a combination of either not having updated the software completely, or your web browsers is caching items that the server has flaged not to cache. That problem was made worse by editing the genmon.conf file. You added comments to the end of the line, which is not allowed in this file format. We are using the standard library configparse to read the conf file and based on the log file genserv.log, it was not recognizing the entries you had comments at the end of the line.

Also, selecting Secure HTTP and a custom port is not a valid combination, which also caused you to look for the web server at the wrong address. If Secure HTTP is selected you use the prefix "https://" in the browser address. This is the equivalent of adding port 443 at the end of the address since HTTPs always runs on port 443. 

if secure web setting enabled:

https://192.168.1.50

if secure web not enabled:

https://192.168.1.15:XXXX
where XXXX is the port number you selected in the UI. If you select port 80 then you can leave off the ":XXXX:

You can not mix a custom port number and Secure HTTP.

I am assuming that you have the latest web update now. I would suggest overwriting your /etc/genmon.conf file with the one your recently downloaded in the /home/pi/genmon directory, then restart genmon via the "bash ./restartgenmon.sh restart" command, then make any changes you need from the web UI. Your current genmon.conf file is generating errors from the comments you added.

This is OK
    # adding a comment on a line by itself is allowed
    setting = True

This is OK:
    # setting = True . remove the setting entirely

This is not:
    setting = True   # add a comment to the end of the line will cause an error

These are the errors in your genser.log file:

2018-04-12 16:10:11,685 : Missing config file or config file entries: Not a boolean: false     #****************true**********************************************************
2018-04-12 16:10:11,687 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2018-04-12 16:14:34,506 : Error Reading Config File (ReadSingleConfigValue): Not a boolean: false     #****************true**********************************************************
2018-04-12 16:14:34,548 : Error Reading Config File (unknown type): email_pw : password
2018-04-12 16:16:39,874 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart
2018-04-12 16:16:44,018 : Missing config file or config file entries: Not a boolean: false     #****************true**********************************************************
2018-04-12 16:16:44,020 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2018-04-12 16:16:44,037 : Error in app.run:[Errno 98] Address already in use
2018-04-12 16:16:46,041 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart
2018-04-12 16:16:50,183 : Missing config file or config file entries: Not a boolean: false     #****************true**********************************************************
2018-04-12 16:16:50,185 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2018-04-12 16:17:57,209 : Script: /bin/bash /home/pi/genmon/genmonmaint.sh updatenp
2018-04-12 16:17:59,150 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart
2018-04-12 16:18:03,284 : Missing config file or config file entries: Not a boolean: false     #****************true**********************************************************
2018-04-12 16:18:03,287 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2018-04-12 16:18:03,308 : Error in app.run:[Errno 98] Address already in use
2018-04-12 16:18:05,312 : Script: /bin/bash /home/pi/genmon/startgenmon.sh restart
2018-04-12 16:18:09,578 : Missing config file or config file entries: Not a boolean: false     #****************true**********************************************************
2018-04-12 16:18:09,581 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2018-04-12 16:22:54,200 : Error Reading Config File (ReadSingleConfigValue): Not a boolean: false     #****************true**********************************************************
2018-04-12 16:22:54,243 : Error Reading Config File (unknown type): email_pw : password
2018-04-12 16:38:32,085 : Error Reading Config File (ReadSingleConfigValue): Not a boolean: false     #****************true**********************************************************
2018-04-12 16:38:32,127 : Error Reading Config File (unknown type): email_pw : password
2018-04-12 16:53:24,767 : Missing config file or config file entries: Not a boolean: false     #****************true**********************************************************
2018-04-12 16:53:24,769 : Starting /home/pi/genmon/genserv.py, Port:8000, Secure HTTP: False, SelfSignedCert: True
2018-04-12 16:55:23,539 : Error Reading Config File (ReadSingleConfigValue): Not a boolean: false     #****************true**********************************************************
2018-04-12 16:55:23,585 : Error Reading Config File (unknown type): email_pw : password

jgyates, 2018-04-12:
As you can see all of the errors are due to the comment at the end of the line.

lqallison, 2018-04-13:
Thank you. I will correct this tomorrow actually today. Any chance to make the buttons remote start, green and the remote stop, red and changing brightness according to whether or not the generator is running or stopped?

jgyates, 2018-04-13:
OK. I will close this issue thread.

Regarding your feature request,, no. The web UI already changes color when the the generator is running. The side bar menu changes color based on running, exercising, off, off-ready, alarm, etc. The buttons also change state so you can not click stop if the generator is not running and you can not click "run" if the generator is already running.

closed 2018-04-13 by jgyates