jgyates/genmon #91: Settings screen

krisasmith, 2018-04-13:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

### Expected Behavior

The settings screen would show that the 'Sync Time' setting is 'on'.

### Actual Behavior

The 'Sync Time' setting is set to true in the /etc/genmon.conf file

### Steps to Reproduce (including precondition)

n/a

### Screenshot on This Problem (if possible)

[image]

[image]

### Your Environment

- Genereator Model: {Please write here}
- Generator Registers: {Please write here}
- Genmon Version: {Please write here}

jgyates, 2018-04-13:
Hi Kris,

I can duplicate this, I think it is related to the UI not updating it settings after Save is clicked. If you go to another page after the Save is complete it does not occur. If you  stay on the Settings page and make a second change, it does not appear to update properly. I will look at this and try to get a fix posted. I *think* this bug is related to the other one your submitted (https://github.com/jgyates/genmon/issues/92).

Thanks
Jason

jgyates, 2018-04-13:
I checked in a change. This should fix the problem. I am not forcing a page refresh after settings are applied. This is a bit brute force to fix the issue, another update may be coming that is more elegant, but this should fix it.

jgyates, 2018-04-13:
The cleaner solution is checked in now. No longer forces a refresh but should handle the settings update as expected.

closed 2018-04-13 by jgyates