jgyates/genmon #92: Settings Screen - Outbound Email Settings

krisasmith, 2018-04-13:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

### Expected Behavior

When i update the 'sender account' and 'SMTP Server SSL Enabled' it should update

### Actual Behavior

After making a change, saving, and then refreshing the page it goes back to the original setting.

### Steps to Reproduce (including precondition)

Change the 'sender account' and 'SSL Enabled' , click save, refresh the page and the account is set back to no-reply@gmail.com and SSL is off.

### Screenshot on This Problem (if possible)

Current settings
[image]

My changes

[image]

Click save

[image]

Saving...

[image]

Looks good

[image]

Click on status tab and then back to settings

[image]

Here is /etc/mymail.conf

[image]

As far as i can tell, SSL is enabled (gmail requires it). Is this possibly a permissions problem?

[image]

### Your Environment

- Genereator Model: {Please write here}
- Generator Registers: {Please write here}
- Genmon Version: {Please write here}

jonathanpisarczyk, 2018-04-13:
I wasn't able to get SSL working unless using port 465 with GMAIL.  I suspect gmail only accepts TLS on 587 now.

jgyates, 2018-04-13:
This appears to be related to this issue (https://github.com/jgyates/genmon/issues/91). Looking at both of them now.

jgyates, 2018-04-13:
An update has been checked in to address this. See https://github.com/jgyates/genmon/issues/91

krisasmith, 2018-04-13:
The code change seems to work however i still am unable to update the 'sender account'. The emails are coming from the correct address, the setting just does not reflect that.

krisasmith, 2018-04-13:
I found the problem, i had an older version of the mymail.conf file, i see you added a new parameter (sender_account) that i didn't have in my file. Should i expect that missing parameter to be written by the UI when clicking save?

jgyates, 2018-04-13:
OK. I also found a problem related to the password. The masking of the password was added but this caused th pw not to be save properly, this change is coming in the next few min...

jgyates, 2018-04-13:
I checked in a change to genserv.py. This change correct a problem saving the email password.

jgyates, 2018-04-13:
It should save the missing parameter.

jgyates, 2018-04-13:
I checked in a final update that handles the bug, but does not force a refresh. Let me know if you still have any issues.

closed 2018-04-13 by jgyates