jgyates/genmon #93: New installation - problems with serial communicatios

berenguerj, 2018-04-15:
Hi all, first of all, this seems like a great project, so many thanks for the creator (Jason) and contributors for it. Let me give a little bit of history on how I got here first. I've been around computer for over 30 years and while I'm pretty decent with hardware, mostly PC environment, I started playing with a RPI3 about a year ago. A few weeks later got another one. I purchased a Generac 9kv (model 7030) generator with an Evolution controller a few months ago and wanted to see if there were remote monitoring options. Learned that Generac have one, but find it kind of expensive and while looking into alternatives, found this site. Since I already have the most important piece of hardware, I them acquired the rest of the components that were need based on this site suggestion. Once the required parts arrived (power cable, RS232-TTL adapter, etc) I started playing with the first RPI3 and installed Raspbian lite to emulate the same environment as the author. I got to the point of testing the serial connection but couldn't make it work. Below is my initial problem when running serialtest.py got this:

GENMON Version 1.6.1
Since this is the starting, I'm making a loop in the tx-rx GPIO ports.

**pi@GeneracRPI3:~/genmon/OtherApps $ sudo python serialtest.py
Loopbacktesting for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to receive data ....
FAILED: Sent data does not match received. Received 0 bytes**

I'm doing the loopback test at the GPIO header.

[image]

After reading and making sure I followed every step:
**content on cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=$**

**Last lines on boot.txt
dtoverlay=pi3-miniuart-bt
enable_uart=1
core-freq=250**

**Serial port listing
pi@GeneracRPI3:/boot $ sudo ls -l /dev/serial0
lrwxrwxrwx 1 root root 7 Apr 15 11:10 /dev/serial0 -> ttyAMA0**

I'm curious as to why I did not get /dev/serial1 -> ttyS0

I also checked the voltage on the GPIO TX-RX and both measured about 3.5v DC.

So using the GPIO this is as far as I got. 

I tested also with a Trendnet USB-Serial cable (model TUS9) and was able to get a successful lop connection up to the molex connector, but is not communicating with the Evolution controller. 

PS. I tried the same steps with the 2nd RP3 with same results, so It is unlikely the GPIO port is defective in both units, so I may have missed something.

Thanks in advance for any help you may provide.

jgyates, 2018-04-15:
Hi @berenguerj 

In your boot.txt, change this line:
     
     dtoverlay=pi3-miniuart-bt

to this:

     dtoverlay=pi3-disable-bt

The BlueTooth and on board serial port share the same hardware so you have to disable BT.

ttyAMA0 is the internal name they raspberry pi folks started using along the way for some reason, I think it started with the 2B or 3, I can't remember..

berenguerj, 2018-04-15:
Hi @jgyates 

Thanks for the fast reply!

After the change and a reboot of the PI, now got this:

**pi@GeneracRPI3:~/genmon/OtherApps $ sudo python serialtest.py
Loopback testing for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to received data....
error communicating...: device reports readiness to read but returned no data (device disconnected or multiple access on port?)**

I googled it but so far nothing. Any comments?

jgyates, 2018-04-15:
Now that you have disabled BT, what does the /dev/serial* show? What is the output of this command?
  
     ls –la /dev/serial*

It is possible that the ordered device number of the serial port is serial1 now instead of serial0. You want to use the port that is associated with ttyAMA0

berenguerj, 2018-04-15:
Seems it remained the same. See below:

**pi@GeneracRPI3:~/genmon/OtherApps $ sudo ls -l /dev/serial*
lrwxrwxrwx 1 root root 7 Apr 15 12:56 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Apr 15 12:56 /dev/serial1 -> ttyS0**

jgyates, 2018-04-15:
I have seen an issue like this before. I think the problem is that due to the order that things were disabled the pi is not trying to use the serial port for login now that BT is disabled.  This page has (https://elinux.org/RPi_Serial_Connection) mentions looking in the /etc/inittab file and commenting out the lines that mention ttyAMA0. The /etc/inittab file will launch the getty program to try to enable login via the serial port.

berenguerj, 2018-04-15:
I will look into it and provide feedback on my findings. Thanks Jason!

berenguerj, 2018-04-15:
I haven't perform any more changes, but wanted to see if there would be different results if I ran the serialtest.py a couple more times, and got these results. They took a little longer than my previous loopback tests:

**pi@GeneracRPI3:~/genmon/OtherApps $ sudo python serialtest.py
Loopback testing for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to received data....
error communicating...: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

pi@GeneracRPI3:~/genmon/OtherApps $ sudo python serialtest.py
Loopback testing for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 284 bytes

pi@GeneracRPI3:~/genmon/OtherApps $ sudo python serialtest.py
Loopback testing for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 186 bytes**

I will continue reading from the link provide, just wanted to shared this in case is relevant.

jgyates, 2018-04-15:
This article has some items I have never seen before related to this:
https://spellfoundry.com/2016/05/29/configuring-gpio-serial-port-raspbian-jessie-including-pi-3/

specifically the section on Disabling the Console

berenguerj, 2018-04-16:
Hi @jgyates 

I think I may have found where the problem is, at least where it starts. I have re-installed rasbian lite a few times already. On one of your comments you mentioned something about the order when things are disabled could affect the port, so I went and installed it again, paying special attention to the steps order as per the other document you suggested for serial communication. This time, right after I configured the serial port and it all looks as you specified in _sudo ls -l /dev/serial*_, I ran _sudo python serialtest.py_ and it said something was missing, so I continue the installation with the automatic script you mentioned and once it finished, I ran again the serialtest.py loopback test with success this time. I rebooted the RPI3 one more time to test and the notice that the loopback started failing again. This told me that there is something in the script that went active after the reboot and caused some kind of conflict making the loopback to fail. Next test was doing the manual installation instead if using the script to identify where exactly was the problem. 
It seems to be related to the startgenmon.sh script. After I finished the manual installation, noticed that once I started startgenmon.sh the serialtest.py failed.

The first time it ran, was right after the RPI3 was reboot, but I set crontab to start stargenmon after 30 seconds, so I would have time to start a serial test before startgenmon was load. Notice how it fails when startgenmon is active and how it passed once I stop the process...

_**pi@GeneracRPI3:~ $ sudo python /home/pi/genmon/OtherApps/serialtest.py
Loopback testing for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful

pi@GeneracRPI3:~ $ sudo python /home/pi/genmon/OtherApps/serialtest.py
Loopback testing for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to received data....
error communicating...: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

pi@GeneracRPI3:~ $ cd genmon

pi@GeneracRPI3:~/genmon $ ./startgenmon.sh stop
Stopping genmon python scripts

pi@GeneracRPI3:~/genmon $ sudo python /home/pi/genmon/OtherApps/serialtest.py
Loopback testing for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful

pi@GeneracRPI3:~/genmon $ sudo python /home/pi/genmon/OtherApps/serialtest.py
Loopback testing for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful

pi@GeneracRPI3:~/genmo./startgenmon.sh start
Starting genmon python scripts
pi@GeneracRPI3:~/genmon $ WARNING: System Initializing. Not receiving data. : ,
 * Running on http://0.0.0.0:8000/ (Press CTRL+C to quit)

sudo python /home/pi/genmon/OtherApps/serialtest.py
Loopback testing for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to received data....
error communicating...: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

pi@GeneracRPI3:~/genmon $ sudo python /home/pi/genmon/OtherApps/serialtest.py
Loopback testing for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to received data....
error communicating...: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

pi@GeneracRPI3:~/genmon $ ./startgenmon.sh stop
Stopping genmon python scripts

pi@GeneracRPI3:~/genmon $ sudo python /home/pi/genmon/OtherApps/serialtest.py
Loopback testing for serial port /dev/serial0...
Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful**_

What do you suggest I can do next?

jgyates, 2018-04-16:
Hi @berenguerj 

The items that was missing when you first ran the test was the python serial library support (python-serial). Both genmon and serialtest use this library as it is the standard method for accessing the serial port with python.

Regarding the test failing if you run startgenmon.sh, this is expected behavior. When genmon.py is running it is using the serial port so serialtest.py will fail. The serial port will only allow one application to access the underlying hardware at at time. 

The order of setting up genmon should be this:
1) setup your pi so it will boot
2) make the changes to raspbian to enable the serial port
3) install the needed libraries
4) test the serial port and cable with a loopback
5) run startgenmon.sh to start genmon

I can tweak the wiki to make this a little more obvious since the serial and software install are on different pages. Let me know if you have any other questions.

berenguerj, 2018-04-16:
Makes sense. On a PC environment you can't share a IRQ or Serial. 
I was doing the loopback test from the GPIO. I will test with the rest of the components (RS232/TTL, Molex, etc) to make sure is all worknig before it goes into the generator and will post results. Thanks for the help!

closed 2018-04-16 by jgyates

berenguerj, 2018-04-19:
Just as a FYI...Got it working now...Received the 8 pin molex yesterday and installed right away as a permanent solution. All Puerto Rico suffered a power outage since yesterday morning  and was able to test in the afternoon. Blackout should last another 12 hrs as per the news. I'm still browsing trough the issues forum for additional information on a few more things I would like to tweak.