jgyates/genmon #94: Connection refused at port 9802

tmorse305, 2018-04-15:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

### Expected Behavior

I am using Automation Shack's NodeLink program to interface to genmon via port 9802.  I have checked the genmon config file and confirmed the default value of 9802.  9802 is the default on the NodeLink side.

### Actual Behavior

genmon refuses the connection, the error on the NodeLink side is: "2018-04-15 17:45:41 - GenMon Socket Error: Connection refused. [genmon1]'.  Is there additional setup that I need to do to enable communication via 9802?  The web port at 8000 is working perfectly

### Steps to Reproduce (including precondition)

N/A

### Screenshot on This Problem (if possible)

N/A

### Your Environment

- Genereator Model: 58751 (20KW)
- Generator Registers: {Please write here}
- Genmon Version: I'm not sure how to tell but I downloaded it in the past few days

Thanks for any help you can offer.  And thanks for writing and sharing such a great program!

jgyates, 2018-04-15:
I would double check the IP address that NodeLink is using for the generator monitor. If you have the wrong IP you will get this error. If the web interface for the generator software is working then port 9802 is functioning.

tmorse305, 2018-04-15:
Thanks for your reply.  I double checked the IP address and it is correct.  Any other ideas?  BTW way the port is 9082, I had it written incorrectly.

jgyates, 2018-04-15:
I am assuming these two systems (genmon and nodelink) are on two separate system. 

1) Can you ping genmon from the nodelink system? If this fails you have an issue with your network (Cable, VLAN, WiFi or something like that)
2) If you can ping, then if you  put ClientInterface.py, myclient.py and mylog.py on the nodelink system you should be able to run ClientInterface.py from that system to validate if comms are working for the IP and port.

     python ClientInterface.py IPaddressOfPi

If the above 2 tests work then the problem is somewhere on the NodeLink side. Otherwise it looks like a problem with the network. 

If ping works, but test 2 does not, let me know and we can go from there.

tmorse305, 2018-04-15:
Hi, the ping was successful so I installed the 3 files you requested.  When I ran ClientInterface.py (from Putty) on the NodeLink machine  I received a warning "Not receiving data. : ,"  but a > prompt.  So I typed in "Generator: status" and it returned the status.  So it points back to the NodeLink side I guess.  Thanks for your help!

jgyates, 2018-04-15:
OK. The warning about not receiving data is the watchdog for serial communications. If you started genmon within a few seconds or your serial cable is unplugged you will get this message (or if you have a problem with your serial port). Once ClientInterface is running you can type "generator: monitor" and it will give you the updated comm status under "Monitor Health".

tmorse305, 2018-04-16:
Thanks, so just to be clear the ClientInterface application uses port 9082?  I want to make sure I'm giving the right information to the NodeLink developer.

jgyates, 2018-04-16:
Correct, this is set in line 15 of myclient.py (https://github.com/jgyates/genmon/blob/master/myclient.py#L15)

tmorse305, 2018-04-17:
Thanks again, tip sent via PayPal

jgyates, 2018-04-17:
No problem. Thanks for the tip! I will close this thread. Feel free to re-open or comment if you have any other issues.

closed 2018-04-17 by jgyates