jgyates/genmon #95: Automatic feedback address - information needed only

steve-bailey1, 2018-04-16:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

### Expected Behavior

email the software team the feedback notices

### Actual Behavior

none at the moment.  Need to know the email domain of the target address to allow relaying via my mail servers

### Steps to Reproduce (including precondition)

turning on the auto feedback option

### Screenshot on This Problem (if possible)

### Your Environment

- Genereator Model: {Please write here}
- Generator Registers: {Please write here}
- Genmon Version:  1.6.2

jgyates, 2018-04-16:
the email domain is gmail.com. The full email address used is generatormonitor.software@gmail.com

steve-bailey1, 2018-04-16:
Thank you.  I just turned it on and it looks like it went through.

closed 2018-04-16 by steve-bailey1