jgyates/genmon #96: Making a Cable?

steven-pozarycki, 2018-04-17:
Any suggestions here would be welcome.

The hardware side... I just got a new 22kw Generac generator which has an Evolution Controller. 
Model G0070431, Production Date 20171129 Controller PN is listed as well (not sure if that is important)

I followed the guidance on https://github.com/jgyates/genmon/wiki/3.1:-Making-a-Cable to make the cable and wanted a second set of eyes in case I missed something.
I wanted to use pin 1 and pin 2 on the molex to power the Raspberry Pi however that doesn't seem like it is working.
Picture 1: Connecting from my Raspberry Pi to the RS232 to TTL - I have a 9 pin femail and connecting to the RS232 Male.
	Power (Blue) going to GPIO-4
	Ground (Orange) going to GPIO-6
	TX (Yellow) going to GPIO-8
	RX (Green) going to GPIO-10

Picture 2 and Picture 3
	RS232 (like the DCE setup in the diagram)
		Ground - Pin 5 connecting to Molex in position 2
		Power - Pin 1 connecting to the Molex in position 1
		RX - Pin 2 connecting to the Molex in position 7
		TX - Pin 3 connecting to the Molex in position 8

Any thoughts?
Thank you!
-Steven
[image]
[image]
[image]

jonathanpisarczyk, 2018-04-17:
Check out this thread:
https://github.com/jgyates/genmon/issues/74

This picture of my setup:
https://user-images.githubusercontent.com/36710072/36503481-43d26680-171b-11e8-8a54-63fc4bc9891a.jpg

The power connection should go directly to the PI pin 2 & 14 from the molex.  Not to the RS232.

jgyates, 2018-04-17:
Hi @steven-pozarycki 

Based on your pictures it appears that you are running power from the controller thru the DB9. This is not correct. Power from the controller should go directly to the GPIO header. See this diagram.

https://raw.githubusercontent.com/jgyates/genmon/master/Diagrams/wiring_schematic.png

The way you have it wired now you are not supplying power to you pi. The pin on the RS232 board labeled VCC is an input from the pi, not an output to the pi.

madisdev, 2018-04-17:
Just a suggestion, It's hard to see a complete picture with your cable color choices. I would alternate yellow/black with +/- and TX/RX.

steven-pozarycki, 2018-04-17:
Thanks, I appreciate it (first time doing the hardware side of things!).

I saw that one thread #74 where it was mentioned that it could draw more from the controller and potentially have an issue... would an alternative be to just power off of USB as also suggested by using this (https://www.amazon.com/gp/product/B01DYE54LI/ref=ox_sc_act_title_2?smid=A3RYN903ALGZTW&psc=1) and a 3 foot USB-to-USB cable like this one https://www.amazon.com/gp/product/B003VTZ070/ref=ox_sc_act_title_1?smid=ATVPDKIKX0DER&psc=1

Thank you all again!

proppilot, 2018-04-17:
save the cable and get this

https://www.amazon.ca/gp/product/B00QTJWRFW/ref=oh_aui_detailpage_o07_s00?ie=UTF8&psc=1

attach to the battery and plug into the PI

jgyates, 2018-04-17:
Each power source and cable combination have a few very minor pros and cons each. It boils down to personal preference really. Some folks like the power from the controller, some go the battery method  with the DC to DC converter. I went with the [this](https://www.amazon.com/gp/product/B01DYE54LI/ref=ox_sc_act_title_2?smid=A3RYN903ALGZTW&psc=1) simpley becaue it was easy and it had a built in fuse, but the DC to DC converter will do the trick as well.

steven-pozarycki, 2018-04-17:
Great, thanks so much to everyone!

closed 2018-04-18 by jgyates