jgyates/genmon #100: Dashboard Verification?

steven-pozarycki, 2018-04-20:
Hi -
I believe I have the cable working and getting results to my webpage. 

Question: Are the stats for the monitoring page showing the "M" and "S" packet numbers increasing good?  I am guessing yes since it does show my serial number as correct and the exercise time every Saturday at 10am.

Screenshots below:
[image]
[image]
[image]

jgyates, 2018-04-20:
Yes, your output looks good. The M and S are for the Master (Raspberry Pi) and Slave (Generator Controller) packet count. These should increase as time goes on. If they do not then you will get an error (i.e. Monitor Health will not be "OK"

jgyates, 2018-04-20:
With minor mods to the genlog.py program you can log any text output. The current genlog.py program only logs the base status (running, stopped, ready, exercising, etc) when the status changes.

Changing the logic in the while loop to call this, for example:

     data = MyClientInterface.ProcessMonitorCommand("generator: status")

would log the status at the interval of your choosing.

closed 2018-04-21 by jgyates