jgyates/genmon #99: Login Failed at mymail.log

berenguerj, 2018-04-20:
Hi, I'trying to get the email configuration work, but keep getting Login Failed error within the mymail.conf. I triple checked my user and credentials and they are both correct. Googled the link provided in the log ( **https://support.google.com/mail/?p=BadCredentials**) and found something that kind of make sense to me, but then, all users could be facing the same issue. Part of the link mentioned that is weak applications are used to access gmail, they may fail due to double authentication method used by gmail. An option they don't recommend would be to decrease the authentication method, but that would result in weaker protection against the gmail account. See more info below...
Do I missed something?

### Expected Behavior

receive email notifications using my gmail account

### Actual Behavior

Getting this error in mymail.log

**2018-04-19 21:13:25,765 : Error SMTP sendmail: (535, '5.7.8 Username and Password not accepted. Learn more at\n5.7.8  https://suppo$
2018-04-19 21:13:25,857 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2018-04-19 21:14:29,580 : LOGIN FAILED!!!
2018-04-19 21:15:27,381 : Error SMTP sendmail: (535, '5.7.8 Username and Password not accepted. Learn more at\n5.7.8  https://suppo$
2018-04-19 21:15:27,476 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2018-04-19 21:15:36,094 : LOGIN FAILED!!!**

### Steps to Reproduce (including precondition)
I tried using both SSL on and off with same results. 
{Please write here}

### Screenshot on This Problem (if possible)

My current mymail.conf file @  /etc
**[MyMail]

# optionally disable email use by the program. set to true to disable all email functionality
disableemail = false

# The password for the email account used for sending and receiving email
email_pw = **************

# the email account
email_account = ****************@gmail.com

# address that email are sent to
email_recipient = *********************@yahoo.com,***************@gmail.com

# the email from whichthe emails are sent (sender email address)
sender_account = no-reply@gmail.com

# the SMTP mail server used for sending email, leave emtpy to disable SMTP, i.e. "smtp_server ="
smtp_server = smtp.gmail.com

# the IMAP server address used for receiving email, leave emtpy to disable IMAP, i.e. "imap_server ="
imap_server = imap.gmail.com

# the SMTP port used on the SMTP server for sending email
# Note about ports and email security: typically port 587 is used for
#  TLS and port 465 is used for SSL
smtp_port = 587

# use SSL encryption to send emails
ssl_enabled = false**

### Your Environment

- Genereator Model: 7030
- Generator Registers: ??????
- Genmon Version: 1.6.2

jonathanpisarczyk, 2018-04-20:
Try using port 465 and enable ssl.

berenguerj, 2018-04-20:
Hi @jonathanpisarczyk 
That was my original setting and changed it to SSL turned off. I just set it up again with SSL as your recommendation, with the results below:

2018-04-19 21:58:25,587 : LOGIN FAILED!!!
2018-04-19 21:58:28,862 : Error SMTP sendmail: (535, '5.7.8 Username and Password not accepted. Learn more at\n5.7.8  https://support.google.com/mail/?p=BadCredentials g16-v6sm2899521oiy.13 - gsmtp')
2018-04-19 21:58:28,965 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2018-04-19 21:59:26,594 : LOGIN FAILED!!!

JustApu, 2018-04-20:
Do you have an "App Password" set in GMail or, if not, have you enabled "Allow less secure apps"?

berenguerj, 2018-04-20:
Hi @JustApu 
I just added the RPI3 into the App Password, and change the gmail provided password into the web configuration screen, but still fails. If I can prevent using the Allow less secure apps option, it will be better.

2018-04-19 22:22:53,423 : LOGIN FAILED!!!
2018-04-19 22:22:56,678 : Error SMTP sendmail: (535, '5.7.8 Username and Password not accepted. Learn more at\n5.7.8  https://suppo$
2018-04-19 22:22:56,767 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2018-04-19 22:24:05,266 : LOGIN FAILED!!!

berenguerj, 2018-04-20:
I just close the thread by mistake.

berenguerj, 2018-04-20:
Ok, found it. When I C&P the gmail provided password from App Password, #01 was added at the end of the text. I manually removed it from /etc/mymail.conf and just received two emails... Thanks for pointing me into the right direction !!

closed 2018-04-20 by berenguerj