jgyates/genmon #102: Time difference between RPi and Evolution controller

berenguerj, 2018-04-26:
Hi all,

I have notice that there is around a minute difference between the RPi3 and the Evolution controller on my generator. 

### Expected Behavior

Sync time between RPI3 and Evolution controller

### Actual Behavior

Around one minute difference where generator time is always one minute ahead of the RPI3 

### Steps to Reproduce (including precondition)

Send time from the web interface, email command and rebooting the RPI3 with always the same results.

### Screenshot on This Problem (if possible)

Time :
            Monitor Time : Wednesday April 25, 2018 **21:59:22**
            Generator Time : Wednesday April 25, 2018 **22:00**

### Your Environment

- Genereator Model: 7030 (9kw air cool Generac)
- Generator Registers: **How do I get the registers?**
- Genmon Version: 1.6.6

jgyates, 2018-04-26:
Hi @berenguerj 

I don't know of a way to set the seconds of the generator time so they way the logic in the code works is that when a request to set the time is issued the software will wait until the system time changes to a new minute, then it will set the generator time. This will result in always being off by some time equal to a 60 seconds or less. The generator also does not report the number of seconds in it's time so waiting for that to turn over will not work either.

Regarding the registers, if you use the program ClientInterface.py you can get a dump of your registers. The Clientinterface program is documented here (https://github.com/jgyates/genmon/wiki/1:-Software-Overview#clientinterfacepy)

    python Clientinterface.py <IPaddress of your PI>

They you will get a prompt, from the prompt type:
    generator: registers

Copy and paste your output and you will have your registers.

If you have email setup to receive commands you can just put "generator: registers" in your subject line and they will be emailed to you.

berenguerj, 2018-04-26:
Hi @jgyates 

Out of curiosity, and understanding that 1 minute is not a big deal. Question, when the app receives the command to sync the time, lets say, at 9h:18m:30s and you will wait until the seconds are 00, do you sent the controller 9h:18m or 9h:19m ?

On the registers, something happened last night with the email settings that, for some reason the app stop processing the commands. They are on the Generator folder in gmail, but they are not processed. I will open another thread for this to see if someone can provide feedback on what to look

jgyates, 2018-04-26:
current system time. So in your example it would be 9:19:00.

https://github.com/jgyates/genmon/blob/master/genmonlib/generac_evolution.py#L1086

closed 2018-04-26 by jgyates