jgyates/genmon #103: Commands sent via email stopped working after incorrect command

berenguerj, 2018-04-26:
Last night I started sending some command via email to see the results and in one instance, I sent the command **generac: status** instead of **generator: status**. After that I started receiving lots of emails saying error in command:

**Time: 08:37:09 PM**
**Date: Wed 25-Apr-2018**

**Invalid GENERATOR command: all commands must be prefixed by "generator: "**

I deleted the incorrect command from the Generator folder and Generator\Processed folder in the hopes of getting rid of the many emails sent\received but they keep appearing. Trying to correct this behavior, I stop the app process and clear the email again, wait about 15 minutes in case there where any other delayed emails, and once everything stop, reboot the RPI and to my surprise, the avalanche of emails started again. after several retries emptying the Generator and Generator\Process folder the problem stopped, but also the ability to send commands via email. 

Any thoughts?

- Genereator Model: Generac 7030, (kw Evolution controller
- Generator Registers: {Please write here}
- Genmon Version: V1.6.6

jgyates, 2018-04-26:
@berenguerj 
Do all of the emails that you receive have the same timestamp "Time: 08:37:09 PM"?

An easy thing to try updating to the latest version. You can do this via the web interface one the monitor screen or by logging into the pi and in the genmon directory type:

     bash ./genmonmaint.sh update

I have never seen this issue before, but there we some changes to how emails are handled on 4/23 (version 1.6.6) and there have been a few updates since then so there is a chance that the update to 1.7 will fix your problem. If it does not let me know.

jgyates, 2018-04-26:
I tried to duplicate this on V1.7 and was not able to see the problem.

berenguerj, 2018-04-26:
Hi @jgyates 
They have incremental times, based on the times received and/or generated. 

[image]

I'll update to the v1.7 once I get home from work and will let you know how it went. Thanks!

jgyates, 2018-04-26:
I updated the subject of the message and checked that change in. Let me know if you still have an issue.

berenguerj, 2018-04-26:
I found out what caused the email commands to stopped working. I looked into the mymail.log and noticed the email app password changed into the one I set originally. Since it wasn't an app password, it was failing the login process. I'm thinking that at some moment, the browser, may have that one cached and caused the change during one of the times I pressed the Save button. 
Once it was set back to an app password, it started communicating with the gmail account. Again the many error emails originally described on my first post started flowing 👎 . 

Will continue looking why this happens and keep everyone updated. 
BTW, I also update the software to v1.7

jgyates, 2018-04-26:
Are you using the same email for both send(email account ) and receive (notifications)? If you are then, depending on your email filter settings, you would see the repeated emails. Make sure you filter on "generator:" and not "generator" or every email sent by the software will get forwarded back, then the software will respond with "Invalid GENERATOR command: invalid prefix", which will also get sent back by the filter,.....

jgyates, 2018-04-27:
If you have any emails in your Processed or Generator folder that are sent from genmon (i.e. Invalid GENERATOR command: all commands must be prefixed by "generator: ") this is caused by one of your email accounts forwarding to your gmail account. The email send and receive threads in the code are completely separate. To validate this you can send two emails:

1) Stop genmon by running:
      bash ./startgenmon.sh stop
2) Send an email from your gmail account to your yahoo account. Include the word generator in the email to make sure if filtering is and issue we test that as well. (Something like "Generator Test").
3) Send an email from your gmail to your gmomail.net account. Include the word generator in the email (Something like "Generator Test").

If either of these emails show up in your gmail in box, or Generator or Generator/Processed mail folders then then the problem is a mail forwarding issue since genmon is not running.

berenguerj, 2018-04-27:
I've been doing some test with the gmail filters/rules and I add an additional condition to the **generator:**  in the subject line. The other condition is to compare the From with my yahoo account. I have sent several emails and so far, no errors 😄 

Below are the conditions:

**Matches: from:(xxxxxxxxxxxxxxxxxx@yahoo.com) subject:(generator:*)
Do this: Apply label "Generator"**

This is preventing the gmail process of any emails not coming from my yahoo account.

I have test with all conditions with the exception of setexercise and settime, and they all seem to work. Thanks for your help.

jgyates, 2018-04-27:
no problem

closed 2018-04-27 by jgyates