jgyates/genmon #105: Question: raspbian stretch

krisasmith, 2018-04-27:
Is anyone running genmon on raspbian stretch? I have been running jessie for a while now without any issues. Today I deiced to rebuild my pi (using a new sd card) on stretch and had nothing but problems. The serial test would randomly succeed or fail. I ran the test a few dozen times. it would fail, then pass, then throw an error about the port being in use and then fail some more and then pass. genmon would not communicate with the controller at all. I use the gensms feature and got a bunch of SSL errors connecting to api.twilio.com. I gave up after about 2 hrs and put my original sd card back in and everything works without an issue.

jgyates, 2018-04-28:
I have not tried this yet. I was going to test this a pi3B+, but I heard that raspbian stretch was not without problems on the 3B+ so I decided to hold off. I would assume it would be stable on the Pi3.

krisasmith, 2018-04-28:
Interesting you mentioned pi 3 b+ has stretch problems. I was having and issue with it booting, the screen would flicker and hang. After I disconnected the serial from the gpio header it would boot. All around it was a total mess. I have another pi running openvpn on stretch without any obvious issues.

lqallison, 2018-05-02:
I am running stretch on my Pi. Have had no problems with it so far. This one is a March 13th kernel. Also have the April 18th kernel running on a 3B and 3B+. This 3B has Genmon and is in my garage and has a poor wifi connection. Has lost connection twice just today. Did not reconnect on its own.

jgyates, 2018-05-03:
Here is a data point: I just took a new Pi 3B+, new SD card, new power supply, installed stretch and genmon (not hooked up to generator yet). Serialtest.py appears to work well. My plan is to swap out my pi in the next few days.

kalex1114, 2018-05-07:
I'm running stretch lite on my pi zero w but haven't hoooked up generator cable yet.  Was doing serial test and had similar results.  where it would fail, fail, pass

Here is a sample after running it just now:

`pi@genmon:~/genmon/OtherApps $ sudo python serialtest.py 

Loopback testing for serial port /dev/serial0...

error open serial port: Could not configure port: (5, 'Input/output error')
Error opening Serial Port /dev/serial0
pi@genmon:~/genmon/OtherApps $ sudo python serialtest.py 

Loopback testing for serial port /dev/serial0...

error open serial port: Could not configure port: (5, 'Input/output error')
Error opening Serial Port /dev/serial0
pi@genmon:~/genmon/OtherApps $ sudo python serialtest.py 

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful
pi@genmon:~/genmon/OtherApps $ sudo python serialtest.py 

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 11 bytes
pi@genmon:~/genmon/OtherApps $ sudo python serialtest.py 

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 2 bytes
pi@genmon:~/genmon/OtherApps $ sudo python serialtest.py 

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful
`

lqallison, 2018-05-08:
I think the 3B+ took away the uart the 3B used on the serial port. Now the serial port is handled by the processor bit by bit. The uart is now used for the bluetooth connection. Heavy processor use could affect the handling of serial data.

kalex1114, 2018-05-09:
I’m running stretch lite on pi zero W

closed 2018-05-20 by jgyates

krisasmith, 2018-05-30:
I got genmon working on a pi 3b+ with stretch. Turns out I can't read and didn't update /boot/cmdline.txt to remove “console=serial0,115200"

kalex1114, 2018-05-30:
Yep same issue for me (facepalm)

jgyates, 2018-06-01:
I added a program to /genmon/OtherApps named serialconfig.py.

This program will check or set the needed serial port settings in the /boot/config.txt and /boot/cmdline.txt files.

To check your settings type:

     sudo python serialtest.py -c

Usage: 
python serialconfig.py [-r -e -c]
       
Example: 

    sudo python serialconfig.py -e
                       
     sudo python serialconfig.py -c

Command line options
      -e  Enable serial port
      -r  Restore modified files
      -c  Check status of serial port