jgyates/genmon #106: Raspberry Pi - under voltage

krisasmith, 2018-04-27:
Anyone seeing under voltage on their pi when powering off the generator controller though the GPIO header? I just realized its a problem for me.

madisdev, 2018-04-27:
How is this issue coming up?

krisasmith, 2018-04-27:
I was trying to rebuild by pi with raspbian stretch (a whole other topic) and had connected a monitor and saw the warnings. Also if you run the cmd: **vcgencmd get_throttled** it returns **throttled=0x50000**. I disconnected the power from the GPIO header (left the ground) and then powered off the microusb port and the warnings went away.

madisdev, 2018-04-27:
I have not heard of that command before. I ran it on my Pi Zero W and it returned 0x0

jgyates, 2018-04-28:
Hi Kris,
I assume you are using the Pi3, which has a higher current draw than than the PiZero. I don't think we know the max current supplied by the evolution but it is a possibility we may be reaching it. Not sure if you are using Wifi or Ethernet but Wifi will significantly increase your current. I have not seen this before in past issues, but it could be a real problem. Where the warnings that alerted you to the problem just messages in the system log?

krisasmith, 2018-04-28:
I am on the pi 3 b+ with wifi disabled using the btoverlay. I could not find any log that showed the problem. I only came across it with a HDMI connection. It has a lighting bolt in the upper right corner and messages on the console about under voltage.

krisasmith, 2018-04-28:
I ordered this https://www.amazon.com/gp/product/B01MDLUSE7/ and paired it with a PoE switch. I no longer get under voltage alerts on the console and **vcgencmd get_throttled** returns **throttled=0x0**

jonathanpisarczyk, 2018-04-30:
I'm running the power from the Evolution controller and appear to be getting under voltage as well.  However everything is working fine still.  

pi@genmon:~ $ vcgencmd get_throttled
throttled=0x50005

The bits represent:
0: under-voltage
1: arm frequency capped
2: currently throttled 
16: under-voltage has occurred
17: arm frequency capped has occurred
18: throttling has occurred

I may switch to the usb battery option in the near future.

krisasmith, 2018-04-30:
I had considered powering directly off the battery also, it would have been more work. I have my pi remotely mounted in my garage next to the transfer switch / electric panel. I have been running powered off the controller for a while without any issues so I don't see it as a big rush to fix. I changed it only because of the way my brain works, I would loose sleep over it. Its a bit of a sickness ;)

jgyates, 2018-04-30:
I added a note about this in the wiki and reference this thread.

NOTE: Some people have reported undervoltage errors on the Pi 3 (not the Pi Zero) when powering through the controller connector. While this has not been reported to prevent the application from running it may not be an ideal way to power the Pi. See [this](https://github.com/jgyates/genmon/issues/106) issue for more details.

madisdev, 2018-04-30:
It feels like the power req's for the PI3 with external Wifi is a bit more than the PI Zero with built in WiFi.

krisasmith, 2018-04-30:
Agreed, i have WiFi and Bluetooth disabled with a btolverlay and it still power hungry.

madisdev, 2018-04-30:
Take this for what you will. http://raspi.tv/2017/how-much-power-does-pi-zero-w-use

lqallison, 2018-05-02:
Have had several problems with 3B and 3B+ while doing development work. Lighting bolt  to the undervoltage message on command line to the point of shutdown altogether. Shut down occurred at 4.87 v measure at the 5v pin on the header. Moved to power supplies with 5.25v @ 2Amp. Voltage
at the header was then 5.1v. no other problems. For Genmon I run 12v from generator battery thru 30FT of 18 AGW  6 wire cable along with RS232 from the Nexus controller into change over switch and out to Raspberry PI thru a motorcycle 12v to USB 5V adapter.  No problems here either. Have not measured the voltage at Pi header though.

jgyates, 2018-05-22:
I have added a program in ./genmon/OtherApps/throttled_check.py. This program will display and monitor the status of throttling on the pi.

to use

     python throttled_check.py

To exit type Ctrl+c

jgyates, 2018-05-24:
FYI, i started having and under voltage problem that I noticed a few weeks ago. I had moved the placement of my enclosure a while back and and I had to extend the USB cable.  Today I replaced the cable with one of these:

https://smile.amazon.com/gp/product/B017VQ3AWU/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1

No under voltage. Of course you milage may vary, but cables do matter. I am powering the pi from the battery.

jgyates, 2018-05-26:
FYI, if the throttled stats are not showing up on the Monitor page you will need to update your raspberry pi firmware. To do so first make sure you are running the latest software / kernel.

     sudo apt-get update
     sudo apt-get dist-upgrade
     sudo rpi-upgrade
     sudo restart

The "sudo apt-get update" and " sudo apt-get dist-upgrade" update the system software.
The  "sudo rpi-upgrade" will update the firmware in the raspberry pi
The "sudo restart" will restart your pi so the updated software can be used.

jgyates, 2018-05-26:
sorry, fixed error in my last post "sudo reboot" will restart your pi. "sudo restart" does nothing.

closed 2018-06-03 by jgyates