jgyates/genmon #107: kW Output Stays Excessively High after Large Load Hits Generator

alphaa211, 2018-05-01:
### Expected Behavior

kW Dial, Graph, and Output Current Line change appropriately with current output/load on generator after Start & Transfer Command.

### Actual Behavior

kW Dial, Graph, and Output Current Line appear to be fine after start up, fine during transfer with light load, and when 24 second HVAC compressor timer expires and exterior AC load hits generator all 3 go exceedingly high and stay.  They go and stay far higher than the possible output of the generator while local live monitoring shows the actual load on the unit at 19-20 (220v) amps.  This load hit makes the generator bog for a second, the lights dim, and then it revs up to provide correct output.

Note: Start & Transfer can cause a slight out of phase transfer between the active utility power and the generator power which is different than a transfer from no utility power to generator power.  I haven't tried simulating a power outage where the AC compressor has longer to unload and takes less power to start to see if the kW output does the same thing.  I'll try it tomorrow.    

### Steps to Reproduce (including precondition)

Start & Transfer with previous Genmon Version from 2 days ago - same effects noted.
Start & Transfer with most current Genmon Version update 2 hours ago - same effects noted.
Start & Transfer multiple times watching AC compressor load hit and monitors go high - repeated.

### Screenshot on This Problem (if possible)

Start & Transfer after house load has been transferred to generator and before HVAC compressor load hits: (latest Genmon Version)

[image]

Start & Transfer after HVAC compressor load has hit: (latest Genmon Version)
 
[image]

Start & Transfer after HVAC compressor load has hit: (2 day prior Genmon Version)

[image]

### Your Environment

- Genereator Model:  G0070330

- Generator Registers: 
Registers :
    Num Regs : 220
    Not Changed : 92468
    Changed : 2840
    Total Changed : 0.03

    Base Registers :
        0239 : 0000
        0039 : 0000
        0038 : 0000
        0034 : 0000
        0037 : 117c
        0036 : 0000
        0237 : 0000
        0033 : 0000
        0032 : 4000
        05f7 : 05c8
        05f6 : 05ad
        05f5 : 0000
        05f4 : 0000
        05f3 : 00c8
        05f1 : 0000
        05ed : 0064
        023b : 0000
        003a : 0000
        023e : 0000
        003b : 0000
        05fa : 0000
        0059 : 0000
        0058 : 0000
        0053 : 0000
        0052 : 0000
        0057 : 0000
        0056 : 0000
        0055 : 0000
        0054 : 0000
        0208 : 03f8
        005c : 0000
        005a : 0000
        005f : 0000
        005e : 0000
        005d : 0000
        003c : 0000
        000d : 0000
        000e : 1507
        000f : 041e
        000a : 0086
        000b : 0000
        000c : 0036
        0005 : 111e
        0006 : 0200
        0007 : 0000
        0000 : 0009
        0001 : 00000000
        0008 : 0000
        0009 : 00f7
        05ee : 0155
        0012 : 0000
        0011 : 00ab
        0010 : 0112
        0019 : 000f
        002b : 0005
        002c : 111e
        002a : 6a74
        002f : 0000
        002d : 0000
        002e : 0002
        0022 : 463b
        0020 : 0000
        0021 : 0000
        001f : 4915
        001e : 0159
        001d : 0000
        001c : 0000
        001b : 463b
        001a : 0091
        01f4 : 33303030373839363736

- Genmon Version: V1.7.4

jgyates, 2018-05-03:
I checked in a new version that should help. Let me know how it behaves on your system. The code now scales the divisor used to calculate current based on the size (nominal kW) of the generator. You can also override this value in the genmon.conf file.

alphaa211, 2018-05-03:
I have tomorrow off and I will try it then.

Thanks!

Doug Schreiner
785-608-6928 - Cell

jgyates, 2018-05-04:
@alphaa211 , make sure you have the latet update as of the time of this message. I just checked in a change that corrects one bug that is related to my fix from the 3rd.

jgyates, 2018-05-07:
@alphaa211 , have you had a chance to test this change?

alphaa211, 2018-05-07:
No, sorry.  Hopefully this afternoon!

Doug Schreiner
785-608-6928 - Cell

alphaa211, 2018-05-07:
Here is what I found.  Please see the last screenshot and notes as my local monitoring shows different amperage than Genmon. I also checked directly with a fluke clamp meter to double check my local monitor readings and they are within 5% of each other. 

Start & Run with No Transfer
[image]

Start & Run with No Transfer
Registers :
    Num Regs : 220
    Not Changed : 10196
    Changed : 828
    Total Changed : 0.08

    Base Registers :
        0239 : 0000
        0039 : 0000
        0038 : ffff
        0034 : 0001
        0037 : 1183
        0036 : 0000
        0237 : 0000
        0033 : 0000
        0032 : 2f22
        05f7 : 05c8
        05f6 : 05ad
        05f5 : 0007
        05f4 : 0007
        05f3 : 00c8
        05f1 : 0000
        05ed : 0064
        023b : 0000
        003a : 0000
        023e : 0000
        003b : 0141
        05fa : 0002
        0059 : 0000
        0058 : 0000
        0053 : 0000
        0052 : 0000
        0057 : 0000
        0056 : 0000
        0055 : 0000
        0054 : 0000
        0208 : 03f8
        005c : 0000
        005a : 0000
        005f : 0000
        005e : 0000
        005d : 0000
        003c : 0259
        000d : 0000
        000e : 0e22
        000f : 0507
        000a : 0085
        000b : 0000
        000c : 0037
        0005 : 111e
        0006 : 0200
        0007 : 0e10
        0000 : 0009
        0001 : 00030000
        0008 : 003c
        0009 : 00f2
        05ee : 05af
        0012 : 00ef
        0011 : 00ab
        0010 : 0112
        0019 : 000f
        002b : 0005
        002c : 111e
        002a : 6a74
        002f : 0000
        002d : 0000
        002e : 0002
        0022 : 463b
        0020 : 0000
        0021 : 0000
        001f : 4915
        001e : 0159
        001d : 0000
        001c : 0000
        001b : 463b
        001a : 0091
        01f4 : 33303030373839363736

Start & Run with Transfer After Transfer & Before AC Compressor Load Hits
[image]

Start & Run with Transfer After Transfer & Before AC Compressor Load Hits
Registers :
    Num Regs : 220
    Not Changed : 15787
    Changed : 1392
    Total Changed : 0.09

    Base Registers :
        0239 : 0000
        0039 : 0000
        0038 : 0000
        0034 : 0000
        0037 : 1131
        0036 : 0000
        0237 : 0000
        0033 : 0000
        0032 : 2ebe
        05f7 : 05c8
        05f6 : 05ad
        05f5 : 0048
        05f4 : 0048
        05f3 : 00c8
        05f1 : 0000
        05ed : 0064
        023b : 0000
        003a : 0000
        023e : 0000
        003b : 0191
        05fa : 0002
        0059 : 0000
        0058 : 0000
        0053 : 0000
        0052 : 0000
        0057 : 0000
        0056 : 0000
        0055 : 0000
        0054 : 0000
        0208 : 03f8
        005c : 0000
        005a : 0000
        005f : 0000
        005e : 0000
        005d : 0000
        003c : 0258
        000d : 0000
        000e : 0e25
        000f : 0507
        000a : 0084
        000b : 0000
        000c : 0037
        0005 : 111e
        0006 : 0200
        0007 : 0e19
        0000 : 0009
        0001 : 00030000
        0008 : 003c
        0009 : 00f5
        05ee : 0ad4
        0012 : 00f0
        0011 : 00ab
        0010 : 0112
        0019 : 000f
        002b : 0005
        002c : 111e
        002a : 6a74
        002f : 0000
        002d : 0000
        002e : 0002
        0022 : 463b
        0020 : 0000
        0021 : 0000
        001f : 4915
        001e : 0159
        001d : 0000
        001c : 0000
        001b : 463b
        001a : 0091
        01f4 : 33303030373839363736

Start & Run with Transfer After Transfer & After AC Compressor Load Hits
Local Monitoring shows approximately 20.5 (240v) amps which = 4920 watts

[image]

Registers :
    Num Regs : 220
    Not Changed : 22465
    Changed : 2143
    Total Changed : 0.10

    Base Registers :
        0239 : 0000
        0039 : 0000
        0038 : 0000
        0034 : 0000
        0037 : 1421
        0036 : 0000
        0237 : 0000
        0033 : 0000
        0032 : 2bdf
        05f7 : 05c8
        05f6 : 05ad
        05f5 : 00cc
        05f4 : 00ce
        05f3 : 00c8
        05f1 : 0000
        05ed : 0064
        023b : 0000
        003a : 0000
        023e : 0000
        003b : 0481
        05fa : 0002
        0059 : 0000
        0058 : 0000
        0053 : 0000
        0052 : 0000
        0057 : 0000
        0056 : 0000
        0055 : 0000
        0054 : 0000
        0208 : 03f8
        005c : 0000
        005a : 0000
        005f : 0000
        005e : 0000
        005d : 0000
        003c : 0257
        000d : 0000
        000e : 0e29
        000f : 0507
        000a : 0083
        000b : 0000
        000c : 0037
        0005 : 111e
        0006 : 0200
        0007 : 0e0b
        0000 : 0009
        0001 : 00030000
        0008 : 003b
        0009 : 00f6
        05ee : 0145
        0012 : 00f0
        0011 : 00ab
        0010 : 0112
        0019 : 000f
        002b : 0005
        002c : 111e
        002a : 6a74
        002f : 0000
        002d : 0000
        002e : 0002
        0022 : 463b
        0020 : 0000
        0021 : 0000
        001f : 4915
        001e : 0158
        001d : 0000
        001c : 0000
        001b : 463b
        001a : 0090
        01f4 : 33303030373839363736

alphaa211, 2018-05-07:
Here is after transfer back to utility about 30 seconds into cool down:

[image]

Registers :
    Num Regs : 220
    Not Changed : 64747
    Changed : 6177
    Total Changed : 0.10

    Base Registers :
        0239 : 0000
        0039 : 0000
        0038 : 0000
        0034 : 0000
        0037 : 123c
        0036 : 0000
        0237 : 0000
        0033 : 00a0
        0032 : 2e0e
        05f7 : 05c8
        05f6 : 05ad
        05f5 : 0007
        05f4 : 0007
        05f3 : 00c8
        05f1 : 0000
        05ed : 0064
        023b : 0000
        003a : 0000
        023e : 0000
        003b : 029c
        05fa : 0002
        0059 : 0000
        0058 : 0000
        0053 : 0000
        0052 : 0000
        0057 : 0000
        0056 : 0000
        0055 : 0000
        0054 : 0000
        0208 : 03f8
        005c : 0000
        005a : 0000
        005f : 0000
        005e : 0000
        005d : 0000
        003c : 0257
        000d : 0000
        000e : 0f05
        000f : 0507
        000a : 0082
        000b : 0000
        000c : 0037
        0005 : 111e
        0006 : 0200
        0007 : 0e0a
        0000 : 0009
        0001 : 00050000
        0008 : 003b
        0009 : 00f6
        05ee : 0352
        0012 : 00ef
        0011 : 00ab
        0010 : 0112
        0019 : 000f
        002b : 0005
        002c : 111e
        002a : 6a74
        002f : 0000
        002d : 0000
        002e : 0002
        0022 : 463b
        0020 : 0000
        0021 : 0000
        001f : 4915
        001e : 0158
        001d : 0000
        001c : 0000
        001b : 463b
        001a : 0090
        01f4 : 33303030373839363736

jgyates, 2018-05-07:
make the following entry in your /etc/genmon.conf file

     currentdivider=56.24

then reastart genmon and let me know if the results are closer. I added a note in the wiki about this, see #5 on this page:

https://github.com/jgyates/genmon/wiki/Appendix-4-Known-Issues

If this works I can make the code use 56.24 as the default divider for 11kw models.

jgyates, 2018-05-08:
@alphaa211 , I checked in an update so make sure you test with the latest version.

jgyates, 2018-05-17:
@alphaa211 

Where you able to test the update?

alphaa211, 2018-05-17:
No.  I have the afternoon off tomorrow and I will try it then.  I have the latest version of GenMon.  Do I still need to make the currentdivder change? Do you still want to know what local monitoring says I am pulling amperage wise vs. what GenMon is displaying? Do you want registers?

jgyates, 2018-05-17:
No, you should not need a divider.  I have the default divider for the 11kw model set to 56.24 in the code so if you have a currentdivider in the conf file you can remove it and just test the latest code.

jgyates, 2018-05-17:
Great! Thanks for the feedback.

closed 2018-05-17 by jgyates