jgyates/genmon #108: E-mail settings

JustApu, 2018-05-01:
(1) I am getting errors trying to login to my e-mail server for outgoing e-mail. Is there any additional SMTP debugging available?

`[mymail.log]`
`2018-05-01 13:33:16,975 : Error SMTP sendmail: (535, '5.7.0 authentication failed')`
`2018-05-01 13:33:16,999 : Error in SendMailThread, sendEmailDirectMIME failed, retrying`

I did connect, from my Pi to the mail server, via the command line, successfully using the Base64-encoded username and password.

`$ telnet mailserver.example.com 587`
`Trying 1.2.3.4...`
`Connected to mailserver.example.com.`
`Escape character is '^]'.`
`220 mailserver.example.com Tue, 1 May 2018 13:29:28 -0400`
`EHLO genmon`
`250-mailserver.example.com Hello customer.provider.net [4.5.6.7], pleased to meet you`
`250-ENHANCEDSTATUSCODES`
`250-PIPELINING`
`250-8BITMIME`
`250-SIZE`
`250-DSN`
`250-AUTH DIGEST-MD5 CRAM-MD5 LOGIN PLAIN`
`250-STARTTLS`
`250-DELIVERBY`
`250 HELP`
`AUTH LOGIN`
`334 VXNlcm5hbWU6`
`c3Bmbxxxxxx=`
`334 UGFzc3dvcmQ6`
`eGJ4RxxxxxxxxxU=`
`235 2.0.0 OK Authenticated`
`QUIT`
`221 2.0.0 mailserver.example.com closing connection`
`Connection closed by foreign host.`

(2) Is there a reason that the "Mail Processed Folder" has to be 11-50 characters long? I was trying to use a shorter name but the WebUI rejected it.

jgyates, 2018-05-01:
Hi @JustApu 

You can add some debug to the file /genmon/genmonlib/mymail.py. To do this you could add the line 

      smtplib.set_debuglevel(5)

see https://docs.python.org/2/library/smtplib.html

 at the top of the SendMailThread function (line 277). It should give more info on error conditions if you do that.

For your 2nd question, no reason for 11-50, probably a typo, I will change this to something more reasonable.

jgyates, 2018-05-01:
the update to allow shorter names is checked in.