jgyates/genmon #110: Modbus address

darkstar2002, 2018-05-06:
has anybody had to use a different modbus address for a nexus controller?  Running a pi  zero w   loopback  works threw the max converter.   main program has errors    tried running the modbu s dump with the default address and get nothing.

darkstar2002, 2018-05-06:
python modbusdump.py -r 9600 -p /dev/serial0 -a 9d -s 5 -e 7
Baud Rate : 9600
Port is : /dev/serial0
Address is : 9d
Start Register : 5
Start Register : 7
OK

2018-05-06 14:00:03,895 : Discarding byte slave: 41
2018-05-06 14:00:03,900 : Error Receiving slave packet for register 05
2018-05-06 14:00:04,172 : Discarding byte slave: 41
2018-05-06 14:00:04,174 : Error Receiving slave packet for register 06

jgyates, 2018-05-06:
Hi @darkstar2002, 

Let's try two things to get some more data:

1) As a data point what error do you get if you unplug your connector (i.e. not connected to the controller and not looped back). Is the error timeout or discard byte? While this will not answer our question it may point us in the right direction.

2) Swap your transmit and receive after the max converter, connect to the controller and repeat the modbusdump test.

Let me know the results of these two tests.

Thanks
-Jason

darkstar2002, 2018-05-06:
I get time outs or serial errors.   Doing some more looking my control panel does not have the same button covers as the nexus but everything else looks like a nexus.   Software 0h0648 rev.d    assembly 0h0643 rev c    k1522.   Model 0057440

jgyates, 2018-05-06:
Is this your generator:

https://www.generac.com/service-support/product-support-lookup/product-support-details?productid=7ad9f06f-56a8-4278-a008-69fe0d0d0f21

Is this a generac generator or some other company branded generator. Generac does license / sell their controllers to other companies like Eaton and Honeywell. 

Timeout are good, that is the expected error. Let me know the results of swapping tx and rx. You should be able to swap these on the RS-232 side of the max board before they go into your cable.

darkstar2002, 2018-05-06:
Yes that is my generator.
swaped lines
Baud Rate : 9600
Port is : /dev/serial0
Address is : 9d
Start Register : 5
Start Register : 7
OK
Exception in thread SerialReadThread (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 754, in run
  File "/home/pi/test/genmon/genmonlib/myserial.py", line 78, in ReadThread
  File "/home/pi/test/genmon/genmonlib/myserial.py", line 131, in Read
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 506, in read
<type 'exceptions.TypeError'>: 'NoneType' object is not callable

2018-05-06 17:00:24,609 : Error: timeout receiving slave packet for register 05 Buffer:0
2018-05-06 17:00:27,622 : Error: timeout receiving slave packet for register 06 Buffer:0
unhooked
Baud Rate : 9600
Port is : /dev/serial0
Address is : 9d
Start Register : 5
Start Register : 7
OK
Exception in thread SerialReadThread (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 754, in run
  File "/home/pi/test/genmon/genmonlib/myserial.py", line 78, in ReadThread
  File "/home/pi/test/genmon/genmonlib/myserial.py", line 131, in Read
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 506, in read
<type 'exceptions.TypeError'>: 'NoneType' object is not callable
2018-05-06 17:03:44,609 : Error: timeout receiving slave packet for register 05 Buffer:0
2018-05-06 17:03:47,614 : Error: timeout receiving slave packet for register 06 Buffer:0

jgyates, 2018-05-06:
That is a good data point. We can swap the tx and rx back to they way you originally had them. 

I want to ask this to rule out any other problems that my be related to the software or installation. Is there anything unique or different about this setup (did you install it with the install script or vary anything in the installation). I don't know of anything in the data we have so far that would make me think this is the issue, but just want to rule that out.

Another thing that is interesting is that your model numbers is 0057440. I register dumps from other Nexus controllers that are 0058xxx. Do you know how old this generator is? Can you look at the controller menu and get the version of the firmware? I could compare that to known good units.

Let me think on this some more. If it is something different about the serial / modbus interface we should be able to track that down and get a fix. I would need to modify or create a new test program but that should be easy.

darkstar2002, 2018-05-06:
I do not know the exact age.  It was slighlty used when I got it a year ago.   I installed with the script no errors.   Redownloaded into another directory so I could have a clean download to test the dumpmodbus script.
firmware is 1.1 hw 1.0 in the menu.  I think the address may be right.  I changed the address to several others and I get the same error as if I am unhooked or swaped lines.

jgyates, 2018-05-06:
OK. Let me compare that to some other fw / hw versions. In the mean time, since this is an older unit, let's try some slower data rates on the command line. Try 1200, 2400 and 4800

jgyates, 2018-05-06:
If that does not work I can mod the program to support something besides No parity, 8 data bits, and 1 stop bit. the fact that we are getting something back is good. Another thing I can check is that generac products like MobileLink and the remote monitor use the same serial modbus interface we are using so if there are notes about compatibility with this generator then that may be a clue, of course they could support multiple data rates (and genmon can to, but we have to figure out what that data rate may be).

jgyates, 2018-05-06:
Look on the inside of your generator housing, there is usually a metal plate that has the model number and rated kw, etc. If you can find this plate look for a manufacture date. This is from generac web site:

http://www.generac.com/all-products/parts-accessories/mobile-link?lang=en-US

 Mobile Link is designed for use with a generator equipped with an LCD display. Mobile Link is compatible with most 2008-and-newer air-cooled standby generators and 2010-and-newer liquid-cooled standby generators from Generac

jgyates, 2018-05-08:
It may be worth running the modbus test at the lower data rates to see if we get some registers back. Once we have a dump we should be able to tell if the registers are close to the Nexus registers.

darkstar2002, 2018-05-13:
I tried 1200, 2400 and 4800 all I get is Error: timeout receiving slave packet for register 05 Buffer:0
If I go back to 9600 I get Discarding byte slave: 41  Error Receiving slave packet for register 05.  I have this semi hooked up.  I should be able to try things out faster and more often then I could before.

jgyates, 2018-05-13:
ok. Let me make some mods to modbusdump.py so we can change the serial parity and stop bits.

jgyates, 2018-05-14:
I checked in an update to modbusdump.py that should let us test other serial parameters:

9600 N,8,1:

      python modbusdump.py -r 9600 -p /dev/serial0 -a 9d -s 0 -e 3

9600 O,8,1:  (Odd parity)
     python modbusdump.py -r 9600 -p /dev/serial0 -a 9d -s 0 -e 3 -x 1

9600 E,8,1: (even parity)

     python modbusdump.py -r 9600 -p /dev/serial0 -a 9d -s 0 -e 3 -x 2

9600 N,8,1.5 (1.5 stop bits)

     python modbusdump.py -r 9600 -p /dev/serial0 -a 9d -s 0 -e 3 -b

The -b will test 1.5 stop bits, if not present it will use one stop bit 
The -x if omitted is no parity, if -x 1 then it uses odd parity, if -x 2 then it uses even parity.

You can mix the data rate, parity and stopbits to see if we can find the right serial parameters.

Let me know if you have any questions.

darkstar2002, 2018-05-14:
I get an error
Error opening serial device...: __init__() got an unexpected keyword argument 'Parity'

jgyates, 2018-05-14:
Did you do a full update of the repository? the program modbusdump.py relies on files in the ./genmonlib directory. If this directory was not updated you will see an error like this.

darkstar2002, 2018-05-14:
I tried all the options with 9600 and got the same error message.    I will be trying other speeds out soon.

darkstar2002, 2018-05-14:
The same error with a correctly working program.  I did update all of the program.  That part is working correctly

jgyates, 2018-05-14:
ok. Sound like the unexpected keyword problem is resolved. Let me know how the other tests go.

darkstar2002, 2018-05-14:
Error: timeout receiving slave packet for register 01 Buffer:0

jgyates, 2018-05-14:
Did it print anything else? It  should print Register 00 or have another error, or was the last error on register 01 and an error on 00 as well. If the later is true it looks like we are only getting one byte back (41 hex). In theory we could try other modbus address (00 - ff) but that is going to be time consuming for you...

darkstar2002, 2018-05-14:
Baud Rate : 9600
Port is : /dev/serial0
Address is : 41
Start Register : 0
Start Register : 3
OK

2018-05-14 12:16:02,001 : Error: timeout receiving slave packet for register 00 Buffer:0
2018-05-14 12:16:05,013 : Error: timeout receiving slave packet for register 01 Buffer:0
2018-05-14 12:16:08,024 : Error: timeout receiving slave packet for register 02 Buffer:0

jgyates, 2018-05-14:
I have another idea, I need to update the code to try this. I need to adjust the timeout value based on the data rate. I will let you know once I update the code so you can test again...

jgyates, 2018-05-15:
I checked a modification that will make the length of time of the timeout based on the data rate + 3000 milliseconds. Not sure if this will help but it is worth a shot...

You will need to update the full repository again as the modbus stuff is handled in one of the library files in the /genmonlib directory.

darkstar2002, 2018-05-15:
I updated.  when using address 00, 41, 9d it respods back quickly with ok and the error messages about discarding byte.  Any other address it takes awhile and errors out with a timeout.

Is there maybe a verbose options  so that we can see more what is going both ways?   It's like its answering something but maybe a little different than you are setup to receive.

darkstar2002, 2018-05-15:
I tried 9600, 4800, 2400.   Maybe its not modbus that this thing uses.

jgyates, 2018-05-15:
Did it respond with a 41 again? Just one byte?

While you have it hooked up you could try these rates: 19200, 38400, 57600, and 115200

darkstar2002, 2018-05-15:
Nothing came back.  I left it running for 2 min each.    Once the rain moves out I will hook the stuff back up and try the other speeds.

jgyates, 2018-05-15:
That is odd that it did not return anything yet it returns 41 with the other test program. On my system it returns the same values. You mentioned that you believe this was a gen 3 panel. Where did you find the info about gen 3 panels?

darkstar2002, 2018-05-15:
This panel works just like a nexus does.  All the menu items line up with a nexus except instead of up and down arrow buttons I have  left and right.  The 8 pin plug on back seems to work the same. I am guessing it is a gen 3 from looking at omnimetrix.net generator monitoring stuff. https://www.omnimetrix.net/wp-content/uploads/2017/03/installation_Generac_GenerationI-II-III_Residential.pdf

darkstar2002, 2018-05-16:
I do have the molex plug on mine but I do not thing this will work.  It must speak something different.  It can not use the cell stuff.   It had before me the advanced wireless monitor.  So it does have some kind of bidirectional communication.

At this point I am ready to give up.  Thanks for trying to get this going.  I will keep an eye out for stuff or manuals on this panel to see if down the road something can be done.

gm58, 2018-05-21:
darkstar2002,

It appears I have the same controller as you;

https://www.zillerelectric.com/products/generac-0h06430srv-2008-air-cooled-control-panel

Reading thru this thread one would think that this controller just doesn't communicate modbus.  Well while playing around trying to get mine going I clicked on the "Start Generator" button and low and behold, the generator started.  Seeing this I would have to say it does speak modbus.  But.. I was unable to stop the generator thru the UI.  Probably due to the Pi not receiving information from the controller telling it the generator was running.  The lettering on the Stop Generator button appeared to be greyed out.  When I went to the generator to stop it, the display read "Running - Radio Start"  

Have you done any other testing or research on the issue?

darkstar2002, 2018-05-21:
I can confirm that the generator does start.

jgyates, 2018-05-21:
When issue the command to start and stop, the following is the sequence of bytes written (9600, N,8,1):

Remote Start:
0x9d, 0x10, 0x00, 0x04, 0x00, 0x01, 0x02, 0x00, 0x00, 0x5E, 0x2D
<wait for response packet>
0x9d, 0x10, 0x00, 0x04, 0x00, 0x01, 0x02, 0x00, 0x01, 0x9E, 0A5
<wait for response packet>

Remote Stop:
0x9d, 0x10, 0x00, 0x04, 0x00, 0x01, 0x02, 0x00, 0x00, 0x5E, 0x2D
<wait for response packet>
0x9d, 0x10, 0x00, 0x03, 0x00, 0x01, 0x02, 0x00, 0x00, 0x5F, 0x65
<wait for response packet> 

In general the packet format is this:

Modbus address = 0x9d
Modbus Write Command = 0x10
Modbus Register to Write = 0x04 0x00 == register 0x0004
Modbus Write Length = 0x00, 0x01  == Length of 0x0001 , one word 
Modbus Data to Write = 0x00, 0x00  == Write data 0x0000
Last two bytes are CRC 

The sequence of commands is :

To start:
write to register 0x0004, value 0x0000
write to register 0x0003, value 0x0001

To stop:
write to register 0x0004, value 0x0000
write to register 0x0003, value 0x0000

So the question becomes, does the generator turn on because of the total of both modbus writes, or is turning on due to a subset of the bytes written?

gm58, 2018-05-24:
Jason,

Yesterday while showing a friend the genmon UI on my IPad.  I was shocked as I watched the needle of the "Battery Voltage" on the Dashboard climb to 12V.  I then clicked on the "Monitor" tab and saw that no packets have been received.  Kind of weird wouldn't you say?

gm58, 2018-05-24:
I should add that now when I look at the Dashboard the "Battery Voltage" is reading 10V

jgyates, 2018-05-24:
Look on the registers page (click the icon in the upper right corner of the web page. This page list all of the registers that are read. Register 000a is battery voltage. What is the value of this register?

If received packets is zero, I would say that this is odd behavior. I will look at the code to see if anything jumps out as to why this could occur.

jgyates, 2018-05-24:
After looking at the code it appears that the web UI has some default values that may be showing up in the gauge. Since this is a nice round number 12.0 instead fo an actual battery voltage (around 12.4V - 13.6V) I expect this is an issue with the web app.

gm58, 2018-05-24:
Hello Jason,

I tried what you asked, although it appears you found the reason I was seeing voltage, but when I clicked on the printer icon "24h" all I got was an empty page.

Well at least I found you something to fix.. 😃