jgyates/genmon #112: kW Output Wacky

dennis48755, 2018-05-16:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

### Expected Behavior

kW Output to be some value less than 16,000kW

### Actual Behavior

kW Output to be greater than 46,000,000kW

### Steps to Reproduce (including precondition)

Run generator, view GenMon Status

### Screenshot on This Problem (if possible)

[image]

### Your Environment

- Genereator Model: 0064622
- Generator Registers: Submitted (hopefully) via GenMon > Monitor > Submit Registers button
- Genmon Version: V1.7.9

dennis48755, 2018-05-16:
Thanks Jason, I'm glad to help when needed.

A note about this incident, this was during a scheduled exercise, so there was no real load. However I had the same symptoms during a power outage last week when the generator was supplying power to my house.

Dennis

jgyates, 2018-05-16:
@dennis48755 

I checked in a mod that should help put the value into a reasonable range. The base problem is that we are reading the value of two registers, these values represent what we think it a byproduct of the current calculation that takes place in the controller for over current protection. In the generators we have tested with this value goes up and down according to the load on the generator. We have also found that the conversion is slightly different based from model to model and it appears to vary based on the rated kW of the model. I have never received feedback on a 16kW model before so your input is a good data point. It appears that you registers are returning a negative value for the registers we are reading. I modified the code to use the absolute value of this register contents. In my simulation it looks good but I just used a snapshot of your registers to test. What I need from you is this:

* Check out the latest code (you can update from the "Monitor" page in the web interface.
* Run your generator with a load (exercising or running with out a load will not get us the info we need)
* Turn on and off some appliances (like a stove or clothes dryer) that use a decent amount of current
* Validate that the current output (and kW output) goes up when the increased load is applied

Note that the generator has a minimum load built in so you may see the value go to about 20% of your rated kw output with a minimum load and stay in that range. Once your house is producing more than the minimum load value you *should* see the kw and current rise. Let me know if this is the case.

If it the kw output decreases as you add a load then I can run another test to get more data.

Also, the code uses an internal divisor to do this calculation. It is in the ball park (i.e. our best guess) but until we have a measured value to validate it remains our best guess and what the divisor should be. If you have a clamp on style multi meter that can measure current we can perform an additional test to make the reading more exact. Let me know if you have this type of meter.

[Here](https://smile.amazon.com/dp/B00NWGZ4XC/ref=sspa_dk_detail_4?psc=1&pd_rd_i=B00NWGZ4XC&pd_rd_wg=DSg3T&pd_rd_r=TE5J2RSGSHDFVBYJDRE3&pd_rd_w=KAGVH) is an example of the type of meter that would measure the current.

I have a note in the wiki about the current calculations here: see item 5 on [this](https://github.com/jgyates/genmon/wiki/Appendix-4-Known-Issues)  page.

Let me know if you have any questions.

dennis48755, 2018-05-16:
Jason

This is generator on normal household load, about 1.57kW
[image]
kW...

Below is adding electric stove, about 3.85kW...
[image]

GenMon is reflecting a more normal range, however the 2kW load change didn't appear to be reflected, or possibly decreased the value.

I did the testing rather quickly to get you some preliminary data.

Dennis

krisasmith, 2018-05-16:
What equipment is that on the left providing the digital readout?

jgyates, 2018-05-17:
@dennis48755, thanks for the pictures, this helps. Can you go to the registers page (click on the icon in the top right corner of the web interface) and then scroll to the bottom and click icon labeled 10m, 1h and 24h. This will allow you to print the register contents for the last 10 minutes, 1 hour or 24 hours (assuming your web interface has been running for that long). If you use chrome you can print to a PDF file. Save the file and attach it to this thread. From the looks of you pictures it appears that I will need to add an offset to the registers we are reading for current. Your multimeter setup is nice. I may try that as well. 

Basically here is what we need to figure out:

We use the contents of the modbus register 003a and 003b to form a 32 bit value. This value is divided by a constant to obtain the current. The constant appears to be different for generators with differing rated kW. Your model appears to have a negative offset. The questions are:

1) is your offset negative because the current calibration in the dealer menu was not performed during install
2) What value should be used for the offset to add to the value read from registers 003a and 003b.

To try to determine the answers let's take this approach:

First, send me another register submission with your generator not running.
Second, run the generator with a load for 10 minutes with the web interface running then send the PDF described above for the 10 minute interval. This will let me know how the variable moves over time.

There also appears to be a minimum load internally on the generator, based on our testing with other generators. I am attaching the graph of the calculated current along with the actual measured current of a 20kw generator. The red line is the current measured with a multimeter, the blue line is the current calculated from register 003a and 003b. This graph should visualize the minimum current situation better.

Once I have the data listed above I hope to have enough info to make another attempt at getting this correct for your generator. Just and FYI, I have another test program that I can send that will log the values of specific registers over time to a file. We can use this file for analysis along with the output of your multimeters.
[image]

jgyates, 2018-05-17:
One other question. Make a note of the values of registers 003a and 003b when the generator is not running, then run the generator (no load required) and stop it. Check to see if the values of 003a and 003b go back to their original values. My current thinking is that your internal current transformers were not calibrated during the install, which would explain the negative numbers. If that is the case then I can take the value of this registers when the engine is not running and use that as an offset to make it non zero when the generator is running. 

The reason I am thinking this way is that I have around 20 different register dumps from Evolution Air cooled models (yours is the only 16kw in that group). All other models I have report these registers as zero when the generator is not running. It should be easy enough to validate this theory with the data point above and a quick mod to the code.

jgyates, 2018-05-17:
I just checked in a new version that we should test. Once you update the new version add this value to /etc/genmon.conf

     currentoffset=201

I am not 100% sure about the value 201. Basically this parameter will add the number to you current register readings from 003a and 003b. I was thinking that I could read the values of the registers when it was not running but your registers indicate the value is zero when not running so we need to imperically change this value. Try it with 201, and you can run your generator, if you get a negative value you need to increase the value, if it is larger than your meters are reading we should increase it.

There are two values in the conf file that are relevant here, the first one, mentioned above and the second one is currentdivisor. Both parameters are optional. Current is calculated like this for Evolution Air Cooled models:

* If the generator is not running current reported back is always zero.
*  currentoffset is zero if not present in the conf file
* currentdivisor is 22/rated kW * 22 unless overridden in the conf file (except for 22kw and 11kw version as this has been tested explicitly and specific values of and 21.48 and 56.24 are used for defaults)
* Current = (combined 003a and 003b register values) + currentoffset / currentdivisor
* currentoffset can be any value, positive or negative, in your case it needs to be positive.
* currentdivisor should be 1 or greater. In your case I would omit this until you got a current offset value that was close, then you can add currentdivisor to your conf file and modify it to make the current reading exactly match your meters. I would start with a value of 30 for currentdivisor

To summarize, add currentoffset=201 to you conf file, restart the software and see if this creates a non-negative current, modify the offset up or down untiul you do not get a negative current value, but try not to make it too high (i.e. as low as you can without a negative current). Once you get that value you can add currerentdivisor to your conf file. Start with currentdivisor=30 and modify it up or down to make your meter match what the software is reporting. For the divisor you may only have to modify it slightly (i.e. 29.5 instead of 30)

Let me know the values that work for you. The currentdivisor value I should be able to add the the software and it should work for others who have an 16kw generator. The currentoffset should be unique to your generator due to the current transformers not being calibrated.

dennis48755, 2018-05-18:
I didn't have as much time to fiddle today as I had hoped.

Using currentoffset=201 yielded negative readings.
[image]

I had to go al the way up to 351 to get it close and even then it was slightly low. See image below...
[image]

I'm a bit confused how currentdivisor=30 fits in. If I finally reach a reasonable current value using currentoffset, wouldn't dividing the value by 30 destroy it?

I'll work on it more tomorrow.

Dennis

jgyates, 2018-05-18:
Because you have a 16kw generator the divisor that is being used now is 30.25. You can adjust this to any positive, non-zero floating point value. For example with no setting in your conf file for currentdivisor the software will use this forumul for the current divisor (22 * Nominal Kw) / 22. Since you have a 16kw Nominal generator that would use 30.25. You can override this value (for example 31.5) and it will slightly vary your current output. If I did not use the currentdivisor the result would be off by a large amount, but since I am not able to test on every generator variant I get relatively close with the simple formula. You can get closer if you want to tweak it. 

If you want to increase the current output value you can decrease the current divisor slightly. Try 30  or 29.5 and see if your results are better.

dennis48755, 2018-05-21:
Sorry for the delay, I went to resume activity with this issue this morning but found I can't connect to the Raspberry Pi 3B including via PuTTY. Red power LED is on but no green activity or RJ45 LED's are on. If I remove/reconnect power, besides the red LED being on, the green LED flashes once or twice then nothing.

The Raspberry is in the basement by the transfer switch. Guess I'll have to bring it up and connect a monitor, mouse and keyboard. Wish I had made a backup of the SD card.

Dennis

jgyates, 2018-06-01:
@dennis48755,

If you are using Wifi your pi may have disconnected from your AP. Linux is notiriously bad about reconnecting automatically. There has been discussion in some of the project threads about using a crontab job to check if wifi is connected. Just a thought...

jgyates, 2018-06-05:
@dennis48755 

Closing this issue for now, feel free to comment if you have any other info.

closed 2018-06-05 by jgyates