jgyates/genmon #114: CPU temp enhancement

steve-bailey1, 2018-05-18:
Would it be possible to add a spot to display cpu temp for the device running the software?  It would need to be configurable since the devices that run this software are different but example of a Pi would be echo /sys/class/thermal/thermal_zone0/temp then divide by 1000 for cpu temp in Celsius.  

Not sure if others would find this helpful but I believe most of us are running this software on rather small hardware installed inside of the enclosure and it would be helpful to monitor the temps of these little things.

JustApu, 2018-05-18:
+1 though perhaps make it flexible/extensible so the end-user can flow other data into the system. genmon wouldn't have to necessarily parse/act on the data, just have a way to receive pre-processed data and (preferably both) display it or report it in the JSON string returned by the client interface.

Besides reporting the Pi's CPU temp, other ideas include WiFi signal strength, environmental data from a BMP280/BME280 connected via I2C, or temperatures from a 1-wire network (so you could monitor battery temp, engine temp, etc.).

Right now, my Pi has a BMP280 that I collect data from outside of genmon but otherwise it is dedicated to genmon so it would be great if this was integrated as well.

jgyates, 2018-05-19:
The monitor page now shows the CPU temp and CPU utilization. You can disable pi specific stats by adding 

     disablepispecific=True 

to your genmon.conf file.

Regarding adding end-user data. That sounds like a good idea too. Let me think of the best way to handle this. At the moment I am just thinking of having a json file in the genmon directory. If it is present it will read the json data and display it.

steve-bailey1, 2018-05-19:
Just updated and I see the area but there are no details.
[image]

jgyates, 2018-05-19:
Does your genmon.log file have any errors?

steve-bailey1, 2018-05-19:
Yes.  

2018-05-18 16:49:26,267 : GenMon Loadded for site: Charles Street
2018-05-18 20:39:32,180 : New version detected: Old = V1.8.1, New = V1.8.2
2018-05-18 20:39:32,272 : GenMon Loadded for site: Charles Street
2018-05-18 20:39:34,724 : Looking up model info on internet
2018-05-18 20:39:36,400 : Found: Model: G0062440
2018-05-18 20:40:14,426 : GenMon Loadded for site: Charles Street
2018-05-18 20:40:25,347 : Error in GetRaspberryPiInfo: [Errno 2] No such file or directory : genmon.py:466
2018-05-18 20:40:28,488 : Error in GetRaspberryPiInfo: [Errno 2] No such file or directory : genmon.py:466
2018-05-18 20:40:33,314 : Error in GetRaspberryPiInfo: [Errno 2] No such file or directory : genmon.py:466
2018-05-18 20:40:33,528 : Error in GetRaspberryPiInfo: [Errno 2] No such file or directory : genmon.py:466

steve-bailey1, 2018-05-19:
Ahh I think I know why.
I am running CentOS on my PI.  I dont have that utility available.
process = Popen(['/opt/vc/bin/vcgencmd', 'measure_temp'], stdout=PIPE)

jgyates, 2018-05-19:
ok. Let me think about that. I may be able to put a fallback plan in there.

steve-bailey1, 2018-05-19:
I found in CentOS that you can take the value in '/sys/class/thermal/thermal_zone0/temp' then divide by 1000 and thats degrees in Celsius then you can multiply by 1.8 and add 32 and that should be Fahrenheit

jgyates, 2018-05-19:
Try it now.

jgyates, 2018-05-19:
@JustApu 

I added some code to import user defined json. Put your json string in a file named "userdefined.json" in the genmon directory. It should show up on the Monitor page. The file is read every time the genmon is asked to display the monitor info.

Let me know hot it works for you.

Jason

steve-bailey1, 2018-05-19:
Very Nice :)

jgyates, 2018-05-19:
I added a brief description of the import JSON to this page:

https://github.com/jgyates/genmon/wiki/Appendix-3--Interfacing-Generator-Monitor-to-External-Applications

JustApu, 2018-05-19:
@jgyates - Thanks. A very quick test with a static JSON file leads me to believe it is working but I will test more tomorrow or the day after. Need to get a better WiFi connection to the Pi as it's marginal right now and the parts to hopefully improve that arrived late today.

zekyl314, 2018-05-19:
I’d love to see your example json file posted when you have time.

-Mike

<https://github.com/jgyates/genmon/issues/114#issuecomment-390375755>, or mute
the thread

JustApu, 2018-05-19:
Sample JSON file attached - just remove the ".txt" file extension. Once I get the WiFi fixed, I'll see if I can work up some code that you could put on your Pi that could be easily customized to report different values back to that JSON file.

[sample.json.txt](https://github.com/jgyates/genmon/files/2019601/sample.json.txt)

JustApu, 2018-05-20:
userjson.py - creates a file named userdefined.json which contains a JSON string of:
1. the current weather (temperature, humidity & pressure) returned by a BME/BMP280 (needs Adafruit_BME280.py from https://github.com/adafruit/Adafruit_BME280_Library)
2. the signal strength of the current wlan0 connection (in dBm)
3. the temperature readings from 1-wire sensors (requires owfs & python-ow packages)

Caveats: (1) my BMP280 is malfunctioning so, while that section of code worked in another script, it isn't actually tested in this script and (2) I'm using fake 1-wire devices as connecting real ones to my Pi is still a "to do" item.

[userdefined.json.txt](https://github.com/jgyates/genmon/files/2020044/userdefined.json.txt)
[userjson.py.txt](https://github.com/jgyates/genmon/files/2020045/userjson.py.txt)

jgyates, 2018-05-20:
I added code to detect the used network interface and if using wifi report signal strength on the Monitor page. You can disable the display of platformspecific info in the conf file. It is enabled by default. 

@JustApu is used some of your code on the signal strength. Thanks!

proppilot, 2018-05-20:
wow has this project ever gotten wings !  Jason, I did the update using the click link on the monitor page and everything seemed fine.  I cleared the cache on my browser and when I open the web page the wifi strength is not displayed anywhere.  Should it ?

jgyates, 2018-05-20:
What is displayed on the Monitor page?

I should display wifi signal strength if you are using wifi as your primary network device, otherwise it will just show which adapter is used.

proppilot, 2018-05-20:
No reference to wlan at all

On iPhone or desktops.  I did the upgrade from the web browser and update button vs the usual way via terminal and SSH into the Pi.  Does anything need to be done after an upgrade to refresh other than clearing caches ?

jgyates, 2018-05-20:
A picture of your Monitor page would help. The only thing that is needed after an update is a browser refresh.

proppilot, 2018-05-20:
Here you go

[image]

proppilot, 2018-05-20:
Here is a screenshot of a monitor program I use to check health.  Wlan0 is an external USB adapter

[image]

proppilot, 2018-05-20:
> <reply+02039b7ab077e38d02e9297841709d1702162025f842fbe392cf0000000117197ee29
> 2a169ce13585f54@reply.github.com>
> <comment@noreply.github.com>
>
> A picture is work a thousand words. The problem is your update is not
> working. You are still on version 1.7.1, the current version is 1.8.5 (soon
> to be 1.8.6).
>
> Open a console and from the genmon directory type:
>  sudo bash ./genmonmain.sh update
> If that works with no errors, restart genmon:
>
> bash ./startgenmon.sh restart
>
> Refresh your browser.
>
> If you get errors on the update post them here.
>
> 

Worked perfectly.  That was the first time I tried an update from the link
on the web browser and I will use the command line for now on

How does one know the current posted version on Github ?

bopGrLzks5t0bRigaJpZM4UFYos> .
type":"ViewAction","target":"https://github.com/jgyates/genmon/issues/114#is
suecomment-390501138","url":"https://github.com/jgyates/genmon/issues/114#is
suecomment-390501138","name":"View Issue"},"description":"View this Issue on
GitHub","publisher":{"@type":"Organization","name":"GitHub","url":"https://g
767c55493e4bb","name":"GitHub"},"entity":{"external_key":"github/jgyates/gen
mon","title":"jgyates/genmon","subtitle":"GitHub
repository","main_image_url":"https://cloud.githubusercontent.com/assets/143
418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"h
ttps://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6
-9aed-b52498112777.png","action":{"name":"Open in
GitHub","url":"https://github.com/jgyates/genmon"}},"updates":{"snippets":[{
"icon":"PERSON","message":"@jgyates in #114: A picture is work a thousand
words. The problem is your update is not working. You are still on version
1.7.1, the current version is 1.8.5 (soon to be 1.8.6).\r\n\r\nOpen a
console and from the genmon directory type:\r\n\r\n     sudo bash
./genmonmain.sh update\r\n\r\nIf that works with no errors, restart
genmon:\r\n\r\n   bash ./startgenmon.sh restart\r\n\r\nRefresh your
browser.\r\n\r\nIf you get errors on the update post them here.\r\n
"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/114#issuecomment-3905
01138"}}}{ "@type": "MessageCard", "@context":
"http://schema.org/extensions", "hideOriginalBody": "false", "originator":
"37567f93-e2a7-4e2a-ad37-a9160fc62647", "title": "Re: [jgyates/genmon] CPU
temp enhancement  (#114)", "sections": [ { "text": "", "activityTitle":
"**jgyates**", "activityImage":
"https://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11
e6-9aed-b52498112777.png", "activitySubtitle": "@jgyates", "facts": [ ] } ],
"inputs": [ { "isMultiLine": true, "@type": "TextInput", "id":
"IssueComment", "isRequired": false } ], "actions": [ { "name": "Comment",
"@type": "HttpPOST", "target": "https://api.github.com", "body":
"{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\":
\"jgyates/genmon\",\n\"issueId\": 114,\n\"IssueComment\":
\"{{IssueComment.value}}\"\n}" } ] }, { "name": "Close issue", "@type":
"HttpPOST", "target": "https://api.github.com", "body": "{\n\"commandName\":
\"IssueClose\",\n\"repositoryFullName\": \"jgyates/genmon\",\n\"issueId\":
114\n}" }, { "targets": [ { "os": "default", "uri":
"https://github.com/jgyates/genmon/issues/114#issuecomment-390501138" } ],
"@type": "OpenUri", "name": "View on GitHub" }, { "name": "Unsubscribe",
"@type": "HttpPOST", "target": "https://api.github.com", "body":
"{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 336955948\n}" } ],
"themeColor": "26292E" }

proppilot, 2018-05-20:
Another question.  Would it make more sense to post link quality as it gives an xx out of 77 ( 51/70 ) relative number vs strength in dBm which few will know if that is good or not ?

jgyates, 2018-05-20:
You can use the browser or the command line. My guess is that a permission was not right and the web update should work going forward. Let me know if you have any issues. I assume that your wifi signal is showing up now?

You can look at the top of the genmon.py file:
https://github.com/jgyates/genmon/blob/master/genmon.py#L28

I will look at adding signal quality, they really represent different things. You can have high signal strength and low quality or vice versa.  If you have good signal strength but bad though put then signal quality is a good indicator that there is interference  or something else going on to make the signal not as readable.

jgyates, 2018-05-20:
@proppilot ,

Try updating via the web interface , I added signal quality. If it does not work with the web interface post the last few lines from your genserv.log file. Current version is 1.8.7

steve-bailey1, 2018-05-20:
I think something went a little sideways with the latest (at least in my environment)

2018-05-20 15:33:08,230 : New version detected: Old = V1.8.4, New = V1.8.7
2018-05-20 15:33:08,322 : GenMon Loadded for site: Charles Street
2018-05-20 15:33:10,762 : Looking up model info on internet
2018-05-20 15:33:12,471 : Found: Model: G0062440
2018-05-20 15:33:25,629 : Error in DisplayMonitor: 'NoneType' object has no attribute 'error' : genmon.py:524
2018-05-20 15:33:27,702 : Error in DisplayMonitor: 'NoneType' object has no attribute 'error' : genmon.py:524

[image]

proppilot, 2018-05-20:
Yes the wifi shows now.

jgyates, 2018-05-20:
@steve-bailey1 , try it now. I was not passing a log variable so if there was an issue it could not log it to genmon.log. It should log it now so we can see what the problem is.

steve-bailey1, 2018-05-20:
Works now.

steve-bailey1, 2018-05-20:
Still has some errors in the log file but the Pi stats are showing again 

2018-05-20 18:02:38,712 : New version detected: Old = V1.8.7, New = V1.8.8
2018-05-20 18:02:38,808 : GenMon Loadded for site: Charles Street
2018-05-20 18:02:41,249 : Looking up model info on internet
2018-05-20 18:02:42,702 : Found: Model: G0062440
2018-05-20 18:02:50,088 : Error in GetLinuxInfo: need more than 1 value to unpack : myplatform.py:138
2018-05-20 18:02:52,929 : Error in GetLinuxInfo: need more than 1 value to unpack : myplatform.py:138

[image]

jgyates, 2018-05-20:
from the console type this: 

     cat /etc/os-release

What is the output?

steve-bailey1, 2018-05-20:
[root@genmon1 ~]# cat /etc/os-release
NAME="CentOS Linux"
VERSION="7 (Core)"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="7"
PRETTY_NAME="CentOS Linux 7 (Core)"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:centos:centos:7"
HOME_URL="https://www.centos.org/"
BUG_REPORT_URL="https://bugs.centos.org/"

CENTOS_MANTISBT_PROJECT="CentOS-7"
CENTOS_MANTISBT_PROJECT_VERSION="7"
REDHAT_SUPPORT_PRODUCT="centos"
REDHAT_SUPPORT_PRODUCT_VERSION="7"

jgyates, 2018-05-20:
ok. I see the problem. Give me a min to fix and test. My code does not like that blank line. Easy fix...

proppilot, 2018-05-20:
sig quality there now as well.  awesome :-)

jgyates, 2018-05-20:
@steve-bailey1 

Try it now. I don't have a CentOS box handy to test but it was a one line fix so it should work.

jgyates, 2018-05-20:
When you test, post the output of the monitor page so I will know that it is working on CentOS, there are a few things that are not displaying. Are you using wifi?

steve-bailey1, 2018-05-20:
Looks awesome.  I dont mind being your CentOS test guy.  Great job with this software! 

[image]

steve-bailey1, 2018-05-20:
It does not and it doesn't look like its easily available on yum package installers.  I will dig into it and see if its something simple to install

jgyates, 2018-05-20:
I might can pull it form /proc/net/wireless

steve-bailey1, 2018-05-20:
Thats probably the easiest way.  CentOS's tool is iw and it does not easily display quality.

[root@genmon1 ~]# cat /proc/net/wireless
Inter-| sta-|   Quality        |   Discarded packets               | Missed | WE
 face | tus | link level noise |  nwid  crypt   frag  retry   misc | beacon | 22
 wlan0: 0000   61.  -49.  -256        0      0      0      0      3        0

jgyates, 2018-05-20:
ok. Try it now. I added the noise level in there too.

jgyates, 2018-05-21:
Closing this thread. Feel free to comment to this thread if there are any issues on this topic.

closed 2018-05-21 by jgyates

zekyl314, 2018-06-01:
Any way to display Fahrenheit instead of Celsius as an option?

jgyates, 2018-06-01:
No presently, but I will put that on the list.

jgyates, 2018-06-03:
This feature has been added.

zekyl314, 2018-06-04:
You're awesome.