jgyates/genmon #115: Block heater status display enhancement?

griffths, 2018-05-19:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot on This Problem (if possible)

{Please write here}

### Your Environment

- Genereator Model: {Please write here}
- Generator Registers: {Please write here}
- Genmon Version: {Please write here}
# #
Requesting an enhancement to display the ON/OFF or actual temp status of a block heater and battery warmer within the UI.  I currently have a liquid cooled unit which has both but I am never sure when they are on or off.  I am not familiar with the Pi or programming in general but could something be accomplished using the GPIO pins if there were a way to detect whether or not power was on to each of the devices?  They are 120V on my unit but are switched ON/OFF with thermostatically controlled outlets.  I was thinking that a current transformer could be used or maybe even a thermistor to detect and report the actual temperature of the warming blanket and block heater?

jgyates, 2018-05-19:
Hi @griffths 

This is possible, however there are a few issues that come to mind.

1) It sounds like this is an accessory / add on item, so one would need to know more about the outlet control
2) The pi's GPIO are designed for reading digital inputs. Current Transformers are typically analog outputs so you would need to add an interface board of some type to the pi if you used a current transformer. If you did not use a transformer you would need to make sure the voltage levels of what you are trying to read are compatible with the pi, otherwise you would need an interface board of some type.
3) Genmon now has the ability to display custom, user supplied data in the web interface so that part is already taken care of
4) There are a lot of temp sensors available for the pi, some of them have python libraries that would make this relatively easy. I have not used any of them before, but I know some others who have.

In general the approach that is likely the easiest is to get a temp sensor that has a python library, write a small python program to read the sensor, write the sensor data to a file formatted in a JSON string. If the file is named "userdefined.json" in the genmon directory, the software will display the sensor data on the Monitor page.

Others may be able to weigh in on a recommended temp sensor or other interfacing options.

griffths, 2018-05-19:
Here are the specifics to my installation:

 

Temp controlled outlet: https://www.amazon.com/Farm-Innovators-TC-3-Thermostatically-Controlled/dp/B0006U2HD2

 

Block Heater: https://www.amazon.com/Generac-5616-Extreme-Weather-Generators/dp/B003UHW0US/ref=sr_1_7?s=lawn-garden <https://www.amazon.com/Generac-5616-Extreme-Weather-Generators/dp/B003UHW0US/ref=sr_1_7?s=lawn-garden&ie=UTF8&qid=1526760267&sr=1-7&keywords=generac+battery+warmer> &ie=UTF8&qid=1526760267&sr=1-7&keywords=generac+battery+warmer

 

Battery blanket:  https://www.amazon.com/Generac-5630-Weather-Liquid-Cooled-Generators/dp/B003UHW0TY/ref=sr_1_11?s=lawn-garden <https://www.amazon.com/Generac-5630-Weather-Liquid-Cooled-Generators/dp/B003UHW0TY/ref=sr_1_11?s=lawn-garden&ie=UTF8&qid=1526760267&sr=1-11&keywords=generac+battery+warmer> &ie=UTF8&qid=1526760267&sr=1-11&keywords=generac+battery+warmer

JustApu, 2018-05-20:
@griffths - While I don't have any DS18B20 sensors hooked up to my genmon Pi (yet), I have a network of them in my HVAC system so I can register supply & return duct temperatures around the house. The Amazon link you posted is basically what I use. I'm planning to use some of those sensors  to monitor my generator battery, engine block & exhaust temperatures. (I don't have a battery or block heater on my unit. It's more curiosity than anything else.)

See https://github.com/jgyates/genmon/issues/114 for a script that should work for getting the 1-wire data into genmon once you have your sensors connected to the Pi. I use a DS9490R for my HVAC monitoring Pi Model A but am planning to use a DS2482S for the genmon Pi Zero W to keep the footprint and adapter count down.

jgyates, 2018-06-06:
Closing this issue. Feel free to post if you still have questions.

closed 2018-06-06 by jgyates