jgyates/genmon #117: LCD blank after installing genmon device

danorum, 2018-05-27:
Has anyone else seen this? Genmon is functioning properly and the switch on the generator is as well, but the lcd on my generator is now blank. I didn't notice if it was that way last night when I plugged it in. Unplugging the monitor does not change the LCD being blank. Also, the green light on the side is still lit.

danorum, 2018-05-27:
Thanks. I removed the genmon and the LCD was blank. I then "rebooted" the controller by disconnecting the back connector on it. When I put it back in, it asked me to set the time (on the LCD). As soon as I plugged the monitor in, the display went blank again. This is a pi zero with a USB ethernet adapter.

jgyates, 2018-05-27:
How are you powering your pi? Also, I would double check your connections on the molex plug. If the symptom is repeatable you could remove a connection to the pi, one at a time, on the molex plug to see which one of them is causing the issue. Another thing you could do is to check the voltage on your battery if you have a volt meter. If you read a voltage below 12.6 you may have a battery that needs replacing.

danorum, 2018-05-27:
Powering the Pi from the molex connector.

Battery is reading 13.36.

If I use an external power supply on the Pi and just remove the red wire (so ground is still connected), is that ok?

jgyates, 2018-05-27:
Yes. That should be OK. As long as you have a ground for the pi from either the battery or the molex connector you should be OK, otherwise you will likely see communication errors on the serial link.

jgyates, 2018-05-27:
That would likely be the easiest fix. The pi zero over ethernet *should* be the lowest current draw for any of the pi models. I suppose you could have a bent pin on your connector. I power my pi from the 12V battery, but I know a lot of folks are also using the molex connector.  Let me know if you have any other issues.

danorum, 2018-05-27:
You can go ahead and close this issue. I think I may just run 3 wire thermostat from the molex to inside the house. Then I can put the Pi inside, which makes a lot of this easier (no weatherproofing of the device needed and I can use a regular plug in power supply). Thanks again for talking through this with me. I did make the troubleshooting faster.

jgyates, 2018-05-27:
no problem.

closed 2018-05-27 by jgyates

smeisner, 2018-06-28:
@danorum, were you using a Nexus controller? I am running into the same thing that you described (RPi Zero with USB wifi adapter, powered from the Molex connector on the Nexus controller) and my display is not working. The backlight will come on, but that's it. I tried pulling the connectors I could see (I missed the one(s) in the back) and it did not fix the display.

I have moved the power connector to the battery instead of the Molex connector. I also moved the USB wifi adapter into the pocket on the outside of genset where the original Generac one used to sit. Reception is now great!

danorum, 2018-06-29:
@smeisner - Yes, a nexus controller. There are 2 big black connectors and I had to disconnect both at the same time to "reboot" the controller. One of the times I did it to confirm the fix, I did it in an order that caused the generator to start up. So, if I was going to do it again, I would switch the generator to off first. Once I did that the display came back on and my heart restarted. Disconnecting the battery didn't do it because it was still getting power from the house.

smeisner, 2018-07-02:
@danorum, thank you, thank you!!! I had tried to reset the controller a couple of times and failed to get the display back. But tonight I had a few moments to try it your way and the display is working again!! I have been looking at the cost of a new controller (mine was $462 for a new one). I am so happy I got the display back!

jonthompson, 2020-01-27:
Reviving an old thread but just want to echo - I also have a nexus controller and wired my pi zero W in today via molex and the pi booted up and connected fine - but then i noticed my screen was lit but blank.  After a similar panic to above - and trial and error - the blue-and-white pair powers my controller and resetting the controller without the pi attached fixed the screen.  I'll go with the same 12v battery power connector everyone else has been using to avoid additional issues.

jgyates, 2020-01-27:
@jonthompson , thanks for the data point