jgyates/genmon #118: Genmon not working - Assistance with troubleshooting

kalex1114, 2018-05-29:
So I finally made a cable and connected everything.  problem is I'm getting no data from the generator.  
I'm using Air cooled 16KW model G0070370 Prod date 20160901.  not sure how to get firmware i'm on from the controller.  

I'm running rasbian stretch lite on raspberry pi zero w.  I followed the guide for software installation.  I'm able to start genmon and get to the web gui but I'm not getting any data from generator.
For cable I'm using ATX 8 pin molex cable and I soldered extension cable in order to reach the raspberry.  I disconnected the mobile link connector and connected this one into its spot.  
Here is how i wired everything:
https://github.com/jgyates/genmon/issues/84

I was able to test serial loopback sucessfully.  But i did run into this issue:
https://github.com/jgyates/genmon/issues/105#issuecomment-387185574

My serial shows this:
pi@genmon:/var/log $ tail myserial.log 
2018-05-07 15:59:59,981 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2018-05-07 16:00:00,300 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2018-05-07 16:00:05,376 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2018-05-07 16:00:05,793 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2018-05-07 16:00:06,056 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2018-05-07 16:00:08,257 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2018-05-07 16:00:08,574 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2018-05-07 16:00:08,599 : Error in SerialDevice:Flush : /dev/serial0:Attempting to use a port that is not open
2018-05-29 14:33:08,497 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2018-05-29 14:33:08,968 : Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?)

mymodbus.log shows this:
pi@genmon:/var/log $ tail mymodbus.log
2018-05-29 14:39:53,673 : Error: timeout receiving slave packet for register 053 Buffer:0
2018-05-29 14:39:56,686 : Error: timeout receiving slave packet for register 052 Buffer:0
2018-05-29 14:39:59,694 : Error: timeout receiving slave packet for register 01 Buffer:0
2018-05-29 14:40:02,701 : Error: timeout receiving slave packet for register 5f1 Buffer:0
2018-05-29 14:40:05,713 : Error: timeout receiving slave packet for register 09 Buffer:0
2018-05-29 14:40:08,719 : Error: timeout receiving slave packet for register 5f5 Buffer:0
2018-05-29 14:40:11,726 : Error: timeout receiving slave packet for register 5f4 Buffer:0
2018-05-29 14:40:14,735 : Error: timeout receiving slave packet for register 5f3 Buffer:0
2018-05-29 14:40:17,745 : Error: timeout receiving slave packet for register 055 Buffer:0
2018-05-29 14:40:20,759 : Error: timeout receiving slave packet for register 036 Buffer:0

Is this a bad cable issue?  Anything i can do to narrow down the troubleshooting?

Pictures of wiring:

https://photos.google.com/share/AF1QipOdhrOF-KJpiICKgap8feLTBYCMHeok9zj6lbfzAu5M7JwhJds3HGybQhjFfTm5Sw?key=dmUzYjFYckhhTEZLbFZJSmtiUVRiSENwc09FTU9n

Thank you

Alex

jgyates, 2018-05-29:
HI @kalex1114 

Couple of questions:

1) Do you still see the intermittent issues described here: https://github.com/jgyates/genmon/issues/105#issuecomment-387185574 .If you still see any issues when running serialtest.py we need to resolve those before trouble shooting genmon.py. 

2) If you still have problems with serial test. Do you see the problems if you use your cable to loopback TX to RX? Do you see problems if you loopback at the Raspberry pi header (i.e. not using the cable)?

3) Have you validated that all steps listed [here](https://github.com/jgyates/genmon/wiki/3.2--Setup-Your-Pi-and-Serial-Port) were performed?

4) If you have a multimeter. Can you validate you have continuity between your battery negative terminal and the ground pin on the raspberry pi header?

5) What is the output of this command: 

     vcgencmd get_throttled

The error " Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?)" is generally caused by either some other software using the port or a power issues.

jgyates, 2018-05-29:
Also, can you post a picture of the Monitor page from the web interface?

kalex1114, 2018-05-30:
wow I feel stupid. Looks like i missed the last part. 
console=serial0,115200

After removing and rebooting it looks like its working:

monitor screenshot:
[image]

status page:
[image]
 

Sorry about false alarm

jgyates, 2018-05-30:
No worries. I am glad you got it working. We've all been there, missing some small thing...

the version you are running is a bit dated, current version is 1.9.6

closed 2018-05-30 by jgyates