jgyates/genmon #119: Email issues

ecadet18, 2018-05-29:
I see the following in the mymail.log and email notifications don't work.  What seems to be the issue?
2018-05-29 17:30:07,708 : Error SMTP Init : SSL:<True>: [SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:661)
2018-05-29 17:30:07,710 : Error in SendMailThread, sendEmailDirectMIME failed, retrying
2018-05-29 17:30:43,615 : LOGIN FAILED!!!

jgyates, 2018-05-29:
I don't have enough info to answer your question. Depending on your mail settings and possibly your email provider, I would guess you have one of the following problems:

1) If using gmail, your SMTP Server Port if using gmail should be 587 and SMTP Server SSL Enabled setting should be off
2) In general if your email provider uses TLS instead of SSL for email encryption use port 587, if you have not encryption use port 25, if you use SSL use port 465
3) It could be and issue with your email provider security settings, but that is very specicific to your email provider.

ecadet18, 2018-05-29:
I use iCloud, and have it set for port 587

jgyates, 2018-05-29:
According to this page (https://support.apple.com/en-us/HT202304) you need to setup and application specific password. Have you done that and are you using the smtp mail server listed in the link?

ecadet18, 2018-05-29:
I have them set correctly and re-checked the password to make sure it's typed correctly

ecadet18, 2018-05-29:
[image]

jgyates, 2018-05-29:
Is two factor authentication enabled for your icloud account?

ecadet18, 2018-05-29:
yes

jgyates, 2018-05-29:
If you have generated an application specific password and it it is not working I would suggest creating an account on another mail provider to use for sending email with the generator monitor. Gmail is known to work well so I would recommend that gmail. I will put icloud on the list of things to look into.

ecadet18, 2018-05-29:
I generated a new app specific password and now I am getting the following:
2018-05-29 18:38:49,848 : Error selecting mail folder! (select)
2018-05-29 18:39:04,765 : Error SMTP Init : SSL:<True>: [SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:661)
2018-05-29 18:39:04,767 : Error in SendMailThread, sendEmailDirectMIME failed, retrying

jgyates, 2018-05-30:
OK. I was able to turn on two factor authentication for my apple account, create and icloud email and generate an application specific password. I tested this with SMTP port set to 587 and SSL disabled. Also, make sure sender account and are both set to your icloud email address.

Based on your settings above you should disable SSL, assuming your password is correct, it should work. 

In my test I can successfully send email from my icloud account with the software.

ecadet18, 2018-05-30:
I am no longer getting "Login failed" messages, but I am still getting the following.  How do you test from the software?

2018-05-30 07:35:16,267 : Error SMTP Init : SSL:<True>: [SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:661)
2018-05-30 07:35:16,269 : Error in SendMailThread, sendEmailDirectMIME failed, retrying

jgyates, 2018-05-30:
Hi @ecadet18 ,

Please post a picture of you settings. without your settings the error code has not context.

ecadet18, 2018-05-30:
The Settings page of the GenMon software or please specify which settings?

jgyates, 2018-05-30:
Yes, the genmon software. I know of any other settings that we have discussed in this thread.  In this thread you posted your email settings earlier. I am asking for them again to see what the values are now.

ecadet18, 2018-05-30:
[image]

ecadet18, 2018-05-30:
That worked, thank you.  I just got 3 emails.

I see a newer version is available, but I use the Update tool to upgrade, it went thru the process of downloading, but it came back with the same version 1.9.2.  Is it worth upgrading and should the process work?

jgyates, 2018-05-30:
Great, I am glad it is working for you. 

The software update will work if you installed the software with the instructions listed on the site. If you  downloaded the software via a method not listed in the installation documentation then it will not work. Specifically, you must have  installed the software the "git" command listed on this page:

https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

if you did not you can fix this but it will require you to log into your raspberry pi and run a few commands. if you log into your pi and type this command:

      git --version

If this command works it should return the version of the git program you are using, for example it would return something like this:

      git version 2.1.4

If that works then change directories into the genmon directory:

     cd genmon

Then type the following command:

     bash ./genmonmaint.sh update

If this works, then type:

     bash ./startgenmon.sh restart

If all of these commands work with not errors you should be updated to the latest software. If you have any errors you can post them to this thread and we can take it form there.

ecadet18, 2018-05-30:
That worked. I upgraded to 1.9.6. 

I see the following on the Status Page about unsupported sensors.  I have a Generac 22KW Model 7043.  Is this an issue?

            Unsupported Sensors :
                Raw RPM Sensor : 0
                Frequency (Calculated) : 0.0 Hz
                Calibrate Volts Value : 1008
                kW Hours in last 30 days : 0.00
                Calibrate Current 1 Value : 1483
                Calibrate Current 2 Value : 1537
                Unsupported Sensor 1 : 0.00
                Unsupported Sensor 2 : 2600
                Unsupported Sensor 3 : 10
                Unsupported Sensor 4 :

jgyates, 2018-05-30:
No, this is not an issue, this is normal. You have "Display Unknown Sensors" enabled the settings page. When enabled the software will display information that is not officially supported. The reason it is not supported varies depending on the model, but typically I put info here if it is either 

a) not immediately useful
b) redundant
c) Not fully tested on all system

If you don't want this info displayed then disable "Display Unknown Sensors"  on the settings page. 

I noticed that you have "Do Not Check for Outages" enabled on your settings page. Since you have this enabled you will not receive notifications if the generator detects and outage. If you want to be notified by email if the generator detects a loss of utility power you should disable this setting. It is intended for people with a "smart" transfer switch. 

It appears that most of the settings on your settings page were enabled, were the default values are typical for most people. If you hold your mouse over each control on the settings page it will display text that describes each entry.

closed 2018-05-31 by jgyates