jgyates/genmon #120: genmon wont run

ewouk61, 2018-05-31:
i am getting an error when i try to run sudo/home/pi/genmon/startgenmon.sh start on the pi.
it is a nexus controller, pi3B, raspbian stretch with desktop

i@raspberrypi:~ $ sudo /home/pi/genmon/startgenmon.sh start
Starting genmon python scripts
Traceback (most recent call last):
  File "/home/pi/genmon/genmon.py", line 818, in <module>
    MyMonitor = Monitor(ConfigFilePath = ConfigFilePath)
  File "/home/pi/genmon/genmon.py", line 117, in __init__
    self.mail = mymail.MyMail(monitor=True, incoming_folder = self.IncomingEmailFolder, processed_folder =self.ProcessedEmailFolder,incoming_callback = self.ProcessCommand)
  File "/home/pi/genmon/genmonlib/mymail.py", line 55, in __init__
    self.GetConfig()
  File "/home/pi/genmon/genmonlib/mymail.py", line 123, in GetConfig
    self.FatalError("ERROR: Unable to read config file " + str(e1))
  File "/home/pi/genmon/genmonlib/mycommon.py", line 87, in FatalError
    raise Exception(Message)
Exception: ERROR: Unable to read config file No section: 'MyMail'
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
Exception in thread InterfaceServerThread:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/genmon/genmon.py", line 756, in InterfaceServerThread
    self.ServerSocket.bind(('', self.ServerSocketPort))
AttributeError: 'Monitor' object has no attribute 'ServerSocketPort'
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs

    func(*targs, **kargs)
  File "/home/pi/genmon/genmon.py", line 792, in Close
    self.ServerSocket.shutdown(socket.SHUT_RDWR)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 107] Transport endpoint is not connected
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/home/pi/genmon/genmon.py", line 792, in Close
    self.ServerSocket.shutdown(socket.SHUT_RDWR)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 107] Transport endpoint is not connected
Exception in thread ProcessThread (most likely raised during interpreter shutdown):Exception in thread SerialReadThread (most likely raised during interpreter shutdown):
Traceback (most recent call last):
Traceback (most recent call last):
pi@raspberrypi:~ $ Error: genmon not loaded.
pi@raspberrypi:~ $ sudo python /home/pi/genmon/OtherApps/serialtest.py

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful

- Genereator Model: 0058851
- Generator Registers: 
- Genmon Version: 1.9.5

This is the first thing I have ever done with a pi, so please excuse the noob questions that will likely follow.

jgyates, 2018-05-31:
According to this log, genmon is failing because it can not find the file mymail.conf in the /etc/ directory. 

      Exception: ERROR: Unable to read config file No section: 'MyMail'

Have you copied this file here? The install script will copy them or you can copy them your self. This information is in the directions for install.

ewouk61, 2018-06-01:
ok i used the  bash ./genmonmaint.sh install  to get the library's. can you add the "copy them into the /etc directory" info to the 3.3 page of the wiki, that is what i used instead of the 3.4. or point to where we should pick up in 3.4?
after some trial and error i got it working following the instructions of 3.4.
i also have access to industrial gens so i will go hop into there once i get that section working on my pi. thanks for the help.

jgyates, 2018-06-01:
the genmonmain.sh script prompts you if you would like to copy the files to the /etc directory. If you select no, it will not copy the files, if you select yes, it will.

Sound good on the industrial gen.

closed 2018-06-01 by jgyates