jgyates/genmon #121: gensms TLS error

krisasmith, 2018-05-31:
I built a new raspberry pi running stretech os. I cannot get gensms.py to function.

### Expected Behavior

gensms sends text notifications

### Actual Behavior

no notifications are sent

### Steps to Reproduce (including precondition)

sudo python /home/pi/genmon/gensms.py

### Screenshot on This Problem (if possible)

pi@raspberrypi:~/genmon $ sudo python /home/pi/genmon/gensms.py
From cffi callback <function _verify_callback at 0x754fdd70>:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 309, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
Error: HTTPSConnectionPool(host='api.twilio.com', port=443): Max retries exceeded with url: /2010-04-01/Accounts/AC0b6470b7eda60c19b6a96e371c04a4b2/Messages.json (Caused by SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),))

jgyates, 2018-05-31:
Try running this command:
    
    sudo pip install cryptography --upgrade 

If this works I will add a note to the wiki. Looks like this thread deals with a similar issue:

https://www.raspberrypi.org/forums/viewtopic.php?t=204149

darkstar2002, 2018-05-31:
I have not tried the sms yet was going to test that next.    I ran that command above and get errors trying to compile for wheel and a few others. 
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-yk9y6D/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-MNld96-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-yk9y6D/cffi/

jgyates, 2018-05-31:
I installed twillo on Jessie

     sudo pip install twillo

Got this error:

    RuntimeError: cryptography requires setuptools 18.5 or newer, please upgrade to a newer version of setuptools

Then I ran this:

    sudo pip install setuptools --upgrade
    sudo pip install twillo

Then I get the same error as @darkstar2002 . Looks like twillo has some issues with their library not working with some of the support libraries it relies on.

@krisasmith can you check the version of twillo on your old system?

      pip freeze | grep "twillo"

Then run the same command on your new system and let us know the versions of each.

jgyates, 2018-05-31:
This at least lets me install, have not tested yet:

    sudo apt-get build-essential
    sudo apt-get libssl-dev
    sudo apt-get libffi-dev 
    sudo apt-get python-dev
    sudo pip install cryptography --upgrade 
    sudo pip install twillio

    pip freeze | grep "twilio"
  
    returns this: twilio==6.14.3

krisasmith, 2018-05-31:
This is on my old genmon pi running jessie.

```
pi@genmon:~ $ pip freeze | grep "twilio"
twilio==6.1.0
```

krisasmith, 2018-05-31:
Tried to update cryptography and got this error / response.

```
pi@raspberrypi:~ $ sudo sudo pip install cryptography --upgrade
From cffi callback <function _verify_callback at 0x76c4f430>:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 309, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
From cffi callback <function _verify_callback at 0x75928070>:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 309, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
Requirement already up-to-date: cryptography in /usr/lib/python2.7/dist-packages
pi@raspberrypi:~ $

```

jgyates, 2018-05-31:
DId you run these commands first?

    sudo apt-get build-essential
     sudo apt-get libssl-dev
     sudo apt-get libffi-dev 
     sudo apt-get python-dev

jgyates, 2018-05-31:
sorry, should be 

    sudo apt-get install build-essential
    sudo apt-get install libssl-dev
    sudo apt-get install libffi-dev 
    sudo apt-get install python-dev

jgyates, 2018-05-31:
The solution for me is: 

   sudo pip install -U pyopenssl

I don't think all that other stuff is needed, only the update to pyopenssl.

I validated this on a stretch system.

krisasmith, 2018-06-01:
I tried all kinds of stuff and kept coming back to the same problem: cryptography was at the wrong version. no matter what i tried it failed to upgrade. i ended up rebuilding the pi and step one after installing python-pip was to upgrade cryptography and everything is working now.

```
sudo apt-get install python-pip
sudo apt-get install build-essential libssl-dev libffi-dev python-dev
sudo pip install -U cryptography
sudo pip install twilio
pip check twilio
```

closed 2018-06-01 by krisasmith