jgyates/genmon #122: Multilevel Password

darkstar2002, 2018-05-31:
Would it be hard to have a dual level password?   One password that would only let you have view only and then another password that has the change information and control of the start stop button.  
Thanks

jgyates, 2018-05-31:
Let me think about the best way to implement that. From a web perspective it should be doable, I just need to think about the implementation details.

danorum, 2018-05-31:
I was just thinking about security and genmon today. I am going to set up 3 of these for family and me. Here are a couple security options that would be cool.

Make the portal read only if you are coming from outside the local network.
Disable any email commands that make changes. (basically, any of the set* commands - so you can request status, but can't make start the generator/change exercise time/etc).

I do like the idea of multiple "users" or passwords for different levels.  I think the 2 other people using this at home might be interested in checking status remotely from their phones, but would be wary of allowing changes to be made. Prevents any issues with the bad passwords they might pick. 

Dan

madisdev, 2018-05-31:
Dan,

Are you talking about having the webpage accessible to the general internet?

danorum, 2018-05-31:
If I could set the portal to be read only (even with any username/password), I would consider it. I think for my house, I am going to use VPN and then look at the page from my phone. I was thinking that email commands that only allowed read only operations would make this interesting to the other family members with generators.

jgyates, 2018-05-31:
The email commands thing is easy. I just check in version 1.9.7. With this versions you can add this to your /etc/genmon.conf file:

readonlyemailcommands = True

As far as making the web portal available on the internet, I would recommend a VPN solution over making the web interface visible to the public internet. The web interface is based on Flask, which allows python integration. Flask can be used as a stand alone server or with a full blown web server such as apache. Flask as a stand alone web server is not intended to be used publicly as it does not have the same security internals as apache. For an internal server it is fine. They people who developed the flask framework discourage its use  as a production web server. Of course you can go through the process of setting up flask to be used with apache on the raspberry pi  however, this is something that I have not done to date.

griffths, 2018-05-31:
I apologize in advance to all users subscribed to this thread but cannot find any other way to contact @danourm to get my question answered so……

 

Is this the same Dan Orum from Microsoft , U of I and Wordlink?

 

If so, small world and shoot me an email at my usual address and lets catch up!

 

Back to the regularly scheduled stuff with my apologies again to everyone else…..

 

Regards,

 

Jay

jgyates, 2018-06-05:
The multi-level password stuff is checked in but the UI additions are not implemented yet. If you want to test this out you can setup a https, username and password with the web interface, then add these two entries to your genmon.conf file:

     http_user_ro = limitedusername
     http_pass_ro = limitedpassword

changing "limitedusername" and "limitedpassword" to the username and password of your choice.

Adding an option for serving the web pages with apache is on the list of things to do, but it may be a month or 2 before I get to it. This would make it safer for an internet facing situation, however you would still need to update your OS software regularly to keep up to date with security patches.

jgyates, 2018-06-05:
Just to clarify, if you enable this an login with the limited user setting the buttons and controls will still show up, but they will not do anything. Once the web UI stuff is complete the buttons and controls will not be visible (or the settings pages) for limited users. Since a limited user can currently see the settings page I would not roll this out to any limited users yet. This is only for those who want to test.

darkstar2002, 2018-06-05:
Looks like its working.
As for Apache it is pretty simple to setup a reverse proxy.  I have done it on a few things before.

jgyates, 2018-06-09:
FYI, V1.9.23 is checked in, it updates the web UI to include limited rights user/password.

darkstar2002, 2018-06-11:
Looks good.

closed 2018-06-11 by jgyates