jgyates/genmon #123: Rerun Settings Lookup

beschaaflh, 2018-06-01:
Is it possible to force genmon.py to re-run the initial settings lookup?  I was testing genmon before it was connected to my generator and the settings are incorrect.  Thanks for your help -- this program is great!

# The following entries are written by genmon.py based on the generator 
# settings, serial number and a one time lookup on the internet of the 
# serial number. If you are not connected to the internet default 
# settings will be used based values read from the generator. Some 
# of these settings are editable via the web interface.
#    fueltype, model, nominalRPM, nominalKW, nominalFreq

jgyates, 2018-06-01:
You can force the software to do the lookup again in two ways.

1) If you upgrade to a new version the lookup will be performed again
2) Trick the software into thinking it has been upgraded by editing the /etc/genmon.conf file to reduce the version stored in the version entry 

Change from:

      version = V1.9.7

To someting lower: 

     version = V1.9.0

beschaaflh, 2018-06-01:
Thanks.  I tricked the software and that did it.  The one setting that isn't correct is fuel type.  My generator is LPG but shows natural gas.  Obviously a very minor item.  Thanks again for the great support!

jgyates, 2018-06-01:
Presently I don't know a way to read the fuel type so you need to set that in the settings page.

closed 2018-06-01 by jgyates

beschaaflh, 2018-06-01:
Perfect. Thanks again.