jgyates/genmon #124: Not getting data from the generator

jcherney, 2018-06-01:
Hello - I just finished setting everything up and I'm not receiving data from my liquid cooled generac QT048.

I'm using raspberry pi 3 running stretch lite.  Installed the latest version of genmon.
serial port test passed without errors.
Tested continuity between battery negative and Pin 6 on the PI3 

monitor screenshot attached.
[image]

I verified all the serial port settings on the wiki page as well.  made sure to remove console redirect.
Not sure what else I can check.

mymodbus.log entries:
pi@raspberrypi:/var/log $ tail mymodbus.log
2018-05-31 21:20:45,085 : Error: timeout receiving slave packet for register 052 Buffer:1
2018-05-31 21:20:48,087 : Error: timeout receiving slave packet for register 01 Buffer:1
2018-05-31 21:20:51,092 : Error: timeout receiving slave packet for register 5f1 Buffer:1
2018-05-31 21:20:54,104 : Error: timeout receiving slave packet for register 09 Buffer:1
2018-05-31 21:20:57,108 : Error: timeout receiving slave packet for register 054 Buffer:1
2018-05-31 21:21:00,118 : Error: timeout receiving slave packet for register 02b Buffer:1
2018-05-31 21:21:03,127 : Error: timeout receiving slave packet for register 0e Buffer:1
2018-05-31 21:21:06,136 : Error: timeout receiving slave packet for register 0f Buffer:1
2018-05-31 21:21:09,147 : Error: timeout receiving slave packet for register 021 Buffer:1
2018-05-31 21:21:12,155 : Error: timeout receiving slave packet for register 02f Buffer:1

any suggestions on what else this could be?

Thank you

Jack

jgyates, 2018-06-01:
Hi Jack,

Were you able to perform a loopback test by connecting tx and rx at on your cable (i.e. looping back through your cable when you cable is disconnected from the generator). Use the program serialtest.py

https://github.com/jgyates/genmon/wiki/1----Software-Overview#otherappsserialtestpy-optional

Let me know if you have and what the result of the test is.

JustApu, 2018-06-01:
`serial port test passed without errors.`

@jcherney  - note that serialtest.py can check that the Pi can talk to/through the TTL/232 adapter (where you made your loopback connection). But you could still have TX/RX reversed between your adapter and the generator. Been there, done that.

jcherney, 2018-06-01:
@jgyates- haven’t done that test, thanks. Will try tomorrow and advise. 

@justapu- have switched those wires back and forth so many times they are wearing thin! I’ll keep doing so to get this resolved, thanks.

jgyates, 2018-06-01:
@jcherney , once we get this up and running, if you don't mind I would like to get a copy of your registers since I don't have a QT048, you can send them to me on the Monitor page in the web interface. Let me know if you have any questions.

jcherney, 2018-06-01:
I Would be happy to provide that info

jcherney, 2018-06-02:
ok just did a test on my cable.  Just to double check i remove the molex connection from the generator and jumpered pins 7 and 8 on the molex end.  then ran serial port test script.  Is this the correct way of testing loopback?
looks like its sending but not 
pi@raspberrypi:~/genmon/OtherApps $ sudo python serialtest.py

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 0 bytes
pi@raspberrypi:~/genmon/OtherApps $

so it looks like cable is not working.   Going to cut the molex and just twist both cables together to see if that helps.

jcherney, 2018-06-02:
just tested without molex and got the same error:
pi@raspberrypi:~/genmon/OtherApps $ sudo python serialtest.py

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 0 bytes
pi@raspberrypi:~/genmon/OtherApps $

So so far loopback on gpio pins is working but these do not.
does this point to a bad ttl converter and/or bad rs232 breakout?

jgyates, 2018-06-02:
It could possibly be a bad converter, bad break out or incorrect wiring of the convert / breakout. Pictures of your wiring would help.

Do the LEDs on your converter light up? If memory serves, there is one for power, tx and rx.

jcherney, 2018-06-02:
How do I post pics on this board?

jgyates, 2018-06-02:
In the text at the bottom of the window you are typing in there are directions on how to post pictures.
[image]

jcherney, 2018-06-02:
I guess that doesn’t show up on an IPhone.  I’ll do not from my desk top. Thanks.

jcherney, 2018-06-02:
[image]
[image]
[image]

The other end of the green and yellow wires are stripped and twisted together.

jgyates, 2018-06-02:
It is hard to tell from the angel of the pictures but you may have both power and ground on the converter tied to power on the pi.

[image]

Try connecting VCC to 3.3V (pin 1 on the pi). It should work on either 5V or 3.3V, but 3.3V is probably a little safer to have it on 3.3V.  Connect ground to pin and Ground to pin 6 on the pi header. pin 8 on the pi goes to pin TDX on the converter, pin 10 on the pi goes to RDX on the converter. 

If that checks out and the loopback still does not work you may want to make sure you connections on the screw terminal are good, a multi-meter to check continuity between the two screws on TX and RX on the green terminal screws should validate that.

jcherney, 2018-06-02:
Getting closer! Ran Serialtest.py and got “Sent data does not match receive. Received 15 bytes”

jgyates, 2018-06-02:
Depending on how recent your code is you can run this test:

     sudo python ./OtherApps/serialconfig.py -c

If it don't have this program then update your software with this command:

     bash ./genmonmaint.sh update

The output of serialcheck.py should look like this:

     Checking : Serial console command line removed: OK
     Checking : Is enable UART in boot config: OK
     Checking : Is BT overlay disabled: OK
     Checking : Serial console service disabled: OK

     Serial port settings are OK. A reboot is needed if changes were made.

jcherney, 2018-06-02:
Looks like I have to update the program. I'll do so and advise.. Thank you.

jcherney, 2018-06-02:
Just updated and ran serialcheck.py  Everything shows as OK

jcherney, 2018-06-02:
Is there a reason the serialtest.py showed a problem before updating, and then not after? Do I still have an issue? (I havent rebuilt my molex yet).

jgyates, 2018-06-02:
I would need to know more info about the problem it is showing. The following command:

     sudo python ./OtherApps/serialconfig.py -c

will just check your settings. I am just trying to determine if you have something else trying to use the serial port. You can take a screen shot and post it here or copy and past the output here if you are accessing your system remotely. We have not changed anyting so I would assume your system still has the same issue. Does the loopback test still fail? If you are accessing your system via ethernet or wireless  or with a keyboard and monitor?

jcherney, 2018-06-02:
System is accessed via wifi, using my desktop keyboard and monitor.

jgyates, 2018-06-02:
ok. just copy and past the output of the problem you saw with serialconfig.py if you still have that on your ssh console. Also lets see the output of this command:

     ls -la /dev/ser*

jcherney, 2018-06-02:
pi@raspberrypi:~/genmon $ sudo python ./OtherApps/serialtest.py

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
error communicating...: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

Now I'm not getting any response. Did I run this correctly?

jgyates, 2018-06-02:
I need to see the output of this command:

     ls -la /dev/ser*

It should look like this:

[image]

If serial0 is not linked to ttyAMA0 then your port is not on serial0. It could be on serial1, depending on how you setup your port.

If you pi serial port is not on /dev/serial0 then you may see this error or if your serial port is not configured properly you could see this error.

jcherney, 2018-06-02:
See above 4 posts.

jgyates, 2018-06-02:
OK, got it. looks like serial0 is indeed pointing to ttyAMA0. That is how it is suppose to be. 

1) Do you have a multimeter?
2) How are you powering your pi, form the battery? If your ground connection is not making a good connection then you could have power issues on the pi, which will make all sorts of things have odd results. If you have a multimeter you can test continuity between the ground on the pi and the ground terminal on your batter.
3) When you ran serialtest.py and got "device reports readiness" was this with the converter or loopedback at the pi connector or thought the cable? Please include these details as it helps with keeping the details of what is working and what is not straight.
4) Let's fall back to the loopback test at the pi connector. What is the result of that with serialtest.py

jcherney, 2018-06-02:
1. I do have a multimeter.
2. Pi is powered from the generator battery. We (Kalex is helping me immensely!) did the ground test yesterday and continuity tested fine. 
3. The cable (yellow and green wire) are twisted where the molex was (I cut and stripped them). The other end are attached to the converter. 
4. How do I run that?

jgyates, 2018-06-02:
Disconnect the converter board
Connect pin 8 to pin 10 on the raspberry pi connector
Run this test:

     sudo python ./OtherApps/serialtest.py

Also, let's see the output of this command:

     vcgencmd get_throttled

jcherney, 2018-06-02:
Do you want the output after I disconnect, or before?

jgyates, 2018-06-02:
we are going to run three tests with serialtest.py.

1) looping back tx to rx at the raspberry pi header with no converter attached (loop pin 8 to 10).
2) looping back tx to rx at the green terminal connector with the converter attached (looping back Tx to Rx on the terminal connector)
3) looping back tx and rx at the molex cable with the converter and cable attached.

Run these in order, if any of these fail, then stop. Run each test 5 times in a row to make sure we have consistent results.

jcherney, 2018-06-02:
Ok, jumped 8 & 10 and got this:

pi@raspberrypi:~/genmon/OtherApps $ sudo python serialtest.py

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
error communicating...: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

jcherney, 2018-06-02:
Ran first one 4 times with same output as above.

jgyates, 2018-06-02:
OK. Send another picture of your connections. I can't tell from your original pic what pin you connected to your pi. Maybe angle you camera a bit instead of a shot directly above..

jgyates, 2018-06-02:
also what is this output:

     vcgencmd get_throttled

jcherney, 2018-06-02:
pi@raspberrypi:~/genmon/OtherApps $ vcgencmd get_throttled
throttled=0x0

jcherney, 2018-06-02:
[image]
[image]
Do these work?

jcherney, 2018-06-02:
Those pics were taken yesterday before you had me move pin 2 to pin 1.

jcherney, 2018-06-02:
[image]

jcherney, 2018-06-02:
I'll take a pic of the loopback also in a minute.

jgyates, 2018-06-02:
Since you initially setup your serial port my modifying /boot/config.txt and /boot/cmdline.txt, have you rebooted your system?

krisasmith, 2018-06-02:
Check /boot/cmdline.txt for “console=serial0,115200", I had the same serial test errors and this was why.

jcherney, 2018-06-02:
Did it reboot after updating?

jgyates, 2018-06-02:
good point Kris,

type this:

     cat /boot/cmdline.txt

jcherney, 2018-06-02:
pi@raspberrypi:~/genmon/OtherApps $ cat /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=672f8420-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

jcherney, 2018-06-02:
[image]
[image]

jgyates, 2018-06-02:
No, did you manually reboot you pi after modifying /boot/config.txt and /boot/cmdline.txt

jgyates, 2018-06-02:
OK. Let's try this.

1) reboot your pi. It is not acting like the serial port settings are in effect. Once it comes backup log back in to you pi
2) Run the loopback test, looping back pin 8 to 10 on the pi connector, but remove all other connections

What are the results of this test?

jcherney, 2018-06-02:
pi@raspberrypi:~/genmon/OtherApps $ sudo python serialtest.py

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 94 bytes

jcherney, 2018-06-02:
Ran it again and got this, again:

pi@raspberrypi:~/genmon/OtherApps $ sudo python serialtest.py

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
error communicating...: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

jcherney, 2018-06-02:
Pin 8&10 are jumped

jcherney, 2018-06-02:
Let me test the ground again, just to make sure. Output is not consistent.

jgyates, 2018-06-02:
ok. I think I know at least one of the problems. I am guessing that you already have genmon loading from your crontab, we need to stop genmon to run serialtest.py.

Type this command from the genmon directory:

    bash ./startgenmon.sh stop

Then type this:

   python ./OtherApps/serialtest.py .   

If you reboot, genmon will load again if you already did that step so we need to make sure it is stopped when running serialtest.py

jcherney, 2018-06-02:
Well, that made a difference:

pi@raspberrypi:~/genmon $ python ./OtherApps/serialtest.py

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful

jcherney, 2018-06-02:
BTW- ground tested and good.

jgyates, 2018-06-02:
Yes, let's run the loopback test with the converter connected, looping back the Tx and Rx  on the terminal connector.

jcherney, 2018-06-02:
Failed: sent data does not match receive. Received 0 bytes

jcherney, 2018-06-02:
I’m using one of the wires for this loop back test. I pulled #8 from the pi. Is that an issue?

jgyates, 2018-06-02:
I don't know what that means. If it means that pin 8 has nothing connected to it then yes. That will cause things to fail the loopback test. Describe or show a picture of how you have everything wired when running this test.

jcherney, 2018-06-02:
Well, 8 is empty so I will have to find another way to jump.

jcherney, 2018-06-02:
[image]
[image]
Still no data received

jgyates, 2018-06-02:
I am not following you. You said 8 was empty then you posted a picture of you pi with 8 connected. Please understand that I can not help if I don't know exactly what you are doing. More details in your description are needed for me to help you.

[37 more replies]

closed 2018-06-03 by jgyates