jgyates/genmon #126: not receiving data

bzuelly, 2018-06-04:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

### Expected Behavior

{Please write here}

### Actual Behavior

not receiving any data on monitor tab or any tabs

### Steps to Reproduce (including precondition)

I have ran loopback test thru cable with TX & RX jumpered as well as a loopback test thru the converter w/o the cable.  All is well, but no data received.

mymodbus.log shows "Error: timeout receiving slave packet for register" over and over with different buffers.  I also ran serialconfig.py -c and all is ok.

### Screenshot on This Problem (if possible)

{Please write here}

### Your Environment

- Genereator Model: 22kw, nexus air cooled
- Generator Registers: {Please write here}
- Genmon Version: 1.9.13

jgyates, 2018-06-04:
If you loopback test is passing, you could have your transmit and receive crossed between the DB-9 connector and the molex / controller end. If this was the case then the loopback would pass but you would not be able to communicate with the generator. Try swapping these lines and try again.

bzuelly, 2018-06-04:
swapped lines and problem continues.

jgyates, 2018-06-04:
1) Pictures of your wiring would help
2) How are you powering your pi
3)  Are the LEDs on your converter board flashing with transmit and receive traffic? You should see them flash during a successful lookpack test, do you see them flash with loopback and when connected to the controller?

bzuelly, 2018-06-04:
I just order another converter.  Will try that and reopen if it doesn't work.  Thanks for the quick answers and for sharing genmon.

jgyates, 2018-06-09:
can you run a loopback test that loops pin 7 and 8 on the molex connector?

jgyates, 2018-06-09:
FYI, i created this doc to aid in serial port problems:

https://github.com/jgyates/genmon/wiki/Appendix-6---Serial-Troubleshooting

bzuelly, 2018-06-09:
Loopback successful on the molex end.  Thanks for the doc. I found that earlier and I've been through all those steps.

jgyates, 2018-06-09:
If you can run the loopback through the molex several times with no errors then you may have your tx and rx crossed on the molex, you should be able to swap them quickly on the green screw down terminal. If your tx and rx are crossed on the molex the loopback will pass but you will not be able to communicate with the controller.

bzuelly, 2018-06-09:
ran 5 times successful, then switched tx and rx.  Ran 5 more times.  Good either way.  

controller: evolution
generator model: G0070431

jgyates, 2018-06-09:
once you have swapped your tx and rx, have you tried connecting it to your generator and running the software?

bzuelly, 2018-06-09:
yes, same problem.  Immediately when is run "bash startgenmon.sh start" it says "Warning: not receiving data"  Also, just updated to most recent version of genmon.

jgyates, 2018-06-09:
On a normal restart you should see the following:

pi@genmon:~/genmon $ WARNING:  Not receiving data. : Auto, Off - Ready
 OK - Init complete.

This is because genserv.py is starting and genmon.py is not finished with the init process and the serial watchdog takes 2 seconds to determine if data is being received. The init complete is genserv.py telling use the init process is complete.

Are you powering the pi off the battery or do you have a common ground on the serial connection? I don't see the later in your pictures so I assume you are powering off the battery, correct? If you do not there is a good chance that you will not be able to receive anything either since the two system will not be on a common ground. You can validate that your ground connections is common if you can use a multimeter and detect continuity between the ground on your converter board and your battery.

If you have tried this with swapping tx and rx and the opposite way, then you may have a bad connection on the molex connector on your generator. You may want to look at the molex socket on the controller and see if anything looks strange.

bzuelly, 2018-06-09:
I am not powering from the generator battery.(although maybe that would be easier)  My pi is inside the house and powered from normal AC in the basement.  I do currently have pin 2 on the molex connected to ground on my serial connection, but maybe that's not correct or good enough?

bzuelly, 2018-06-10:
Would I use pin 5 for ground on the serial connectors?  I definitely don't have a common ground.  My green serial connector has a ground pin which is what my black wire coming from the molex connector is connected to.

jgyates, 2018-06-10:
If you check continuity between pin 5 on the serial connector and the pin labeled GND on your converter board, then yes, that would be fine. If there is not continuity then you can run a wire from the pin 5 on the 9 pin serial to one of the ground connections on the pi expansion header. In short, you need the converter board, the pi and the generator controller all on a common ground.

dprus, 2018-06-17:
Also:

pi@genmon-pi:/var/log $ ls -all /dev/serial*
lrwxrwxrwx 1 root root 7 Jun 16 18:09 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Jun 16 18:09 /dev/serial1 -> ttyS0

jgyates, 2018-06-17:
This page has a guide for serial troubleshooting:

https://github.com/jgyates/genmon/wiki/Appendix-6---Serial-Troubleshooting

From your picture you are powering your converter board on 5 volts, 3 volts is recommended (pin 1 on the pi header)

Your lack of connectivity with the generator is likley due to the two system not being on a common ground or you have your pins on the molex wired backwards. You can easily test the crossed wire theory by swapping transmit and receive on the green screw down terminal. If you wires are crossed, loopback will work but comms with the generator will not. Also, if you are not on a common ground with the generator then loopback will work but comms between the pi and generator will not.

The above thread describes how to test a for a common ground. You can also connect the ground pin  (pin 2) on the molex to the screw down terminal #5 on the green terminal connector.

dprus, 2018-06-17:
I switched the terminals on the screws and things did indeed start working. But this doesn't seem to line up with what's on the wiki now?

Also, I believe I am using a common ground because I'm using the 12v to USB 5v setup the wiki recommends I use. You can see it in the pics, kinda. So the Pi is being powered directly from the battery.

Anyway, everything's working now. That's what matters :) thanks!

jgyates, 2018-06-17:
Depending on if you have a DTE or DCE RS-232 pin out the cable connections will be reversed. This diagram shows both scenarios.

https://github.com/jgyates/genmon/blob/master/Diagrams/Cable_Connections.jpg

jgyates, 2018-06-25:
@bzuelly Are you still having problems? I am closing this thread due to lack of activity. Feel free to comment in this thread if you still have issues.

closed 2018-06-25 by jgyates

Cleo1, 2018-08-25:
Just wanted to add my experience to this post. I have been fighting communications with my 20KW Guardian Centurion air-cooled generator with Evolution Controller for weeks. Made my cable, have tried DTE, DCE moved rx/tx nothing. I purchased another MAX3232 board with rx/tx lights to help with troubleshooting. Still nothing. Today I grounded the Pi to the generator. Saw rx/tx lights and knew I had resolved the issue. Everything is working. I am using USB converter from battery to power the Pi. Also settled with DCE. Thank you for your work on this! I have been wanting something like this for years. Will donate to your project.

jgyates, 2018-08-25:
@Cleo1 
Glad it is working for you now. Thanks for the feedback. If you had to make a ground connection from the pi to the generator in addition to the battery to USB connection, my ***guess*** is that the terminal on the battery may not be making a good connection to the cable due to some corrosion on the battery. This solution should workaround that nicely.

Thanks
Jason

jgyates, 2018-08-26:
If you have a multimeter you can measure continuity between your negative battery terminal and generator ground (before you make the last connection). If you did not have continuity then either you have a bad connection on your battery terminal or the DC to DC converter in the Battery to USB converter has an open circuit in the wire. On the USB to Battery converter, the negative battery wire has continuity with the ground wire, if this is not the case then you have an open circuit on that connection.

Cleo1, 2018-08-26:
If I was to use the 5 VDC from the controller do I plug into the 5VDC on the GPIO?

jgyates, 2018-08-26:
yes

Cleo1, 2018-08-26:
Thank you sir!