jgyates/genmon #127: Settings not being saved from settings GUI

danorum, 2018-06-04:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

### Expected Behavior

Open Settings
Change Disable Email Write Commands to On
Setting persists after saving and refreshing the page

### Actual Behavior
Setting does not get saved

### Steps to Reproduce (including precondition)

updated to the latest version 1.9.8
Attempted to set the new setting in the GUI

I have also had issues setting the weather settings via GUI, but I am not sure if that is related to this or my API key or location not working properly.

jgyates, 2018-06-04:
I fixed and issue last night that may cause this issue. Update to 1.9.14 and let me know if you still have and issue.

danorum, 2018-06-04:
The new version didn't fix it. I had to insert the following lines into the /etc/genmon.conf. Once I did that, the setting was editable with the web GUI. It looks like the upgrade doesn't add new entries to the conf file.

# If True, email commands that write to the controller will be disabled
# (settime, setexercise, setremote, etc.). The default is False 
readonlyemailcommands = False

jgyates, 2018-06-04:
I think you are right. Let me look at this. We had that functionality to add missing / new items in there at one point, I will have to go back and see what the situation is here. Thanks for the feedback.

jgyates, 2018-06-05:
@danorum , I checked in an update that corrects this issue. Let me know if you have any issues.

closed 2018-06-06 by jgyates