jgyates/genmon #128: Multiple emails with each "event"?

danorum, 2018-06-04:
I have noticed that when I make a settings change or restart genmon, I get 2 emails with subject "Generator Monitoring Started at xxx"

This weekend, when my exercise started, I received 2 emails at the beginning and 2 at the end. It appears that the first is when the event happens and the second is because of the new log entry.

Are both of these expected? What are the best settings to know what is going on, but not get a bunch of duplicates in my inbox.

Thanks!

jgyates, 2018-06-04:
The two emails when starting are because genmon is starting twice. The first time due to a restart for settings change and the second time due to an issue with the Flask library not letting go of it's resources which cause a failure on reload, so this is expected, but not ideal. 

The other emails are because the generator switch or engine state changed and engine state includes alarm conditions. 

Regarding the best setting, that is very subjective, I suggest you look at the notification page in the web IU.

closed 2018-06-04 by jgyates

danorum, 2018-06-04:
Thanks. Just to make sure I am clear. When the exercise starts, I get one email when the engine status is exercising, but the last start stop log at the bottom is the previous stop event. Then, right after, I get another email while it is still exercising and it is showing the last start stop event as that "Running - Exercise" event.  Then I get another pair when the engine shuts down, one for the engine change and another for the log entry being added.

I will probably play around with the notification options, but wasn't sure if there was a list of what was in each category (I can probably go through the python to find that) or what everyone else was using.

Dan

jgyates, 2018-06-04:
With each email that you get on exercise, look at these two entries in your email:

       Switch State : Auto 
       Engine State : Off - Ready

What should happening is that if either of these change, you get an email. So when starting an exercise cycle, if you have all email notices enabled, you will get an email when starting (cranking) and also when running, another when stopping, and another when the state goes back to ready. the switch state and engine state is read from register 0001 (actually 0001 and 0002 but I combine those two in one 32 bit modbus read), if that changes you get an email.

danorum, 2018-06-04:
In both "starting" emails, they are the same. 
        Switch State : Auto
        Engine State : Exercising

The difference between the 2 is 

first one - 
        Logs : 
            Alarm Log : 05/27/18 16:51:04 Check Battery 
            Start Stop Log : 05/27/18 16:51:06 Stopped - Auto 

second one -
        Logs : 
            Alarm Log : 05/27/18 16:51:04 Check Battery 
            Start Stop Log : 06/02/18 10:00:05 Running - Exercise 

It isn't a huge deal, I was just trying to figure out how to get the right info without it being too chatty.

Dan