jgyates/genmon #129: Settings page stopped working with update to V1.9.14

smeisner, 2018-06-04:
This setup was working fine until I updated the software (via the web page). I believe this is when the Settings page stopped responding. Shortly before that I also enabled Twilio support and was missing some libraries, so I had a few "mis-steps" until I figured out I needed "apt-get install build-essential libssl-dev libffi-dev python-dev". Then I did the 'pip' commands to add Twilio support.

### Expected Behavior

To be presented with the settings page when I click "Settings" in the left hand bar.

### Actual Behavior

No response in browser window. Activity continues, just no settings page is generated.

### Steps to Reproduce (including precondition)

In Google Chrome web browser, click on "Settings".

### Screenshot on This Problem (if possible)

When I run genmon in the foreground, I see the following when I click on "Settings":

192.168.0.200 - - [04/Jun/2018 13:33:47] "GET /cmd/settings HTTP/1.1" 200 -
----------------------------------------
Exception happened during processing of request from ('192.168.0.200', 64142)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 596, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 331, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 654, in __init__
    self.finish()
  File "/usr/lib/python2.7/SocketServer.py", line 713, in finish
    self.wfile.close()
  File "/usr/lib/python2.7/socket.py", line 283, in close
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 307, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
error: [Errno 32] Broken pipe
----------------------------------------

### Your Environment

Raspberry Pi Zero W
Raspbian 9 Lite 

pi@generator:~/genmon $ python -V
Python 2.7.13

- Genereator Model: G0060331 (Honeywell/Generac 14kW air-cooled with Nexus)
- Genmon Version: V1.9.14
- Generator Registers:
Registers :
    Num Regs : 170
    Not Changed : 641
    Changed : 209
    Total Changed : 0.33

    Base Registers :
        0239 : 0000
        0039 : 0000
        0038 : 0000
        0034 : 0000
        0037 : 115e
        0036 : 0000
        0237 : 0000
        0033 : 0000
        0032 : 2ea2
        05f7 : 03fb
        05f6 : 03fb
        05f5 : 03fb
        05f4 : 03fb
        05f3 : 03fb
        05f1 : 03fb
        05ed : 0000
        023b : 0000
        003a : 0000
        023e : 0000
        003b : 0000
        05fa : 03fb
        0059 : 03fb
        0058 : 03fb
        0053 : 03fb
        0052 : 03fb
        0057 : 03fb
        0056 : 0000
        0055 : 03fb
        0054 : 03fb
        0208 : 03fb
        005c : 0000
        005a : 0000
        005f : 0000
        005e : 0000
        005d : 0000
        003c : 0000
        000d : 0000
        000e : 0d33
        000f : 0604
        000a : 008d
        000b : 0000
        000c : 0058
        0005 : 081d
        0006 : 0101
        0007 : 0000
        0000 : 0003
        0001 : 00000000
        0008 : 0000
        0009 : 00e5
        05ee : 0000
        0012 : 0000
        0011 : 009c
        0010 : 0112
        0019 : 0003
        002b : 0000
        002c : 0000
        002a : 668e
        002f : 0000
        002d : 0000
        002e : 0000
        0022 : 45e9
        0020 : 4576
        0021 : 016c
        001f : 00a4
        001e : 00a4
        001d : 4576
        001c : 0072
        001b : 4576
        001a : 00a4
        01f4 : 30303036393032383232

smeisner, 2018-06-04:
As a followup, if I use `/home/pi/genmon/startgenmon.sh stop` (as root), I see the following dumps:

```
./startgenmon.sh stop
Stopping genmon python scripts
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/home/pi/genmon/genmonlib/mymail.py", line 141, in Close
    self.Mailbox.close()
  File "/usr/lib/python2.7/imaplib.py", line 395, in close
    typ, dat = self._simple_command('CLOSE')
  File "/usr/lib/python2.7/imaplib.py", line 1088, in _simple_command
    return self._command_complete(name, self._command(name, *args))
  File "/usr/lib/python2.7/imaplib.py", line 910, in _command_complete
    typ, data = self._get_tagged_response(tag)
  File "/usr/lib/python2.7/imaplib.py", line 1017, in _get_tagged_response
    self._get_response()
  File "/usr/lib/python2.7/imaplib.py", line 929, in _get_response
    resp = self._get_line()
  File "/usr/lib/python2.7/imaplib.py", line 1027, in _get_line
    line = self.readline()
  File "/usr/lib/python2.7/imaplib.py", line 1189, in readline
    return self.file.readline()
  File "/usr/lib/python2.7/socket.py", line 451, in readline
    data = self._sock.recv(self._rbufsize)
  File "/usr/lib/python2.7/ssl.py", line 766, in recv
    return self.read(buflen)
  File "/usr/lib/python2.7/ssl.py", line 653, in read
    v = self._sslobj.read(len)
TypeError: Close() takes exactly 1 argument (3 given)

Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/home/pi/genmon/genmonlib/mymail.py", line 141, in Close
    self.Mailbox.close()
  File "/usr/lib/python2.7/imaplib.py", line 395, in close
    typ, dat = self._simple_command('CLOSE')
  File "/usr/lib/python2.7/imaplib.py", line 1088, in _simple_command
    return self._command_complete(name, self._command(name, *args))
  File "/usr/lib/python2.7/imaplib.py", line 910, in _command_complete
    typ, data = self._get_tagged_response(tag)
  File "/usr/lib/python2.7/imaplib.py", line 1017, in _get_tagged_response
    self._get_response()
  File "/usr/lib/python2.7/imaplib.py", line 929, in _get_response
    resp = self._get_line()
  File "/usr/lib/python2.7/imaplib.py", line 1027, in _get_line
    line = self.readline()
  File "/usr/lib/python2.7/imaplib.py", line 1189, in readline
    return self.file.readline()
  File "/usr/lib/python2.7/socket.py", line 451, in readline
    data = self._sock.recv(self._rbufsize)
  File "/usr/lib/python2.7/ssl.py", line 766, in recv
    return self.read(buflen)
  File "/usr/lib/python2.7/ssl.py", line 653, in read
    v = self._sslobj.read(len)
TypeError: Close() takes exactly 1 argument (3 given)
Exception in thread ProcessThread (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 754, in run
  File "/home/pi/genmon/genmonlib/controller.py", line 160, in ProcessThread
  File "/home/pi/genmon/genmonlib/generac_evolution.py", line 622, in MasterEmulation
  File "/home/pi/genmon/genmonlib/mymodbus.py", line 140, in ProcessMasterSlaveTransaction
  File "/home/pi/genmon/genmonlib/mymodbus.py", line 175, in ProcessOneTransaction
  File "/usr/lib/python2.7/threading.py", line 216, in __exit__
  File "/usr/lib/python2.7/threading.py", line 203, in release
<type 'exceptions.TypeError'>: 'NoneType' object is not callable
Last : <READY>, New : <>
Generator Ready End

```

jgyates, 2018-06-04:
Hi, Regarding the errors on stop. That is because the version I checked in last night has two shutdown handlers instead of one. I corrected that issue a few min ago. Thoes errros are due to the second shutdown handler trying to cleanup after we have already cleaned up. An update to 1.9.15 should fix that.

Regarding your fist issue, let's try this from the /home/pi/genmon directory:

1) Update to the latest

        bash ./genmonmaint.sh update

2) recopy your /home/pi/genmon.conf file to /etc/genmon.conf

        sudo cp /home/pi/genmon.conf /etc

3) restart

     bash ./startgenmon.sh restart

You will likely see the error messages on restart since V1.9.14 is stopping and V1.9.15 is starting

If you still do not see a settings page they, enter this address in your browser:

      http://MYIPADDRESS:8000/cmd/sendlogfiles

replace MYIPADDRESS with the address of you pi. If you changed the port number change the 8000 to the port you are using. This will send your log files to me so I can see if there are any errors occurring, or you can post them here. Specifically interested in /var/log/genmon.log and /var/log/genserv.log

Also, if you still see a problem with the setting page after update, open the developer view on Chrome (Settings -> More Tools > Developer Tools, then look at the console output. Do you see any errors here when running the web app?

smeisner, 2018-06-04:
I sent the log files a few moments ago.  Settings page still not showing up after running `bash ./genmonmaint.sh update`. After the git update and copying the genmon.conf to /etc, I did not modify it, so I ran with the default conf.

> Also, if you still see a problem with the setting page after update, open the developer view on Chrome  then look at the console output. Do you see any errors here when running the web app?

Yes, I saw the following on the Chrome console:
```
jquery-3.3.1.min.js:2 XHR failed loading: GET "http://generator:8000/cmd/settings".
send @ jquery-3.3.1.min.js:2
ajax @ jquery-3.3.1.min.js:2
DisplaySettings @ genmon.js:1260
MenuClick @ genmon.js:1887
(anonymous) @ genmon.js:54
dispatch @ jquery-3.3.1.min.js:2
y.handle @ jquery-3.3.1.min.js:2

```
Also, when I run genmonstart.sh stop, it is throwing (what seems to be) more errors;

```
bash ./startgenmon.sh stop
Stopping genmon python scripts
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/home/pi/genmon/genmonlib/mymail.py", line 141, in Close
    self.Mailbox.close()
  File "/usr/lib/python2.7/imaplib.py", line 395, in close
    typ, dat = self._simple_command('CLOSE')
  File "/usr/lib/python2.7/imaplib.py", line 1088, in _simple_command
    return self._command_complete(name, self._command(name, *args))
  File "/usr/lib/python2.7/imaplib.py", line 910, in _command_complete
    typ, data = self._get_tagged_response(tag)
  File "/usr/lib/python2.7/imaplib.py", line 1017, in _get_tagged_response
    self._get_response()
  File "/usr/lib/python2.7/imaplib.py", line 929, in _get_response
    resp = self._get_line()
  File "/usr/lib/python2.7/imaplib.py", line 1027, in _get_line
    line = self.readline()
  File "/usr/lib/python2.7/imaplib.py", line 1189, in readline
    return self.file.readline()
  File "/usr/lib/python2.7/socket.py", line 451, in readline
    data = self._sock.recv(self._rbufsize)
  File "/usr/lib/python2.7/ssl.py", line 766, in recv
    return self.read(buflen)
  File "/usr/lib/python2.7/ssl.py", line 653, in read
    v = self._sslobj.read(len)
TypeError: Close() takes exactly 1 argument (3 given)
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/home/pi/genmon/genmonlib/mymail.py", line 141, in Close
    self.Mailbox.close()
  File "/usr/lib/python2.7/imaplib.py", line 395, in close
    typ, dat = self._simple_command('CLOSE')
  File "/usr/lib/python2.7/imaplib.py", line 1088, in _simple_command
    return self._command_complete(name, self._command(name, *args))
  File "/usr/lib/python2.7/imaplib.py", line 910, in _command_complete
    typ, data = self._get_tagged_response(tag)
  File "/usr/lib/python2.7/imaplib.py", line 1017, in _get_tagged_response
    self._get_response()
  File "/usr/lib/python2.7/imaplib.py", line 929, in _get_response
    resp = self._get_line()
  File "/usr/lib/python2.7/imaplib.py", line 1027, in _get_line
    line = self.readline()
  File "/usr/lib/python2.7/imaplib.py", line 1189, in readline
    return self.file.readline()
  File "/usr/lib/python2.7/socket.py", line 451, in readline
    data = self._sock.recv(self._rbufsize)
[... 20 lines cut ...]

  File "/usr/lib/python2.7/threading.py", line 754, in run

  File "/home/pi/genmon/genmonlib/mypipe.py", line 83, in ReadPipeThread<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'path'  File "/home/pi/genmon/genmonlib/myserial.py", line 90, in ReadThread  File "/usr/lib/python2.7/threading.py", line 754, in run

  File "/home/pi/genmon/genmonlib/controller.py", line 160, in ProcessThread<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'path'
  File "/home/pi/genmon/genmonlib/controller.py", line 179, in CheckAlarmThread

  File "/home/pi/genmon/genmonlib/myserial.py", line 143, in Read
  File "/home/pi/genmon/genmonlib/generac_evolution.py", line 629, in MasterEmulation
  File "/home/pi/genmon/genmonlib/mymodbus.py", line 140, in ProcessMasterSlaveTransaction
  File "/home/pi/genmon/genmonlib/mymodbus.py", line 175, in ProcessOneTransaction
  File "/usr/lib/python2.7/threading.py", line 216, in __exit__
  File "/usr/lib/python2.7/threading.py", line 203, in release
<type 'exceptions.TypeError'>: 'NoneType' object is not callable

  File "/home/pi/genmon/genmonlib/generac_evolution.py", line 1318, in CheckForAlarms  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 498, in read

<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'error'
  File "/home/pi/genmon/genmonlib/generac_evolution.py", line 1244, in CheckForOutage
  File "/home/pi/genmon/genmonlib/generac_evolution.py", line 2624, in GetUtilityVoltage
  File "/home/pi/genmon/genmonlib/controller.py", line 306, in GetParameter
<type 'exceptions.TypeError'>: 'NoneType' object is not callable

```

smeisner, 2018-06-04:
Another update...

If I enter into the Chrome address bar, `http://generator:8000/cmd/settings`, I get a large block of text:

```
{"autofeedback":["boolean","Automated Feedback",29,false,"If enabled, an email will be sent to the software maintainer if critical data is missing in the software for your generator. The email will contain the contents of the generator registers. Enabling this option helps support the widest possible set of features and supported generators." ,""],"certfile":["string","https Certificate File",205,"","If Use 'Self-signed Certificate' is disabled, denotes the path and filename of the cert file","UnixFile"], "disableemail":["boolean","Disable Email Usage",300,false,"Disable email use by the program.",""],"disableoutagecheck":["boolean","Do Not Check for Outages",17,false,

etc

```

I did not include everything since it's huge and has passwords and email addresses. Either way, the above is not formatted as an HTML page.

jgyates, 2018-06-04:
Yes, regarding the http://generator:8000/cmd/settings command. This is the expected behavior. This is the command that the java script sends to the web server to request settings. This is not related to your original issue.

Regarding the error messages on stop. This is known behavior. Some of the threads that are used are stopped abruptly on a stop command. This has always occurred in the background, but as part of some new changes I am trapping the shutdown event. In a future change this will be reduced. This is not related to your original issue.

Michael and I are discussing your original issue (settings page not showing up). For the moment I think I have all of the data I need. We are trying to track down the issue now.

smeisner, 2018-06-05:
I updated to 1.9.16. When I click Settings, the page still does not show, but genmon outputs:
```
192.168.0.157 - - [05/Jun/2018 07:12:55] "GET /cmd/settings HTTP/1.1" 200 -
----------------------------------------
Exception happened during processing of request from ('192.168.0.157', 59672)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 596, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 331, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 654, in __init__
    self.finish()
  File "/usr/lib/python2.7/SocketServer.py", line 713, in finish
    self.wfile.close()
  File "/usr/lib/python2.7/socket.py", line 283, in close
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 307, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
error: [Errno 32] Broken pipe
----------------------------------------

```

On shutdown, it does look much cleaner:

```
pi@generator:~/genmon $ bash ./startgenmon.sh stop
Stopping genmon python scripts
 Exception in thread SocketWorkThread (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 754, in run
  File "/home/pi/genmon/genmon.py", line 811, in SocketWorkThread
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'error'

```
The output in the Chrome console is still the same when I click Settings. Although the last time I tested, I had to click Settings 3 times before I saw the Python traceback and got the following in the Chrome console:
```
jquery-3.3.1.min.js:2 XHR failed loading: GET "http://192.168.0.169:8000/cmd/settings".
send @ jquery-3.3.1.min.js:2
ajax @ jquery-3.3.1.min.js:2
DisplaySettings @ genmon.js:1260
MenuClick @ genmon.js:1887
(anonymous) @ genmon.js:54
dispatch @ jquery-3.3.1.min.js:2
y.handle @ jquery-3.3.1.min.js:2

```

Is there any other info I can gather to help? Even if you want remote access, I can set that up.

smeisner, 2018-06-05:
Forgot to mention; As of the 1.9.14 update, a new error is showing up. From /var/log/genmon.log:

`Note LookUpSNInfo (parse request 1), (product ID not found): list index out of range : generac_evolution.py:452`

This causes a red banner at the bottom of the web pages claiming an unknown error occurred.

If I go to the site `https://www.generac.com/GeneracCorporate/WebServices/GeneracSelfHelpWebService.asmx/GetSearchResults?query=<my_serial_number>`, I get the following response:

```
<string xmlns="http://tempuri.org/">
{"SerialNumber":{"Number":"<my_serial_number>","ModelNumber":"G0060331","Registered":true,"ValidSerialNumber":true},"Results":[],"ManualModelNumber":"G0060331","ResultsText":"Model/Serial Number not found."}
</string>
```

smeisner, 2018-06-05:
**It's working now!!**

On another system I had a web browser open to my genmon site. Having two browsers open seemed to cause problems. When I closed one of the browsers, the settings page showed up on the other.

I also enabled the Autmatic Feedback, got an error about sending email and then the red banner (mentioned above this reply) disappeared.

```
Exception in thread SendMailThread (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 754, in run
  File "/home/pi/genmon/genmonlib/mymail.py", line 294, in SendMailThread
  File "/home/pi/genmon/genmonlib/mymail.py", line 262, in sendEmailDirectMIME
  File "/usr/lib/python2.7/smtplib.py", line 582, in login
  File "/usr/lib/python2.7/smtplib.py", line 542, in ehlo_or_helo_if_needed
  File "/usr/lib/python2.7/smtplib.py", line 414, in ehlo
  File "/usr/lib/python2.7/smtplib.py", line 362, in getreply
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'error'
Starting genmon python scripts

```

jgyates, 2018-06-05:
If you see the message "most likely raised during interpreter shutdown)" this is due to the way we are stopping the app to restart (kill). This error i not relevant.

Are you using a login/password to log into the web site? What other options do you have enabled. I use with multiple browers all the time with no issues. 

What is the other browser? Firebox, Internet Explorer, ? 

Is the error isolated to one browser? If so the disable any extensions to determine which one is causing a problem.

regarding this error: "LookUpSNInfo (parse request 1), (product ID not found): list index out of range : generac_evolution.py:452"

Please check your logs, old logs are stored as /var/log/genmon.log.1, /var/log/genmon.log.2, etc. Nothing related to this code has changed in 1.9.14. When you say that this is not working since V1.9.14, can you validate this with the logs or is this an error that you have just noticed that may have been happening. This matters because if there is a problem the info you tell us helps us track down the issue. The function LookUpSNInfo is run every time a new version is installed and the version info along with the success or failure of this call is also logged.

jgyates, 2018-06-05:
If the browser error is limited to Chrome, please test in incognito mode also.

closed 2018-06-05 by jgyates

jgyates, 2018-06-05:
Also, in your logs you have an unknown log entry that the software has not encounter. The first several entries in your alarm log are: 

       01/01/18 09:01:21 Overcrank 
        01/01/18 09:01:01 Check Battery 
        01/14/17 12:37:07 Check Battery 
        01/14/17 12:16:01 Low Battery 
        01/14/17 12:15:01 Check Battery 
        01/13/17 08:17:06 Check Battery 
        01/19/33 08:17:22 Unknown 0x05 
        09/08/16 17:08:54 Inspect Spark Plugs 

Please review your alarm log and let me know the text your controller displays for the message that corresponds to "01/19/33 08:17:22 Unknown 0x05 " in your actual log. With this info I can add that log entry to the software for log code 0x05.

jgyates, 2018-06-05:
The unknown entry in your alarm log is likely the cuase to the unknown condition that cause a red icon. This is expected behavior if you have unknown log entries. You should have a notice on your web UI log page about this.

jgyates, 2018-06-05:
I went back and did some testing regarding the following error:

     Note LookUpSNInfo (parse request 1), (product ID not found): list index out of range : generac_evolution.py:452

Short story is that, this is not an error, it is a notice. It is letting me know that the first phase of the serial number lookup could not find the product ID so it is using the 2nd phase.

I know you said that "As of the 1.9.14 update, a new error is showing up. ". I think you will find that if you look at your logs, that this is not the case.

smeisner, 2018-06-05:
Interesting...I did check the logs and the message began showing up on every start of genmon after the .14 update.

> Are you using a login/password to log into the web site? 

No...not yet.

> What other options do you have enabled.

SMS (via Trelio). But since the last update,I have not reenabled that. Would you like to see my conf files to assess all the options enabled?

> What is the other browser? Firebox, Internet Explorer, ?
> Is the error isolated to one browser? 
> If the browser error is limited to Chrome, please test in incognito mode also.

I was using only Chrome. I tried 2 browsers both in incognito mode, but the problem still occurred.
I tried IE, but got a number of script errors on the developer console:
```
HTML1300: Navigation occurred.
generator:8000
HTML1202: http://generator:8000/ is running in Compatibility View because 'Display intranet sites in Compatibility View' is checked.
generator:8000
SCRIPT5007: Unable to get property 'createElement' of undefined or null reference
jquery-3.3.1.min.js (2,6997)
SCRIPT5009: 'jQuery' is undefined
jquery-ui.min.js (6,73)
SCRIPT5009: 'jQuery' is undefined
jquery.ui.touch-punch.min.js (11,2)
SCRIPT1028: Expected identifier, string or number
libraries.min.js (874,5)
SCRIPT1028: Expected identifier, string or number
genmon.js (406,5)

```
I also tried a Firefox browser and a Chrome browser; could not access Settings.

> Please check your logs, old logs are stored as /var/log...

I grep'd all the genmon log files and got 3 hits. The first two worked fine. The third (after the update this morning) showed the error message. Based on what you described, I would presume the first 2 are after the initial 1.9.13 install and the 1.9.14 update. The last is after the 1.9.15 update today.

> Please review your alarm log and let me know the text your controller displays for the message that corresponds to "01/19/33 08:17:22 Unknown 0x05 " in your actual log.

I will tonight. I need to physically access the generator to get this info. (at work now).

smeisner, 2018-06-05:
Explicit detail:

```
root@generator:/var/log# grep 'New version detected' genmon*
genmon.log:2018-06-04 12:20:13,239 : New version detected: Old = V1.9.13, New = V1.9.14
genmon.log:2018-06-04 15:38:59,107 : New version detected: Old = V1.9.15, New = V1.9.15
genmon.log:2018-06-05 07:11:43,346 : New version detected: Old = V1.9.15, New = V1.9.16
root@generator:/var/log#
root@generator:/var/log# grep LookUpSNInfo genmon*
genmon.log:2018-06-04 12:20:23,193 : Note LookUpSNInfo (parse request 1), (product ID not found): list index out of range : generac_evolution.py:452
genmon.log:2018-06-04 15:39:08,593 : Note LookUpSNInfo (parse request 1), (product ID not found): list index out of range : generac_evolution.py:452
genmon.log:2018-06-05 07:11:54,962 : Error LookUpSNInfo (parse request 1), (product ID not found): list index out of range : generac_evolution.py:456

```

jgyates, 2018-06-05:
Regarding the LookUpSNInfo, if it did start showing up on a specific date, must be due to generac changing their data base. This code works in two steps. Step one: use the serial number to get the product ID, then lookup model name and kw rating based on the product ID. If it cant find the product ID then it looks up from a different URL (for older models) to find the same info. this message is to let me know that it is switching from step one to step two. The software is finding your model and kw rating, or at least it is when I run simulation based on your registers so this is not really an issue.

Are you able to see the same errors in the browser when running on a local network and are you using a VLAN?

A screen cap of the developer view of this error would be helpful.

smeisner, 2018-06-05:
[image]

jgyates, 2018-06-05:
Are you able to see the same errors in the browser when running on a local network and are you using a VLAN?

    Nope...no VLANs in use here.

Was that, no, you can not see the same error when running on a local network or, no, no VLANs?

jgyates, 2018-06-05:
Also, can you send a screen cap of the network page of the developer tools when the error occurs?

To recap, is the way  you use to reproduce this is having two browsers open at the same time and one can not access the settings page?

smeisner, 2018-06-05:
> Was that, no, you can not see the same error when running on a local network or, no, no VLANs?

The latter; No VLANs.

> is the way you use to reproduce this is having two browsers open at the same time and one can not access the settings page?

Correct. If there is more than 1 browser running with a genmon page open, the Settings page is inaccessible by any of the browsers. Once all but 1 browsers are closed, the Settings page becomes accessible.

I will post a screen cap of the network page soon.

smeisner, 2018-06-05:
[image]

jgyates, 2018-06-05:
Can you update to the latest and retest. Then send me your log files again? I added some additional error checking.

jgyates, 2018-06-05:
You can send the log files by using this address:

     http://MYIPADDRESS:8000/cmd/sendlogfiles

jgyates, 2018-06-05:
I did not change the version since genmon.py did not change. I only updated two files to include or change the log output so I did not change the versions number. Try to access the settings page and then send your log files.

smeisner, 2018-06-05:
OK, did it. Sorry I poked around a bit after trying to access the settings page, so the logs may be a bit noisy.

jgyates, 2018-06-05:
Nothing indicating an error in the logs. Michael will see if there are any things in the javascript that my be causing this issue.

smeisner, 2018-06-05:
As I mentioned before, I can set up remote access if you (and Michael) want to check things out directly.

I am happy to help in any way!

smeisner, 2018-06-05:
> Please review your alarm log and let me know the text your controller displays for the message that corresponds to "01/19/33 08:17:22 Unknown 0x05 " in your actual log.

As shown below, the error is "Underspeed"

[image]

jgyates, 2018-06-06:
I checked in an update the reads some of the data on the settings page before everything is loaded. This may help speed things up on the settings page.  There is still quite a bit of reading the SD card going on in the settings page, but this removes some of it.

I also added the alarm "Underspeed"

When you have a moment can you look at your SD card and see the speed printed on SD card. This link has some info on SD card speeds.

https://www.sdcard.org/developers/overview/speed_class/

If you don't see it just take a picture of your SD card and post it here. I would like to include this info in some of the documentation regarding minimum SD card speed.

smeisner, 2018-06-06:
V1.9.17 looks perfect! The time it takes for the Settings page to come up is now very quick with 2 browsers running. I am also seeing no "Unknown errors" with the additional Underspeed code.

Also no failure when looking up the serial number:

```
2018-06-05 21:18:46,391 : New version detected: Old = V1.9.16, New = V1.9.17
2018-06-05 21:18:49,815 : Looking up model info on internet
2018-06-05 21:18:50,102 : ModelID : Model ID register is unknown
2018-06-05 21:18:55,435 : Found: Model: G0060331
2018-06-05 21:18:55,437 : Notice: product ID not found online. Using serial number to complete lookup. : generac_evolution.py:456

```
I will do some research on the SD card I used. I didn't think too much about it when I grabbed a spare to set this up.

jgyates, 2018-06-06:
Also, is your generator a 15kw?