jgyates/genmon #130: Openweathermap settings not saving

kalex1114, 2018-06-06:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

### Expected Behavior

Saving and enabling openweathermap

### Actual Behavior

After clicking save and restarting genmon openweathermap is set back to off and api/location info is missing

### Steps to Reproduce (including precondition)

1) click enable
2) copy/paste api key from the website
3) enter 7 digit city id from the website
4) click save
5) after saving and restarting genmon weather data is not displayed and going into settings screen shows it being turned off

### Screenshot on This Problem (if possible)
1) entering and saving data:
[image]

2) after clicking save and going back into settings:
[image]

### Your Environment
Tried it on both Safari on a OSX and Chrome on windows 10 laptop
- Genereator Model: G0070370
            Hardware Version : V1.06
            Firmware Version : V1.16
- Generator Registers: {Please write here}
- Genmon Version: V1.9.18

jgyates, 2018-06-06:
I check in a partial fix. The on/off button in the gui for weather is not working, but you can now save the settings. To disable the weather, remove the api key. Should have a fix for the UI soon.

kalex1114, 2018-06-06:
Updated and able to save it now.  After saving, GUI icon shows as ON

Where do I see the weather?  not seeing it on the status and monitor pages

jgyates, 2018-06-06:
You would see the weather on the monitor page. Check /var/log/genmon.log for errors relating to your key or location lookup.

kalex1114, 2018-06-06:
was missing pyowm libarary.  Installed it but now getting this:

2018-06-06 13:52:29,498 : Generator Monitor Shutdown
2018-06-06 13:52:40,544 : Error in InitOWM: Close() takes exactly 1 argument (3 given) : myweather.py:69
2018-06-06 13:52:40,575 : GenMon Loadded for site: Lawton
2018-06-06 13:53:08,889 : Generator Monitor Shutdown
2018-06-06 13:53:11,868 : Validation Error: Invalid register length (Start) 012c 0002

kalex1114, 2018-06-06:
reboot didn't help

2018-06-06 13:53:11,868 : Validation Error: Invalid register length (Start) 012c 0002
2018-06-06 13:55:45,843 : Error opening controller device: Close() takes exactly 1 argument (3 given)
2018-06-06 13:55:45,926 : Error getting thread name in KillThread: TimeSyncThread
2018-06-06 13:55:45,931 : Error getting thread name in KillThread: ComWatchDog
2018-06-06 13:55:49,258 : Error Closing Monitor: [Errno 107] Transport endpoint is not connected : genmon.py:885
2018-06-06 13:55:49,260 : Generator Monitor Shutdown
2018-06-06 13:56:19,255 : Error opening controller device: Close() takes exactly 1 argument (3 given)
2018-06-06 13:58:25,696 : GenMon Loadded for site: Lawton

kalex1114, 2018-06-06:
Weather is showing in monitor page.  But status page is not loading in the browser.

jgyates, 2018-06-06:
How did you install pyowm? Check your crontab to see if you are loading genmon twice.

kalex1114, 2018-06-06:
first i did pip install pyowm still saw the error and then did sudo pip install pyowm

genmon is only listed once in crontab

jgyates, 2018-06-06:
Not sure what is happening. Send me your log files with this command:

http://MYIPADDRESS:8000/cmd/sendlogfiles

replace the url with your IP address and port

jgyates, 2018-06-06:
One thing that the log snippet you posted could point to is that genserv.py is failing and trying to reload. If that is the case the url for logs may not work. Post the log for genserv.py : 
 /var/log/genserv.log

jgyates, 2018-06-06:
You have quite a history of serial errors in myserial.log. Let's run this test.

     sudo python ./OtherApps/serialcheck.py -c

kalex1114, 2018-06-06:
Are those old or recent?  took me a while to get the cable built so its been operational only for 5 days or so.  

here is the output :  
Checking : Serial console command line removed: OK
Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: OK
Checking : Serial console service disabled: OK

I uninstalled pyowm (both instances) deleted api key for openweather and rebooted. Currently genmon is running and status page is working.

jgyates, 2018-06-06:
Try this:

     sudo pip install pyowm
     pip freeze | grep 'pyowm'

The version of pyowm I have on my system is:  pyowm==2.8.0

then reboot, do not setup your openweathermap api key yet.

jgyates, 2018-06-06:
then send me your logs again.

jgyates, 2018-06-06:
If this works then that points to some issue with your system and pyowm

kalex1114, 2018-06-06:
pi@genmon:~ $ sudo pip install pyowm
Collecting pyowm
Requirement already satisfied: requests<2.19,>=2.18.2 in /usr/local/lib/python2.7/dist-packages (from pyowm)
Requirement already satisfied: urllib3<1.23,>=1.21.1 in /usr/local/lib/python2.7/dist-packages (from requests<2.19,>=2.18.2->pyowm)
Requirement already satisfied: idna<2.7,>=2.5 in /usr/local/lib/python2.7/dist-packages (from requests<2.19,>=2.18.2->pyowm)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /usr/local/lib/python2.7/dist-packages (from requests<2.19,>=2.18.2->pyowm)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python2.7/dist-packages (from requests<2.19,>=2.18.2->pyowm)
Installing collected packages: pyowm
Successfully installed pyowm-2.8.0
pi@genmon:~ $ pip freeze |grep 'pyowm'
pyowm==2.8.0

rebooting now

kalex1114, 2018-06-06:
rebooted and logs sent

jgyates, 2018-06-06:
ok. Got your logs, no errors on reboot, I assume it is running OK. Correct?

If it is then let's add your weather api key and location directly to /etc/genmon.conf

Then manually restart

     bash ./startgenmon.py restart

Do you see any errors on the console and send me your logs again.

kalex1114, 2018-06-06:
Yep loaded up fine no errors.

just added api and location to the conf file, restarted manually.  loaded without issues.  logs sent
weather works too.

console looks clean as well

pi@genmon:~/genmon $ bash ./startgenmon.sh restart
Stopping genmon python scripts
Starting genmon python scripts
pi@genmon:~/genmon $ OK : Auto, Off - Ready
 OK - Init complete.
 * Running on http://0.0.0.0:8000/ (Press CTRL+C to quit)
192.168.0.29 - - [06/Jun/2018 16:26:36] "GET /cmd/status_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:36] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:36] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:38] "GET /cmd/monitor_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:38] "GET /images/transparent.png HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:39] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:41] "GET /cmd/monitor_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:41] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:41] "GET /images/transparent.png HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:42] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:45] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:45] "GET /cmd/status_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:46] "GET /cmd/status_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:46] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:47] "GET /cmd/monitor_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:48] "GET /images/transparent.png HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:48] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:51] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:51] "GET /cmd/monitor_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:51] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:51] "GET /images/transparent.png HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:54] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:56] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:56] "GET /cmd/monitor_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:56] "GET /images/transparent.png HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:26:57] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:01] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:01] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:01] "GET /cmd/monitor_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:01] "GET /images/transparent.png HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:04] "GET / HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:04] "GET /jquery-3.3.1.min.js HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:04] "GET /jquery.ui.touch-punch.min.js HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:04] "GET /libraries.min.js HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:04] "GET /jquery-ui.min.js HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:04] "GET /jquery-ui.css HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:04] "GET /genmon.css HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:04] "GET /libraries.min.css HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:04] "GET /genmon.js HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:05] "GET / HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:05] "GET /cmd/getfavicon HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:05] "GET /favicon.ico HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:05] "GET /cmd/getreglabels HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:05] "GET /cmd/gui_status_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:05] "GET /cmd/power_log_json?power_log_json:%2010000 HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:05] "GET /cmd/registers_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:05] "GET /images/transparent.png HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:05] "GET /images/sprites.png HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:06] "GET /cmd/sendlogfiles HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:06] "GET /cmd/start_info_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:06] "GET /favicon.ico HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:06] "GET /cmd/monitor_json HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:06] "GET /cmd/sendlogfiles HTTP/1.1" 200 -
192.168.0.29 - - [06/Jun/2018 16:27:07] "GET /favicon.ico HTTP/1.1" 200 -

jgyates, 2018-06-06:
interesting. I guess that pip install without the sudo was the thing that made it go bonkers. Strange. I don't really understand why, but at least it is working. Let me know if you have any other issues. 

I have added a bit more error checking to be on the safe side. I will check in those changes shortly..

kalex1114, 2018-06-06:
Yep seems like it was.  Thank you

closed 2018-06-07 by jgyates