jgyates/genmon #131: Data missing/unknown for Honeywell/Generac 15KW

smeisner, 2018-06-07:
Today, I updated from V1.9.17, then to V1.9.18 and then directly to .20 and started seeing Unknown Log Entries, missing serial number, service due is grouped into A and B instead of Air filter, Oil, oil filter, sparkplug, battery.

Log after upgrade from .16 to .17:
```
2018-06-05 21:18:46,391 : New version detected: Old = V1.9.16, New = V1.9.17
2018-06-05 21:18:49,815 : Looking up model info on internet
2018-06-05 21:18:50,102 : ModelID : Model ID register is unknown
2018-06-05 21:18:55,435 : Found: Model: G0060331
2018-06-05 21:18:55,437 : Notice: product ID not found online. Using serial number to complete lookup. : generac_evolution.py:456
2018-06-05 21:18:58,443 : GenMon Loadded for site: Nashua
2018-06-05 22:59:37,291 : Excpetion in InterfaceServerThread[Errno 4] Interrupted system call : genmon.py:833
2018-06-05 23:01:59,203 : GenMon Loadded for site: Nashua
2018-06-06 10:16:37,819 : Generator Monitor Shutdown

```
After the upgrade from .17 to .18, the log was pretty clean:
```
2018-06-06 10:16:40,577 : New version detected: Old = V1.9.17, New = V1.9.18
2018-06-06 10:16:43,281 : Looking up model info on internet
2018-06-06 10:16:52,820 : Found: Model: G0060331
2018-06-06 10:16:52,822 : Notice: product ID not found online. Using serial number to complete lookup. : generac_evolution.py:457
2018-06-06 10:16:52,924 : GenMon Loadded for site: Nashua
2018-06-06 10:16:57,484 : Generator Monitor Shutdown
2018-06-06 10:17:01,988 : GenMon Loadded for site: Nashua
2018-06-06 17:19:07,757 : Generator Monitor Shutdown
```
After the .18 to .20 upgrade, it didn't look as clean. At this time I also added my openweather API key and rebooted:

```
2018-06-06 17:19:10,912 : New version detected: Old = V1.9.18, New = V1.9.20
2018-06-06 17:19:13,786 : Looking up model info on internet
2018-06-06 17:19:19,748 : Found: Model: G0060331
2018-06-06 17:19:19,759 : Notice: product ID not found online. Using serial number to complete lookup. : generac_evolution.py:457
2018-06-06 17:19:23,671 : Error in InitOWM: Close() takes exactly 1 argument (3 given) : myweather.py:83
2018-06-06 17:19:23,715 : GenMon Loadded for site: Nashua
2018-06-06 17:19:51,612 : Generator Monitor Shutdown
2018-06-06 17:21:32,535 : Excpetion in InterfaceServerThread[Errno 4] Interrupted system call : genmon.py:839
2018-06-06 17:22:33,241 : Error opening controller device: Close() takes exactly 1 argument (3 given)
2018-06-06 17:22:45,298 : Error opening controller device: Close() takes exactly 1 argument (3 given)
2018-06-06 17:24:06,929 : GenMon Loadded for site: Nashua
```

While running at .17 everything worked wonderfully. After the upgrades, I get a bunch of unknown data and missing details.

### Expected Behavior

For log entries to be interpreted correctly and all data to be displayed as with .17.

### Actual Behavior

A bunch of "Unknown 0x00 : Alarm Code: 12850" and "01/29/18 09:00:03 Controller Fault 
        01/29/18 09:00:03 Unknown 0x00 "

Missing generator details such as serial number and service details (as described above).

### Steps to Reproduce (including precondition)

Updated to 1.9.20 while using a Generac/Honeywell 15KW.

### Screenshot on This Problem (if possible)

[Generator Monitor.pdf](https://github.com/jgyates/genmon/files/2078671/Generator.Monitor.pdf)

[Generator Monitor2.pdf](https://github.com/jgyates/genmon/files/2078674/Generator.Monitor2.pdf)

### Your Environment

- Genereator Model: G0060331 (Honeywell/Generac 15kw), Air cooled, Nexus
- Genmon Version: V1.9.20 (maybe present since .19)
- Generator Registers: 
Registers :
    Num Regs : 219
    Not Changed : 407838
    Changed : 61291
    Total Changed : 0.15

    Base Registers :
        0239 : 0000
        0039 : 0000
        0038 : 0001
        0034 : 0000
        0037 : 12a9
        0036 : 0000
        0033 : 0064
        0032 : 2d57
        05f7 : 03fb
        05f6 : 03fb
        05f5 : 03fb
        05f4 : 03fb
        05f3 : 03fb
        05f1 : 03fb
        05ed : 0000
        05ee : 0000
        0237 : 0000
        023b : 0000
        003a : 0000
        023e : 0000
        003b : 0000
        05fa : 03fb
        0059 : 03fb
        0058 : 03fb
        0053 : 03fb
        0052 : 03fb
        0057 : 03fb
        0056 : 0000
        0055 : 03fb
        0054 : 03fb
        0208 : 03fb
        005c : 0000
        005a : 0000
        005f : 0000
        005e : 0000
        005d : 0000
        003c : 0000
        000d : 0000
        000e : 161a
        000f : 0606
        000a : 0086
        000b : 0000
        000c : 0058
        0005 : 081d
        0006 : 0101
        0007 : 0000
        0000 : 0003
        0001 : 00000000
        0008 : 0000
        0009 : 00e8
        002a : 668e
        0012 : 0000
        0011 : 009c
        0010 : 0312
        0019 : 0003
        002b : 0000

smeisner, 2018-06-07:
I am very confused. I just found there was another update, so I updated to .21. Now I get the serial number, all the service details and the log entries are being interpreted. But now I don't see power output (graph or gauge) on the Monitor page.

jgyates, 2018-06-07:
Show picture of your Montor page and your Maintenace page.

smeisner, 2018-06-07:
These are from after the .21 upgrade. I didn't take any screen shots prior.

[image]

[image]

jgyates, 2018-06-07:
I asked for  the monitor and maintenance page, you gave me the maintenance and status page.

jgyates, 2018-06-07:
@smeisner ,

Based on the data, it looks like, at some point, you have modified your genmon.conf and uncommented the "evolutioncontroller" and possibly the "liquidcooled" entries. You may have just undone the modification recently and I expect that when you did your logs returned to normal. When you did this you have told the software to ignore the type of controller you have (You have a Nexus Air Cooled unit) and force the software to treat your controller like an Evolution Controller. The alarm registers are completely different between the Nexus and Evolution controllers so considering you modification to the configuration file, this is expected. Another indicator that you have done this is that you mentioned seeing a power gauge and graph. The ability to read current (and calculate power) is not supported on the Nexus controller. The only way you could have seen this is to modify the setting above since your registers indicate you have a Nexus Air Cooled unit (Register 0000=0003 -> Nexus Air Cooled). Put your configuration file back to the default an you will likely be fine.

See this page:
https://github.com/jgyates/genmon/wiki/Appendix-4-Known-Issues

In short if you have a Nexus and tell the software you have a Evolution you will get a lot of unknown values since the location of the alarms  is completely different. 

Regarding the errors in the log, as i mentioned in your query yesterday, these are messages related to shutting down(for example when you save your settings, the software is shutdown and restarted) and can be ignored.

closed 2018-06-07 by jgyates

smeisner, 2018-06-07:
I cannot understand how those settings could've been set accidentally. I have not modified the genmon.conf in a while...and I understand the layout of the registers is different for different controllers.

> Regarding the errors in the log, as i mentioned in your query yesterday, these are messages related to shutting down(for example when you save your settings, the software is shutdown and restarted) and can be ignored.

Yep...understood. I only included them for debug purposes. Wasn't complaining about them at all.

Again, (as I mentioned last night @ 10:38PM EST) since I updated to .21 (without modifying the conf files), it started working again...just no power info.

jgyates, 2018-06-07:
@smeisner ,

All of the problems you have reported can be tracked back to your SD card. If you have a SD card that is bad or a speed that is below what is recommended for use by a raspberry pi then you will see odd results from software that interacts with the disk. I would highly recommend shutting down your pi and removing you SD card and validating the conclusion that we can to in your other post (your SD card is slow or bad). You will likely only see other "strange" problems. Take a picture of the SD card and post it here or let me know the class number as I would like to add this to the wiki (i.e. warning don't use this speed of card). If it is not a class 10 SD card, get a new one that is. If it is a class 10 card, run a disk check on your system.

smeisner, 2018-06-07:
Thanks.

Based on your prior statement (about the quality/speed of the card), I put an order in for a good Class 10 card which will be delivered tomorrow. At that point, I will move the config to the new card. I will also document the old one with whatever details you want.

jgyates, 2018-06-08:
Sounds good. Thanks